xoseperez/espurna #2033: RGB, brightness and groups

kordian-kowalski, 2019-11-30:
It seems there is a bug with how brightness value is handled in connection with MQTT groups for colors.

I can't really pinpoint the cause of the issue, but if any device in an MQTT group for an RGB light has a brightness value lower than the maximum (255), then changing color or brightness on any device in that group will tend to decrease the brightness to 0. It seems that it has something to do with how the brightness value is recovered from RGB values - there is some inconsistency there.

mcspr, 2019-12-01:
Hey,

You sure that is still the case with the 1.14.0? Current group code sends and receives raw channel values that are independent of brightness. 1.13.5 was sending channel * brightness values, thus the eventual setting all channels to zero when brightness is less that max.

kordian-kowalski, 2019-12-01:
Hey,

Indeed - I was using 1.13, after updating to 1.14 everything seems to be working fine. Thanks!

closed 2019-12-01 by kordian-kowalski