xoseperez/espurna #2035: WIFI Network can't be removed

lblabr, 2019-12-01:
on 1.14 i tried to delete my Wifi Settings. my configured Network (1) could not be deleted.

i would like to give my plug to my neighbor, so it need to be removed...

Thanks in advance

Lars

mcspr, 2019-12-01:
Notably, this was not broken in 1.14.0 and was a problem in 1.13.5 too.
Workaround is to not “Delete network”, but to just erase ssid and passphrase in an existing one and “Save”.
Or, use DEBUG panel and do `del ssid0` and `del pass0` (note of the index. you can check all keys with `keys` or each key with `get ssid0` etc.)

When saving the data it never sends any “ssid” or “pass” arrays, thus settings backend never knows that we want those gone. Will be fixed by the https://github.com/xoseperez/espurna/pull/1986, so please don’t close this yet.

lblabr, 2019-12-01:
okay, thanks!

PR #1986 (2019-12-18): WebUI fix change detection of input elements
- use setOriginalsFromValues function right after elements had changed and
**only** with those elements
- add checkbox handling to setOriginalsFromValues to match hasChanged
- apply hasChanged after value is set to refresh originals after saving
- change parts that already set 'original' attribute manually

This was sort-of broken, because nothing really bad happens besides sending all of the available key-values to the device. 

<s>And I am not really sure what is up with the `schSwitch: [0xff]`. Need to look at it again and then I will commit this [... cut]
closed 2019-12-18 by mcspr