xoseperez/espurna #2040: problem with MQTT- domoticz

gatocamaron, 2019-12-03:
Hi,
First of all, congratulate you on the project since I find it very useful when setting up my home.
The problem I have is that espurna communicates very well in the MQTT-domoticz sense. When I change the status to espurna, domoticz receives well and updates the status.
The problem is when I change state in domoticz, it does not change the state in espurna. MQTT receives the order well but it seems that espurna does not interpret well and does not change the state of the relay.
The log  in espurna is as follows:
Espurna-domoticz:
[865905] [WEBSOCKET] Requested action: relay
[865906] [RELAY] #0 scheduled ON in 0 ms
[865910] [RELAY] #0 set to ON
[865913] [MQTT] Sending domoticz/in => {"idx": 2, "nvalue": 1, "svalue": ""} (PID 60)
[865914] [MQTT] Sending luzsofa/relay/0 => 1 (PID 61)
[865975] [MQTT] Publish ACK for PID 60
[866204] [MQTT] Publish ACK for PID 61
[866205] [MQTT] Received domoticz/out => {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "luzsofa",
   "dtype" : "Light/Switch",
   "id" : "00014052",
   "idx" : 2,
   "name" : "luzsofa",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}
[866224] [DOMOTICZ] Received value 1 for IDX 2
[866918] [RELAY] Setting relay mask: 0b1

Domoticz-espurna:

[952574] [MQTT] Received domoticz/out => {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "luzsofa",
   "dtype" : "Light/Switch",
   "id" : "00014052",
   "idx" : 2,
   "name" : "luzsofa",
   "nvalue" : 0,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "On/Off",
   "unit" : 1
}
[952592] [DOMOTICZ] Received value 0 for IDX 2

The problem is that it simply ignores the message and does not change the relay.
Thanks in advance.

m-kozlowski, 2019-12-03:
it's two issues down on the list: #2034
including solution.

gatocamaron, 2019-12-03:
Solucionado. Muchas gracias por la rapida respuesta. Saludos.

PR #2049 (2019-12-09): domoticz: proper datastructure for mask
fix #2034 
fix #2040
closed 2019-12-09 by mcspr