xoseperez/espurna #2041: How to get HEX codes for more 'IR_BUTTON_SET == XX' controllers.

davebuk, 2019-12-04:
Looking in `general.h` there are two configurations for Remote Buttons SET 1 and Remote Buttons SET 2. Neither of these control my device properly.

Is it possible to get the HEX codes **easily** to put into the `general.h` file? I presume I'd have to monitor the serial data somewhere while pressing the current remote buttons?

mcspr, 2019-12-16:
See https://gitter.im/tinkerman-cat/espurna?at=5df6944dac14cc652c8c8e91
https://github.com/xoseperez/espurna/commit/7efaaa8a8054cb4cb3c421ca92a0d98f39922b32#diff-a2f5f4b053c6da1d45efa792923e5b30 should be reverted, brightness is incorrectly applied based on the green color value.

I do see logging is done here:
https://github.com/xoseperez/espurna/blob/5cf687ffc1a08e4fd6656d3a2e746c9edc32f118/code/espurna/ir.ino#L375

> Secondly, should the functions such as: IR_BUTTON_MODE_EFFECT, LIGHT_EFFECT_FLASH activate the light effect modes, or do these functions stop working due to espurna being flashed?

`EFFECT` are just not handled in any way. I am not sure of the history, but I think the idea was to adapt a digital LED library (FastLED) first and try to utilize it's methods to create software effects. But that library change was never implemented

davebuk, 2019-12-16:
Pressing the IR buttons give various outputs at the debug webUI. I guess it's a matter of recording the received IR and aligning with the correct RGB values? It's not immediately obvious to me what the `irin` codes represent. It's there a reference somewhere?
```
[993979] [IR] Code does not match any action
[993980] [MQTT] Sending RGB001/irin => 3:16712445:32 (PID 1)
[994817] [IR] IRIN: 3:16771095:32
[994819] [MQTT] Sending RGB001/irin => 3:16771095:32 (PID 1)
[995584] [IR] IRIN: 3:16750695:32
[995584] [IR] Code does not match any action
[995586] [MQTT] Sending RGB001/irin => 3:16750695:32 (PID 1)
[996903] [IR] IRIN: 3:16750695:32
[996904] [IR] Code does not match any action
[996906] [MQTT] Sending RGB001/irin => 3:16750695:32 (PID 1)
[997457] [IR] IRIN: 3:16750695:32
[997457] [IR] Code does not match any action
[997459] [MQTT] Sending RGB001/irin => 3:16750695:32 (PID 1)
[998238] [IR] IRIN: 3:16750695:32
[998239] [IR] Code does not match any action
[998241] [MQTT] Sending RGB001/irin => 3:16750695:32 (PID 1)
[000271] [IR] IRIN: 3:16775175:32
[000272] [IR] Code does not match any action
[000273] [MQTT] Sending RGB001/irin => 3:16775175:32 (PID 1)
[000945] [IR] IRIN: 3:16775175:32
[000946] [IR] Code does not match any action
[000948] [MQTT] Sending RGB001/irin => 3:16775175:32 (PID 1)
[001910] [IR] IRIN: 3:16756815:32
[001912] [MQTT] Sending RGB001/irin => 3:16756815:32 (PID 1)
[002013] [MQTT] Sending RGB001/rgb => 0,0,142 (PID 1)
[002014] [MQTT] Sending RGB001/hsv => 240,100,56 (PID 1)
[002015] [MQTT] Sending RGB001/mired => 0 (PID 1)
[002017] [MQTT] Sending RGB001/channel/0 => 0 (PID 1)
[002021] [MQTT] Sending RGB001/channel/1 => 0 (PID 1)
[002026] [MQTT] Sending RGB001/channel/2 => 142 (PID 1)
[002031] [MQTT] Sending RGB001/brightness => 191 (PID 1)```

mcspr, 2019-12-16:
Pretty much, yes. I have reverted that commit in `dev`, so it should work as intended.

`...:<code>:...` is the button code value, but you need to convert it to hexadecimal first to compare with the button set codes (or convert hardcoded codes into decimal, that'll work too)
Comparing with the second buttonset, it is `{ 0xFFB04F, IR_BUTTON_MODE_RGB, 0x00BEBF }`, so it should set ch1 to 190 (0xbe) and ch2 to 191 (0xbf), but it does only set the ch2

davebuk, 2019-12-16:
Your reverted commit has fixed my issue. All commands work correctly using `set 1`. Like you said, the only thing that doesn't work are the "effects".

closed 2019-12-16 by davebuk

mcspr, 2019-12-17:
Also see #320 
Still unfinished as of 2y, and I was not able to find the original work on bitbucket that was the base for the [`fastled branch`](https://github.com/xoseperez/espurna/tree/fastled) :(
But my naive guess would be is that we could easily replace the existing transitions for something else. And I probably need to buy some esp-based led controller that is not for digital leds

davebuk, 2019-12-17:
I see there are some basic 'schemes' in the lights section of Tasmota.
https://github.com/arendst/Tasmota/wiki/Lights
All the other schemes are for addressable LEDs. I guess the fast led branch was for a similar setup to Tasmota?