xoseperez/espurna #2044: BROKER_REAL_TIME=1 causes crashes in the INFLUXDB module

mcspr, 2019-12-06:
The problem is with POW R2, it crashes few minutes after reboot when BROKER_REAL_TIME is set to 1. In web debug i see INFLUXDB messages in every 5-6 seconds.

_Originally posted by @MihataBG in https://github.com/xoseperez/espurna/issues/1968#issuecomment-562615803_

---

90% certain this is related to the issue with ESPAsyncTCP, fixed by the https://github.com/me-no-dev/ESPAsyncTCP/commit/75c2513c57ac8415eab3670dba5c64e73090f5ee

I don't really remember why exactly I skipped the update, besides the ASYNC SSL build problem. Need to reassess this again.
Or, do things like https://github.com/xoseperez/espurna/pull/1806 and just use AsyncClient directly.

MihataBG, 2019-12-10:
Also have a problem with INFLUXDB on "magichome-led-controller". I tried with BROKER_REAL_TIME=0 but module restart itself every time when do INFLUXDB message.

kordian-kowalski, 2019-12-12:
Same here, enabling InfluxDB will crash a Wemos D1 mini after sending a couple of measurements.

mcspr, 2019-12-13:
Right, previous version only sent influx message each sensor report, while BROKER_REALTIME=1 now makes it every reading.
Considering I already broke broker apart, I wonder if it is also viable to just remove REALTIME flag altogether and just have SensorReportBroker and SensorReadingBroker and make influx subscribe to report one.

Problem with SyncClient that I mentioned previously I quite random and depends on how well network behaves. But it is thoroughly broken, as discovered in the https://github.com/me-no-dev/ESPAsyncTCP/pull/115. We need to throw out old SSL stuff first though

closed 2019-12-16 by mcspr

mcspr, 2019-12-16:
#2061 reverts to the old behaviour and sends data every report, please check again with the current `dev` build

There are also two additional changes:
- string measurements are quoted
- http request now sends multiple measurements instead of one at a time.

PR #2061 (2019-12-16): Influxdb: fix sensor send frequency, use async client, send data in batches
fix #2044 

- use separate Broker instances for sensor read and report
- use async client directly
- send multiple measurements in a single request
- quote string measurements

Avoiding SyncClient for now, since there is no clear benefit of using it over WiFiClient. Move to async client directly (possibly this will be a base for generic client object). For helper object to work it needs to:
- have easily accessible buffer data structure for direct writes. control allocation, maybe allow persistent structures instead of [... cut]