xoseperez/espurna #2054: Is there a variable to get Time/Date in Epoch for King Wi-Fi Curtain switch.

AlbertWeterings, 2019-12-09:
I am trying to make the "King Wi-Fi Curtain Switch" work in ESPURNA. At this point I have a reasonable idea how to communicate with the MCU on the button board.

To complete this communication I need the date and time in Epoch. Is there a global variable available to use or a other simple routine to get this?

mcspr, 2019-12-11:
`now()`?
https://github.com/xoseperez/espurna/blob/5cf687ffc1a08e4fd6656d3a2e746c9edc32f118/code/espurna/ntp.ino#L128

AlbertWeterings, 2019-12-13:
Thanks @mcspr now() is working but only if I use now() in a file name that starts later in the alphabet than n It looks like Arduino IDE is compiling files in alphabetic order. so any file name after ntp.ino working. Is there a solution for it?

An other thing I had to solve is I needed a 13 digit time value now() is returning only 10 digits, so I concatenated the last 3 digits of millis to now()

mcspr, 2019-12-13:
Just add `#include <TimeLib.h>` to the file. IDE is combining all of .ino together alphabetically in a single .cpp, it just does not know about the now(). We just happen to not have any calls to timelib functions causing this issue

closed 2020-02-18 by stale[bot]