xoseperez/espurna #2055: LOHAS bulbs

cro, 2019-12-09:
Greetings. I purchased some Lohas bulbs from Amazon and successfully flashed Espurna on them. I bought both the standard lamp bulbs (https://www.amazon.com/gp/product/B07T7W7ZMW) and the BR30 version (https://www.amazon.com/gp/product/B07TMY394G/). The BR30s work fine, but the lamp bulbs do not. After closer inspection it appears that they are the "daylight" versions, offering both RGB color and daylight (2700k-6000k). Is there an Espurna build that will work with them? I've tried several and all I get is a dim blue flash no matter what I set the sliders to. 

The BR30-style bulbs say `10W E26 RGBW 19-10` on them.

The A19 bulbs say `9W E26 19-8 2700K-6000K+RGB`.

**Device information**
```
[117756] [MAIN] ESPURNA 1.14.0 (978cf2af)
[117757] [MAIN] xose.perez@gmail.com
[117757] [MAIN] http://tinkerman.cat

[117758] [MAIN] CPU chip ID: 0xB83549
[117761] [MAIN] CPU frequency: 80 MHz
[117764] [MAIN] SDK version: 1.5.3(aec24ac9)
[117768] [MAIN] Core version: 2.3.0
[117774] [MAIN] Core revision: 9826c6d
[117775] [MAIN] Build time: 1575072831
[117780] 
[117780] [MAIN] Flash chip ID: 0x144051
[117783] [MAIN] Flash speed: 40000000 Hz
[117786] [MAIN] Flash mode: DOUT
[117789] 
[117790] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[117800] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[117806] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[117811] [MAIN] Firmware size       :   510816 bytes /  125 sectors (   1 to  125)
[117818] [MAIN] Max OTA size        :   512000 bytes /  125 sectors ( 126 to  250)
[117826] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[117833] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[117840] 
[117842] [MAIN] EEPROM sectors: 251, 250
[117848] [MAIN] EEPROM current: 251
[117848] 
[117849] [MAIN] EEPROM:  4096 bytes initially |  1107 bytes used (27%) |  2989 bytes free (72%)
[117858] [MAIN] Heap  : 34008 bytes initially | 16576 bytes used (48%) | 17432 bytes free (51%)
[117866] [MAIN] Stack :  4096 bytes initially |  1396 bytes used (34%) |  2700 bytes free (65%)
[117873] 
[117874] [MAIN] Boot version: 31
[117877] [MAIN] Boot mode: 1
[117882] [MAIN] Last reset reason: External System
[117885] [MAIN] Last reset info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[117898] 
[117899] [MAIN] Board: LOHAS_E27_9W
[117902] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[117917] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[117920] [MAIN] WebUI image: LIGHT
[117923] 
[117927] [MAIN] Firmware MD5: dff3371bd8d2fff1b83cc1ea3990abce
[117932] [MAIN] Power: 3144 mV
[117933] [MAIN] Power saving delay value: 10 ms
[117937] 
```

**Tools used**
* macOS Catalina
* Firefox 72.0b3*

mcspr, 2019-12-11:
Do you have any hires photos of the board?

LOHAS_E27_9W is using MY92XX lights:
https://github.com/xoseperez/espurna/blob/5cf687ffc1a08e4fd6656d3a2e746c9edc32f118/code/espurna/config/hardware.h#L3370
Perhaps some of the pins are configured differently or the way lights are controlled at all

cro, 2019-12-11:
I figured it out, it works with Tasmota.  The bulb is this one: https://blakadder.github.io/templates/lohas_ZN070.html

I did manage to brick one bulb in my testing, it flashes slowly when I apply power, but no wifi access point ever appears.  I'm guessing I will have to disassemble it and try to connect via serial unless someone has a better idea.

closed 2019-12-11 by cro

cro, 2019-12-11:
[image]

mcspr, 2019-12-12:
Sorry, no idea about the cause of the brick. Is still running Tasmota? I think there is a failsafe when power-toggling the device ~5 times which should clear out internal settings storage. Ours does not have that extension yet, but should it reboot 5 times (due to possible crasehs) while still powered into a failsafe that would've brought up the AP.
If the device is crashing constantly, serial connection might help in at least showing the exception code and stack trace.

---

Just to explain what I am doing for similar issues when encountering tasmota template syntax:
https://github.com/arendst/Tasmota/blob/bfe2dc68cba880961ac4c18cb6512d0903c08172/tasmota/tasmota_template.h#L64-L68

GPIO number first row, second row are template numbers that directly map to the enum, so one can easily find out the value by counting line offset (using any decent text editor locally, for example):

 0|1|2|3|4|5|9|10|12|13|14|15|16|
-|-|-|-|-|-|-|-|-|-|-|-|-
0 | 0| 0| 0|38|37| 0| 0|41|39|40| 0| 0|

Number|Channel|GPIO
-|-|-
37|R|5
38|G|4
39|B|13
40|W|14
41|W|12

Which is close to the existing Arilux bulb, but the Blue and White channel are swapped:
https://github.com/xoseperez/espurna/blob/5cf687ffc1a08e4fd6656d3a2e746c9edc32f118/code/espurna/config/hardware.h#L1835-L1839

I think https://blakadder.github.io/templates could use at least GPIO numbering explanation for every template, need to look into how it is generated...

anthonycarboni, 2020-11-13:
Which version did you flash your b30s with?