xoseperez/espurna #2056: Is this smart plug already supported? (with photos)

jeannotvh, 2019-12-09:
I am a newcomer to the awesome Espurna universe, and instead of asking for creating support for a new device type and knowing there are dozens of synonyms out there, I decided to ask if anyone recognizes this ESP8265 smart plug, and can direct me to the synonym's binary.

I already flashed Espurna firmware that comes with tuya-convert, only to realize there is no MQTT support in the web page, and I need MQTT. So here it is:

Thanks guys!

[image]
[image]
[image]
[image]

jeannotvh, 2019-12-09:
This looks dangerously close to the Tonbux XS-SSA01.
Can anyone confirm?

jeannotvh, 2019-12-10:
Ok I dared to flash espurna-1.14.0-tonbux-xsssa01.bin, and got this message. Old version was still in there after reboot:
[image]

jeannotvh, 2019-12-10:
This appears to be a Teckin sp10, here's a photo for comparison:
 
[image]

mcspr, 2019-12-11:
See https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices-(HTTP-firmware)#socketsmulti-sockets
> GPIO0: LED2i-Red, GPIO2: Led1i-Blue, GPIO13: Button1, GPIO15: Relay1

Have you tried to compare pin numbers with any entries in the hardware.h?
https://github.com/xoseperez/espurna/blob/dev/code/espurna/config/hardware.h
TECKIN_SP21 and WORKCHOICE_ECOPLUG seem to be close enough.

jeannotvh, 2019-12-11:
Max, thank you for your quick reply!

I shall try flashing with these, and will get back to you!

Jean

jeannotvh, 2019-12-11:
Max,
  SP21 binary flashed flawlessly, and relay responds to commands.

Thank you!

closed 2020-02-17 by stale[bot]