xoseperez/espurna #2057: Found an old version of MagicHome PCB (MINI-WF-5CH-A _v1.0) if there any support for it

KeyTwoGleb, 2019-12-10:
Hello, i have found an old and rare version of MagicHome cotroller (board: MINI-WF-5CH-A_V1.0) with RF-remote and 5ch LED OUT (RGB+CCT/RGBWW)... I not found any info or photos it internet about this PCB... Soildering quality is very weak... Prevousily i have used it over 2 years 24\7 powered on, but currenty i need support for WebUI or other way to control LED strip from PC...

Did anyone have practice with this board? Are there any major differences compared to more popular versions of MagicHome? 

Today i will dump original FW, in near 2 days i want try to build and flash ESPurna on this board...
But this is my first practice, before I had no experience with ESP controllers...

There is photo 
[image]
[image]

davebuk, 2019-12-10:
[https://github.com/xoseperez/espurna/wiki/Hardware-Magic-Home-LED-Controller](url) and [https://github.com/xoseperez/espurna/wiki/Hardware-Arilux-AL-LCxx](url) has some general information about these devices. Device `MAGICHOME_ZJ_ESPM_5CH_B_13` might work but the GPIO configuration could be wrong. If so, you'll have to build your own binary with the correct config.

KeyTwoGleb, 2019-12-10:
This board for RF using IC marked "480R", it's Synoxo SYN480R RF Transceiver, and red wire soldered to "E1" pin - i think is antenna, black wire upped - is unsolderd at box...

This controller using standart 21 button RF remote,,,

Today i have received new "MagicHome RGBWW\RGB-CCT LED Controller" from Aliexpress, and it have similar RF remote and sticker on controller housing, but housng shape slighty differemt,  tommorow i want to dissasamble it to check and dump PCB)! 

Sorry for my bad English...

davebuk, 2019-12-11:
From the pre-built binaries, `MAGICHOME_ZJ_WFMN_B_11` is RGBW with RF. `MAGICHOME_ZJ_ESPM_5CH_B_13` is RGBWW without RF. You could try either of those to start with and see what works?

KeyTwoGleb, 2019-12-17:
> From the pre-built binaries, `MAGICHOME_ZJ_WFMN_B_11` is RGBW with RF. `MAGICHOME_ZJ_ESPM_5CH_B_13` is RGBWW without RF. You could try either of those to start with and see what works?

I back to this project, today i will test MAGICHOME_ZJ_WFMN_B_11 and MAGICHOME_ZJ_ESPM_5CH_B_13 FW (also i will try to fix it -  but as I wrote earlier - this is my first experience with this platform and code on subject module), and i will upload original binary of these boards. Sorry for late reply (currently i have some health problems)....

KeyTwoGleb, 2019-12-19:
I have dump original binary MINI-WF-5CH-A_V1.0.bin (4Mb) https://drive.google.com/file/d/1pJkCen55tWuZOgxpKXDn6mbJsNShEBck/view?usp=sharing

i tested magichome-zj-wfmn-b-11.bin - its not boot (i cant see any WiFi AP), and arilux-al-lc02.bin, its works, but color channels incorrectly assigned....

closed 2020-02-24 by stale[bot]