xoseperez/espurna #2058: Is there a standard for the messages sent in the rfin topic?

wsw70, 2019-12-10:
I use a Sonoff RF Bridge to gather the messages from my RF 433 wall switches and send them to MQTT for further processing. A typical message from a wall switch is

    xxxxxxxxxxxxyyyyyy

- The first 12 chars (shown as `x` above) are unpredictable (maybe random) and I do not know what their meaning is. They change between the switch presses. 
- The last 6 (`y` above) are unique to the switch and this is how I know which one was used.

What is told above is purely experimental.

I also see, from time to time messages such as `3DD6008203DE000000`. They do not come from my switches and must be sent be something else, probably a device at the neighbour's.

This leads to my question: **is there a common standard for the breakdown of such RF 433 messages?** Or some "commonly accepted practices" - maybe across one producer or a type of device (switch vs. meteo station for instance)?

mcspr, 2019-12-11:
See https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA4

We cut protocol header and tail when sending things to MQTT, so we end up with: `24 E0 01 40 03 84 D0 03 58`:
Data | Description
-- | --
0x24, 0xE0 | Tsyn (9440µs)
0x01, 0x40 | Tlow (320µs)
0x03, 0x84 | Thigh (900µs)
0xD0, 0x03, 0x58 | 24bit Data

Where Tsyn, Tlow and Thigh describe the protocol of that specific device.

Note that those numbers can change a bit each transmission. If the code was previously learned and ESPurna knows about it, we will return the learned string instead of the real payload:
https://github.com/xoseperez/espurna/blob/5cf687ffc1a08e4fd6656d3a2e746c9edc32f118/code/espurna/rfbridge.ino#L285-L287

wsw70, 2019-12-11:
@mcspr Thanks Max for the information. I have a few questions:

>  we cut raw data header and tail when sending things to MQTT, 

In your example (`24 E0 01 40 03 84 D0 03 58`), the first 12 bytes "describe the protocol of that specific device". What does this mean exactly?   
My lack of understanding mainly comes from the fact that these 12 bytes change between presses (for the case of a simple wall switch) so I do not get how they can "describe the protocol". You also mention that "those numbers can change a bit each transmission" → is that the change I see between presses in the header part? (I will have to capture a few and see whether the changes on the relevant bytes are small)

Also - would you know if the payload of the device is standardized to some point? In other words - whether I can make sense of it or is it just proprietary information I need to check device by device (or maybe chipset by chipset)?

mcspr, 2019-12-12:
ref: https://github.com/sui77/rc-switch/wiki/KnowHow_LineCoding
Timings stay around the same value of μs, with 1/3 and 3/1 length for bit pulses and 1/31 for sync

wsw70, 2019-12-12:
Thanks @mcspr for all the info!

closed 2019-12-12 by wsw70