xoseperez/espurna #2060: longer interval/specific date for scheduler

coelner, 2019-12-14:
I can't find any directly related issue.

I need to activate a switch by a longer period than on daily base. Once in a month I want to start a system. Sometimes I would need a specific date and not an interval.

mcspr, 2019-12-24:
As I am not really in favour of adding yet another kv pair to the scheduler engine... I guess we can extend RPN rules instead? See this example:
https://github.com/xoseperez/espurna/wiki/RPN-Rules#open-light-at-night-if-there-is-presence

It matches the existing functionality of scheduler to compare hour, minute and day of week. e.g., here's the piece of code to return weekday match:
https://github.com/xoseperez/espurna/blob/37763f1ad42b77008c0a13aa460db87ceb81345c/code/espurna/rpnrules.ino#L134-L140

We can add an additional operator `day` / `dom` / or longer `dayofmonth` to match the month day instead

mcspr, 2024-06-15:
Part of the scheduler update, YYYY-MM-DD can be specified in addition to DOW and HH:MM
https://github.com/xoseperez/espurna/wiki/Scheduler#time-string

closed 2024-06-15 by mcspr