xoseperez/espurna #2062: Platformio lib_deps instaling error

oscarsan1, 2019-12-14:
When building for first time with platformio I get an error every missing library, but every building task Platformio intalls only one more library. The error happens only with web referenced libraries like:
https://github.com/marvinroger/async-mqtt-client#v0.8.1
https://github.com/xoseperez/debounceevent.git#2.0.5
https://github.com/xoseperez/eeprom_rotate#0.9.2

but not with other "direct" libraries, that are instaled without error:
    Brzo I2C
...
    Embedis

Then I have to run build task 21 times until the build success. Each time one more library is instaled.

The error is in this case with Installing NtpClient, next I build I get the next library error RFM69:

Processing wemos-d1mini-relayshield (board: esp12e; platform: espressif8266@1.5.0; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 1.5.0 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 1.20300.1 (2.3.0), tool-esptool 1.409.0 (4.9)
Converting espurna.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: **Installing NtpClient**
git version 2.24.0.windows.2
Cloning into 'c:\_proyecto\_Espurna\espurna-master 1.14.0\code\.pio\libdeps\wemos-d1mini-relayshield\_tmp_installing-wmpzjmhu-package'...
HEAD is now at 0942ebc Merge remote-tracking branch 'upstream/develop' into develop
PermissionError: [WinError 5] Acceso denegado: 'c:\\_proyecto\\_Espurna\\espurna-master 1.14.0\\code\\.pio\\libdeps\\wemos-d1mini-relayshield\\_tmp_installing-wmpzjmhu-package\\.git\\objects\\pack\\pack-efbfd8063d05b562ed75cf3b48da369c59a5fe32.idx':
  File "C:\Users\Osr\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Osr\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Osr\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Osr\.platformio\platforms\espressif8266@1.5.0\builder\main.py", line 375:
    target_elf = env.BuildProgram()
  File "C:\Users\Osr\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Osr\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\Osr\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Osr\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Osr\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\Osr\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "C:\Users\Osr\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 301:
    force=force,
  File "C:\Users\Osr\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "C:\Users\Osr\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 497:
    return self._install_from_tmp_dir(_tmp_dir, requirements)
  File "C:\Users\Osr\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 583:
    shutil.move(tmp_dir, pkg_dir)
  File "C:\Users\Osr\.platformio\python37\lib\shutil.py", line 578:
    rmtree(src)
  File "C:\Users\Osr\.platformio\python37\lib\shutil.py", line 516:
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\Osr\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\Osr\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\Osr\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\Osr\.platformio\python37\lib\shutil.py", line 400:
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\Osr\.platformio\python37\lib\shutil.py", line 398:
    os.unlink(fullname)
===================================================== [FAILED] Took 5.12 seconds =====================================================

oscarsan1, 2019-12-14:
I also found error with missing libraries:

    SPIFlash
    DNSServer 

After I include this libraries to the lib_deps in the platformio.ini finaly build without errors.

mcspr, 2019-12-14:
Does it change anything if you use global library storage?
```diff
diff --git a/code/scripts/pio_pre.py b/code/scripts/pio_pre.py
index adcdc13b..c8b63546 100644
--- a/code/scripts/pio_pre.py
+++ b/code/scripts/pio_pre.py
@@ -93,4 +93,4 @@ if os.environ.get("ESPURNA_PIO_SHARED_LIBRARIES"):
         storage = get_shared_libdeps_dir("common", "shared_libdeps_dir")
         print("using shared library storage: ", storage, file=sys.stderr)

-    subprocess_libdeps(env.GetProjectOption("lib_deps"), storage)
+subprocess_libdeps(env.GetProjectOption("lib_deps"), None
```
This will install things to ~/.platformio/lib

```
SPIFlash
DNSServer 
```
Are builtin libraries, there is no need to add them unless you have modified .ini even more while testing this and now use a different library dependency finder.

---

I am not sure what exactly changed in PIO 4.1.0, but this is not the first report of broken library installs with git:
https://github.com/xoseperez/espurna/issues/2006

I'd post this to the https://github.com/platformio/platformio-core instead, with some example project like this:
- platformio.ini
```ini
[env:d1_mini]
platform = espressif8266
framework = arduino
board = d1_mini
lib_deps =
    https://github.com/marvinroger/async-mqtt-client#v0.8.1
    https://github.com/me-no-dev/ESPAsyncTCP#7e9ed22
    https://github.com/me-no-dev/ESPAsyncWebServer#b0c6144 
```
- src/main.cpp
```cpp
#include <Arduino.h>
#include <ESPAsyncTCP.h>
#include <AsyncMqttClient.h>
#include <ESPAsyncWebServer.h>
void setup() { }
void loop() { }
```
With a couple of lib_deps pointing to git. Funny thing is installation is actually successful, but the temporary file removal is broken because of either VSCode or some antivirus solution.

closed 2019-12-14 by mcspr