xoseperez/espurna #2072: Not working domoticz control on S20

kaju666, 2019-12-18:
Hi

I buyed S26 and flashed it with new espruna 1.14.
Unfortunatly after adding next device in Domoticz and configure the espruna it's somehow not working.
Debug say's that is correctly receiving command from domoticz but wor some reason it not chnaging the switch.
Have other SONOFF Basic,RF,T1 in house with espruna 13.5 and they are working fine.
[image]
[image]

Carmelo66, 2019-12-19:
is a known problem and not solved for all sonoff devices, or rather it is solvable for those who know where to put their hands in the firmware. With the firmware 14.0 does not work neither domoticz nor Alexa, you have to downgrade to version 12.5 and then upgrade to version 13.2 to find both Domoticz and Alexa.

Maestro74, 2019-12-19:
It's the same with version 1.14 for me. There seem to be problems with MQTT communication here. Apparently, this doesn't work. Solution: If you have already flashed to 1.14, then simply downgrade via WebGUI 1.13.5. It works perfectly. Not only S20 devices are affected.

mcspr, 2019-12-19:
Guys, please use search before creating new issues. Issue template is there for a reason.
Duplicate of #2034, #2040 

Same for Alexa, I don't have any ability to test it, but you may use any other external integration in the mean time (OpenHAB, HomeAsssistant etc. all have voice integrations running on top of their own code):
#1904 , #1993 

Nightly binaries are there too, so you don't have to build things:
https://github.com/mcspr/espurna-nightly-buillder/releases

closed 2019-12-19 by mcspr