xoseperez/espurna #2076: Support for Woox GU10 Lamp

jkalousek, 2019-12-19:
[image]

I bought few of thees lamps as they looked similar to alredy supportet GU10 lamps.
I managed to flash them with Tuya and generic GU10 firmware but the light was constantly on.

I tried few other FWs via OTA until I got to iWoole LED Table Lamp. With that firmware light turned off and I was able to controll white chanel with (I assume) green value.
So I took source code for iWoole and started to tinkering with it, unfortunatelly after another OTA light stopped responding and as I do not see any obvious labeled pads to solder to I wanted to ask firts if there isn't a light with similar chips or somone have any Idea where I could start.

[image]
[image]
[image]

jkalousek, 2019-12-19:
PIN labeled as D seems to be connected to pin 9 OF ESP8266EX.
PIN labeled as C seems to be connected to pin 4 OF ESP8266EX.

jkalousek, 2019-12-19:
[image]
I didn't realize it but there are some labels on bottom of the board.

IO04: C
IO14: D

jkalousek, 2019-12-19:
It seems that this could be quite similar to https://github.com/xoseperez/espurna/issues/568

mcspr, 2019-12-20:
If it is, we need to implement it first. Thanks for the research!

From the referenced issue links, see https://github.com/arendst/Tasmota/pull/4991 for raw sw SPI code that drives the leds. I'd expect that sm16717 tasmota configuration to work just fine with this bulb.

And interestingly enough, FastLED lib supports it - https://github.com/FastLED/FastLED/wiki/Chipset-reference#other-stuff
> SM16716 - implemented because a couple people asked for it. Terrible protocol.

(Since there was an issue regarding fastled integration anyway)

BlueBull010, 2020-03-01:
I figured this one out I think, it has exactly the same pin layout as the below bulb

https://templates.blakadder.com/cleverio-51398.html

It's working in my case now. Thank you for putting me on the right track, you guys rock!