xoseperez/espurna #2078: Change/Set the MAC address?

mattster98, 2019-12-20:
I have two different devices that I've flashed with espurna and have discovered the hard way that they have the same MAC address.  I'd like to change one of them, but from what I can tell this has to be done in software - I can't change it with some utility and have it stick in the hardware.

My suggestion is to add this as a settable variable in the debug terminal in the wifi module.  That would let me fire one of the devices up, make up a new non-conflicting MAC, and save it so I can have both devices online at the same time since now they'll have different MAC addresses.

Thanks!

mcspr, 2019-12-20:
Based on:
https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf **3.5.52.** *wifi_set_macaddr*
And https://github.com/esp8266/Arduino/issues/413#issuecomment-458551253

Settings are driven through a C++ class, so some adjustments need to happen there so it could work without it (or at least data parsing part)
For local build I think there is even no need to modify .ino files and it will work by dropping somefile.c with `preinit()` function inside. It will be picked up by the build system and linked into the binary.

padcom, 2023-09-02:
Any progress on this?

mcspr, 2023-09-02:
Follow-up for the above
https://github.com/esp8266/Arduino/issues/8554#issuecomment-1162376481

https://github.com/xoseperez/espurna/blob/4f22960a90408ebce1f4a38530a290818172ceab/code/espurna/wifi.cpp#L1532
At least for now, modifying the source is the way. Meaning here, just before the connection happens where we definitely know interface is UP
```cpp
const uint8_t mac[] { 0xaa, 0xbb, 0xcc, 0xdd, 0xee, 0xff };
wifi_set_macaddr(STATION_IF, &mac[0]);
```

(or through settings / build flags, depending on how you'd want this to work)

padcom, 2023-09-02:
Thanks! I don't mind rebuilding the firmware - it's simple enough