xoseperez/espurna #2082: Mqtt group problem of 4 devices.

tonymerce, 2019-12-23:
Hello everyone, I have a problem with the synchronization of 4 devices in the Switch group mqtt section. The problem is that once in a while, when I turn on a switch (sonoff) they turn on and off all 4 in an endless loop. Forgive my English. Thank you

mcspr, 2019-12-24:
Do you mean 4 separate devices or 4 switches on the same board?
Any changes with the current release?

tonymerce, 2019-12-25:
4 separate devices.  I have not tried the latest version because it does not work with domoticz.

mcspr, 2019-12-25:
Have you tried the latest release or development build?
1.14.0 had the domoticz bug, 1.14.1-dev should not.

tonymerce, 2019-12-25:
Ok, thanks a lot.

tonymerce, 2019-12-25:
I'll try today

tonymerce, 2019-12-25:
Hello, I have already tried the latest version and effectively with domoticz it already works but I still have the same problem of the 4 synchronized switches in mqtt group. If I activate it by mistake 2 times in a second, all loops and there is no way to stop them. To stop them I have to turn off the mosquitto service. My installation is as follows:
31 light devices with espurna
Mosquitto server in windows 10.
Domoticz in windows 10.
Homebridge in windows 10.
Maybe if there was a time in the synchronization mqtt group to not attend messages (feedback) of change of state until that time or something similar passes. I have not found that time, Thanks in advance and Forgive my english.

mcspr, 2019-12-25:
Well, you can specify `relayDelayOnN` or `relayDelayOffN` (where N is the switch number starting from 0) [in the settings](https://github.com/xoseperez/espurna/wiki/Terminal#settings), but I don't think that will do anything in this scenario. With those settings, after any switch to ON or OFF device will wait for specified time in milliseconds before actually turning ON or OFF and sending any data to MQTT group and base topics.

I am not really sure how we can solve this when domoticz is also in play, what it sounds like you are receiving commands from dcz out-of-sequence with group topic.
Does using the same domoticz idx for every device also solve this? There could be an additional "virtual" switch created and is in sync with the real one.

tonymerce, 2019-12-27:
Hi, it seems that putting the same idx and removing the mqtt group works. Thank you.

mcspr, 2019-12-27:
Cool. I'll look at groups code in any case, since we do need some upgrade for lights too.
I wonder if we need to extend payload with some additional data, like sequence number or something like that, to avoid such cycling

closed 2019-12-27 by mcspr