xoseperez/espurna #2084: PLUGIN:  Reading sensors within plugin not working on sample plugin.

keejayzee, 2019-12-27:
Uncommenting the following lines of plugin1.ino:
<code>
#if SENSOR_SUPPORT
        unsigned char _sensorID;
        double _sensorValue;
        _sensorValue = _getSensorValue(_sensorID, "current");
        _sensorValue = _getSensorValue(_sensorID, "filtered");
        _sensorValue = _getSensorValue(_sensorID, "reported");
#endif
</code>

And also:
<code>
#if SENSOR_SUPPORT
    double _getSensorValue(unsigned char id, const char * type ) {
        char * p = ltrim((char *)type);        
        if (strcmp(p, "current") != 0) return _magnitudes[id].current;
        else if (strcmp(p, "filtered") != 0) return _magnitudes[id].filtered;
        else if (strcmp(p, "reported") != 0) return _magnitudes[id].reported;
        return 0;
    }
#endif
</code>

Does not even compile as it's not aware of <code>_magnitudes</code>. I tried <code>#include "sensors/BaseSensor.h"</code> and then adding:
<code>`
#if SENSOR_SUPPORT

struct sensor_magnitude_t {
    BaseSensor * sensor;        // Sensor object
    BaseFilter * filter;        // Filter object
    unsigned char local;        // Local index in its provider
    unsigned char type;         // Type of measurement
    unsigned char decimals;     // Number of decimals in textual representation
    unsigned char global;       // Global index in its type
    double last;                // Last raw value from sensor (unfiltered)
    double reported;            // Last reported value
    double min_change;          // Minimum value change to report
    double max_change;          // Maximum value change to report
};

    std::vector<BaseSensor *> _sensors;
    std::vector<sensor_magnitude_t> _magnitudes;

#endif`
</code>

And all I get are compile errors  on duplicated definitions of <code>_sensors</code> and <code>_magnitudes</code>

I'm stuck here, any help would be appreciated.

PS: The plugin sample itself has quite a few errors and does not compile. I fixed those, but as it's not part of the code itself I can't ask for a PR

mcspr, 2019-12-27:
Since that .zip file is not part of the code base, as you noticed, we can't easily keep track of it's contents. And since the original author is long gone and the plugin idea was never improved upon, I am inclined to remove the .zip file mention and replace the page with an example code instead that references the actual working and up-to-date API.
I'll try to rewrite it this weekend.
 
As previously mentioned, you would have better idea of internals by using something like thingspeak.ino or influxdb.ino as an example instead of relying on that page.
There is a single entry-point for plugin code - `extraSetup()`. Structure is weird in Arduino, all *.ino are concatenated together alphabetically + IDE tries to magically generate function prototypes, but it can fail and you'd need to add `#include` directly. So I am not even sure how that example code with sensor worked at all originally.

keejayzee, 2019-12-27:
> Since that .zip file is not part of the code base, as you noticed, we can't easily keep track of it's contents. And since the original author is long gone and the plugin idea was never improved [...]

Thanks Max, that'd be awesome!

> As previously mentioned, you would have better idea of internals by using something like thingspeak.ino or influxdb.ino as an example instead of relying on that page.

Thanks again for the tip. I'm going to look through over those files right away.

> Structure is weird in Arduino, all *.ino are concatenated together alphabetically + IDE tries to magically generate function prototypes, but it can fail and you'd need to add `#include` directly. 

I'm using platformio, but I'll take that into account from now on, just in case.

closed 2019-12-27 by keejayzee