xoseperez/espurna #2085: Problems with Sonoff touch devices

tonymerce, 2019-12-27:
Hello, lately I have problems with the latest sonoff touch devices that I have acquired. After putting Spurna on them and signing up for home automation, homebridge, etc. they work, but every time they reset and turn on by themselves, leaving them on until I realize, I have other sonoff touch that do not have this problem. Does this problem happen to someone? Thanks in advance.

mcspr, 2020-01-03:
*ping*
Could it be built-in RF? When button is held, it will learn any incoming 433 RF code after the beep, causing button press whenever it receives that code again. I think the on-board "reset" button removes any codes.... Or holding the button for even more time and repeating the code.

tonymerce, 2020-01-06:
No, the ones that fail me are the touch ones, those don't have RF. I also have T1 but those don't fail. I fail 2 of 20. I will change them to see if it is solved. Thank you.

closed 2020-03-13 by stale[bot]