xoseperez/espurna #2094: MQTT: Relay `query` / `3` command

mcspr, 2019-12-31:
**Bug description**
Citing https://github.com/xoseperez/espurna/wiki/MQTT#relays
> The special query payload does not change the relay state but triggers a state topic message.

This was removed at some point during the 1.14.0, but I have not seen any bug reports so far.

**Steps to reproduce**
\- 

**Expected behavior**
(?) Perhaps the correct way is to use a different method of trigger it instead of hijacking relay setter. It is not really effective way to receive up-to-date status from multiple relays, perhaps there should be an "action" to send back relay status for all relays (or full heartbeat, with every other status)
But, not sure why do that at all, since we *should* know the status anyway

**Screenshots**
\-

**Device information**
\-

**Additional context**
\-

jordifm2377, 2020-01-14:
Will this feature be enabled again in the next versions ?
I've found it very useful to been able to query the status on demand. Or maybe is there any other way to know the status of the device on demand ?

mcspr, 2020-01-14:
Depends. My main question is, why it is needed at all? If client is connected to broker, it already received latest status at some point. Would it not be easier to remember latest status and use that as a reference?

jordifm2377, 2020-01-15:
You're right, the client will get the status at some point.
In my case, I use the query option when the client boots up and during this boot up it wants to know the status of the relay. And not need to wait for the heartbeat.
Also you're right about remembering the last status and start from here. But this would mean that the client has to keep this status persisted somewhere, because if it only keeps the status in memory as soon as the client reboots, it loses the status. (it happens to me when a power cut occurs)

Anyway, it is not a big deal :) ... just I think that we might be losing functionality that maybe it's not much used but it doesn't disturb.

mcspr, 2020-01-15:
Yes, but is there a need to wait for the heartbeat if [retained messages](https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages/) are configured? Relay status by default is retained, so you would receive relay status message at the moment that you are connected to the broker. Note that broker configuration needs to have some persistence configured too for retained messages to work in case broker itself restarts.
[For mosquitto](https://mosquitto.org/man/mosquitto-conf-5.html) it is `persistence true`

`3` payload is never used by the code, so I assumed that was part of webui process at some point and removed that :) I am more in favour of adding a command action that will send relay status for all of the relays (ref: #1546)

jordifm2377, 2020-01-15:
Retained messages seem a good alternative. Thanks for the suggestion :)

I like the idea that you mentioned on #1546

mcspr, 2020-03-15:
Yes, forgot to update this, there is an added `heartbeat` action similar to `reboot`, simply parsing text of the action to either MQTT or HTTP /action/set.

I could add arguments though, to filter allowed heartbeat values like `hbReport` does through a bitmask. e.g. to only send relays it would mean to send `{root}/action/heartbeat/set` `1024` (meaning 1 << 10) bit, see utils.ino / utils.h).