xoseperez/espurna #2104: Software Watchdog. Fatal exception:4 flag:3 (SOFT_WDT) reboots.

davebuk, 2020-01-13:
I have recently upgraded all my devices to 1.14.1 core 2.6.3. Two devices (SWA1 and Sonoff Basic) keep rebooting due to, I believe, the Software Watchdog timer. They were both fine on version 1.13.6-dev core 2.5.2 and had an uptime of 90 days before upgrading. The uptime seems to vary and the devices reconnect to WiFi/mqtt and work, but the relay output turns OFF after the reset (as configured in the 'SWITCHES' section).

Does the following help point to a particular reason why the Watchdog timer is being triggered?

**SWA1**

```
[994800] [WEBSOCKET] Requested action: dbgcmd
[994802] 

---8<-------

[994803] [MAIN] ESPURNA 1.14.1 (62ad7da3)
[994803] [MAIN] xose.perez@gmail.com
[994803] [MAIN] http://tinkerman.cat

[994803] [MAIN] CPU chip ID: 0x854E0A
[994803] [MAIN] CPU frequency: 80 MHz
[994804] [MAIN] SDK version: 2.2.2-dev(38a443e)
[994804] [MAIN] Core version: 2.6.3
[994808] [MAIN] Core revision: 3d128e5c
[994808] [MAIN] Build time: 1578863237
[994809] 
[994809] [MAIN] Flash chip ID: 0x15400E
[994809] [MAIN] Flash speed: 40000000 Hz
[994809] [MAIN] Flash mode: DOUT
[994809] 
[994810] [MAIN] Flash size (CHIP)   :  2097152 bytes /  512 sectors (   0 to  511)
[994815] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[994816] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[994818] [MAIN] Firmware size       :   540048 bytes /  132 sectors (   1 to  132)
[994819] [MAIN] Max OTA size        :   483328 bytes /  118 sectors ( 133 to  250)
[994819] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[994820] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[994820] 
[994820] [MAIN] EEPROM sectors: 251, 250
[994823] [MAIN] EEPROM current: 250
[994823] 
[994823] [MAIN] EEPROM:  4096 bytes initially |   489 bytes used (11%) |  3607 bytes free (88%)
[994826] [MAIN] Heap  : 42896 bytes initially | 17104 bytes used (39%) | 25792 bytes free (60%)
[994827] [MAIN] Heap  : 24008 contiguous bytes available (7% fragmentation)
[994827] [MAIN] Stack :  4096 bytes initially |  1600 bytes used (39%) |  2496 bytes free (60%)
[994828] 
[994828] [MAIN] Boot version: 4
[994831] [MAIN] Boot mode: 1
[994832] [MAIN] Last reset reason: Software Watchdog
[994832] [MAIN] Last reset info: Fatal exception:4 flag:3 (SOFT_WDT) epc1:0x4025ab86 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[994832] 
[994833] [MAIN] Board: LINGAN_SWA1_DALLAS
[994833] [MAIN] Support: API BROKER BUTTON DEBUG_TELNET DEBUG_WEB LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL WEB 
[994834] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[994834] [MAIN] Sensors: DALLAS 
[994836] [MAIN] WebUI image: SENSOR
[994836] 
[994836] [MAIN] Firmware MD5: 31c731b6ec136d293dab6ccc5a9180f4
[994837] [MAIN] Power: 3409 mV
[994838] [MAIN] Power saving delay value: 10 ms
[994838] 

---8<-------
```

**Sonoff**
```
[857342] [WEBSOCKET] Requested action: dbgcmd
[857348] 

---8<-------

[857348] [MAIN] ESPURNA 1.14.1 (62ad7da3)
[857348] [MAIN] xose.perez@gmail.com
[857349] [MAIN] http://tinkerman.cat

[857349] [MAIN] CPU chip ID: 0x9785FC
[857349] [MAIN] CPU frequency: 80 MHz
[857349] [MAIN] SDK version: 2.2.2-dev(38a443e)
[857349] [MAIN] Core version: 2.6.3
[857353] [MAIN] Core revision: 3d128e5c
[857353] [MAIN] Build time: 1578576965
[857353] 
[857353] [MAIN] Flash chip ID: 0x144051
[857354] [MAIN] Flash speed: 40000000 Hz
[857354] [MAIN] Flash mode: DOUT
[857354] 
[857355] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[857357] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[857357] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[857357] [MAIN] Firmware size       :   544688 bytes /  133 sectors (   1 to  133)
[857358] [MAIN] Max OTA size        :   479232 bytes /  117 sectors ( 134 to  250)
[857358] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[857359] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[857359] 
[857359] [MAIN] EEPROM sectors: 251, 250
[857361] [MAIN] EEPROM current: 251
[857362] 
[857362] [MAIN] EEPROM:  4096 bytes initially |  1096 bytes used (26%) |  3000 bytes free (73%)
[857363] [MAIN] Heap  : 42400 bytes initially | 18152 bytes used (42%) | 24248 bytes free (57%)
[857363] [MAIN] Heap  : 23144 contiguous bytes available (5% fragmentation)
[857363] [MAIN] Stack :  4096 bytes initially |  1776 bytes used (43%) |  2320 bytes free (56%)
[857363] 
[857364] [MAIN] Boot version: 7
[857366] [MAIN] Boot mode: 1
[857366] [MAIN] Last reset reason: Software Watchdog
[857367] [MAIN] Last reset info: Fatal exception:4 flag:3 (SOFT_WDT) epc1:0x4025b9f2 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[857367] 
[857368] [MAIN] Board: ITEAD_SONOFF_BASIC_BMX280
[857368] [MAIN] Support: API BROKER BUTTON DEBUG_TELNET DEBUG_WEB I2C LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL WEB 
[857368] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[857368] [MAIN] Sensors: BMX280 
[857370] [MAIN] WebUI image: SENSOR
[857370] 
[857371] [MAIN] Firmware MD5: 604c5a60e8fdfb34a55721b3bf95b228
[857372] [MAIN] Power: 3386 mV
[857372] [MAIN] Power saving delay value: 10 ms
[857372] 

---8<-------
```

mcspr, 2020-01-13:
`crash` command prints out stack trace, exception addresses need to be decoded though:
https://github.com/mcspr/EspArduinoExceptionDecoder
(you may need to remove `[...timestamp...]` strings from the beginning)

Can you also attach resulting .elf and .bin from BMX build?
See `.pio/build/<env-name>/firmware.<extension>`

davebuk, 2020-01-13:
I don't always have a  `crash` output, but one I saved from yesterdayfor the SWA1 was:

```
---8<-------

[128447] +OK
[154001] [WEBSOCKET] Requested action: dbgcmd
[154003] [DEBUG] Latest crash was at 345409 ms after boot
[154003] [DEBUG] Reason of restart: 3
[154003] [DEBUG] Exception cause: 4
[154003] [DEBUG] epc1=0x4025ab86 epc2=0x00000000 epc3=0x00000000
[154003] [DEBUG] excvaddr=0x00000000 depc=0x00000000
[154004] sp=0x3fffeef0 end=0x3fffffb0 saved=0x0080

[154004] [DEBUG] >>>stack>>>
[DEBUG] 3fffeef0: 40277df0 00000000 00000001 4023aa3d 
[DEBUG] 3fffef00: 00000000 3ffee6e8 00000000 00000000 
[DEBUG] 3fffef10: 4025a9a4 3ffee6e8 3ffef6a0 60000600 
[DEBUG] 3fffef20: 4025c615 3ffee6e8 3ffef6a0 14620beb 
[DEBUG] 3fffef30: 4025c65a 3fffdab0 00000000 3fffdcb0 
[DEBUG] 3fffef40: 3ffef6b8 3fffdad0 3fff0884 4023091b 
[DEBUG] 3fffef50: 40000f49 40000f49 3fffdab0 40000f49 
[DEBUG] 3fffef60: 40000e19 40001878 00000001 03f52cdc 
[DEBUG] 3fffef70: 8d603ffe 84c83ffe 31f33fff 00044022 
[DEBUG] <<<stack<<<
[154145] +OK
```

davebuk, 2020-01-13:
[build.zip](https://github.com/xoseperez/espurna/files/4053023/build.zip)
PlatformIO had deleted the folders so have re-built firmware. The .zip has 'basic_bmx.bin' which is the one currently loaded. The other two files are directly from the folders requested.

mcspr, 2020-01-13:
Ok. I had a similar module already set up with august's build which I two-step upgraded just now. Hopefully something breaks.

btw stack decoding is busted in current version :/ I need to rewrite the output to fit the serial output that the tool expects and simplify c/p.
https://github.com/me21/EspArduinoExceptionDecoder can decode the stuff between `>>>stack>>>` and `<<<stack<<<`, but will complain with the full output. but both timestamps and `[DEBUG] ` prefixes need to be removed manually.

davebuk, 2020-01-13:
Hopefully you'll see something within two hours.
Last `info`:

```
---8<-------

[065910] [MAIN] ESPURNA 1.14.1 (62ad7da3)
[065910] [MAIN] xose.perez@gmail.com
[065910] [MAIN] http://tinkerman.cat

[065910] [MAIN] CPU chip ID: 0x9785FC
[065910] [MAIN] CPU frequency: 80 MHz
[065911] [MAIN] SDK version: 2.2.2-dev(38a443e)
[065911] [MAIN] Core version: 2.6.3
[065915] [MAIN] Core revision: 3d128e5c
[065915] [MAIN] Build time: 1578576965
[065916] 
[065916] [MAIN] Flash chip ID: 0x144051
[065916] [MAIN] Flash speed: 40000000 Hz
[065916] [MAIN] Flash mode: DOUT
[065916] 
[065917] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[065919] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[065919] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[065920] [MAIN] Firmware size       :   544688 bytes /  133 sectors (   1 to  133)
[065920] [MAIN] Max OTA size        :   479232 bytes /  117 sectors ( 134 to  250)
[065920] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[065921] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[065921] 
[065921] [MAIN] EEPROM sectors: 251, 250
[065923] [MAIN] EEPROM current: 250
[065923] 
[065924] [MAIN] EEPROM:  4096 bytes initially |  1096 bytes used (26%) |  3000 bytes free (73%)
[065924] [MAIN] Heap  : 43368 bytes initially | 19240 bytes used (44%) | 24128 bytes free (55%)
[065925] [MAIN] Heap  : 23024 contiguous bytes available (5% fragmentation)
[065925] [MAIN] Stack :  4096 bytes initially |  1664 bytes used (40%) |  2432 bytes free (59%)
[065925] 
[065925] [MAIN] Boot version: 7
[065927] [MAIN] Boot mode: 1
[065928] [MAIN] Last reset reason: Software Watchdog
[065928] [MAIN] Last reset info: Fatal exception:4 flag:3 (SOFT_WDT) epc1:0x4025b9f2 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[065928] 
[065929] [MAIN] Board: ITEAD_SONOFF_BASIC_BMX280
[065929] [MAIN] Support: API BROKER BUTTON DEBUG_TELNET DEBUG_WEB I2C LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL WEB 
[065930] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[065931] [MAIN] Sensors: BMX280 
[065933] [MAIN] WebUI image: SENSOR
[065933] 
[065933] [MAIN] Firmware MD5: 604c5a60e8fdfb34a55721b3bf95b228
[065934] [MAIN] Power: 3387 mV
[065934] [MAIN] Power saving delay value: 10 ms
[065935] 

---8<-------
```
`crash`
```
[145692] [WEBSOCKET] Requested action: dbgcmd
[145695] [DEBUG] Latest crash was at 6327820 ms after boot
[145696] [DEBUG] Reason of restart: 3
[145696] [DEBUG] Exception cause: 4
[145696] [DEBUG] epc1=0x4025b9f2 epc2=0x00000000 epc3=0x00000000
[145697] [DEBUG] excvaddr=0x00000000 depc=0x00000000
[145697] sp=0x3fffeef0 end=0x3fffffb0 saved=0x0080

[145697] [DEBUG] >>>stack>>>
[DEBUG] 3fffeef0: 402573f0 3fff53bc 3fff470c 3fff3abc 
[DEBUG] 3fffef00: 00000000 3ffee6e8 00000000 00000000 
[DEBUG] 3fffef10: 4025b810 3ffee6e8 3ffef6a0 4023b580 
[DEBUG] 3fffef20: 4025d481 3fff470c 3fff53bc 402618b7 
[DEBUG] 3fffef30: 4025d4c6 3fffdab0 00000000 3fffdcb0 
[DEBUG] 3fffef40: 3ffef6c8 00000000 3fff53bc 4025d0c3 
[DEBUG] 3fffef50: 40000f49 3fffdab0 3fffdab0 40000f49 
[DEBUG] 3fffef60: 40000e19 40001878 00000004 3fffffb0 
[DEBUG] 3fffef70: 00000000 00000020 40100eb2 3ffeb025 
[DEBUG] <<<stack<<<
[145805] +OK
```

davebuk, 2020-01-13:
I have tried to look at the information you posted about stack decoding. I'll have to set some time aside to get my head around how to do it! I have found 'decoder.py' in: `C:\Users\<user name>\AppData\Local\Programs\Python\Python38-32\Lib\json` but not sure how to run it from within platformIO. `decoder.py -e .pioenvs\lingan_swa1\firmware.elf stack.txt` doesn't run from the platformIO terminal command line running from: `C:\Users\<user name>\Documents\PlatformIO\ESPurna dev\espurna\code>`.

Let me know if you want to try the stack decode and I'll see what I can get.

Cheers.

davebuk, 2020-01-13:
I have also just had a HARDWARE WATCHDOG reboot. No `crash` though.

```
---8<-------

[104323] [MAIN] ESPURNA 1.14.1 (62ad7da3)
[104323] [MAIN] xose.perez@gmail.com
[104323] [MAIN] http://tinkerman.cat

[104323] [MAIN] CPU chip ID: 0x9785FC
[104324] [MAIN] CPU frequency: 80 MHz
[104324] [MAIN] SDK version: 2.2.2-dev(38a443e)
[104325] [MAIN] Core version: 2.6.3
[104328] [MAIN] Core revision: 3d128e5c
[104328] [MAIN] Build time: 1578576965
[104328] 
[104329] [MAIN] Flash chip ID: 0x144051
[104329] [MAIN] Flash speed: 40000000 Hz
[104329] [MAIN] Flash mode: DOUT
[104329] 
[104330] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[104332] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[104332] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[104333] [MAIN] Firmware size       :   544688 bytes /  133 sectors (   1 to  133)
[104333] [MAIN] Max OTA size        :   479232 bytes /  117 sectors ( 134 to  250)
[104333] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[104333] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[104334] 
[104334] [MAIN] EEPROM sectors: 251, 250
[104336] [MAIN] EEPROM current: 250
[104336] 
[104337] [MAIN] EEPROM:  4096 bytes initially |  1096 bytes used (26%) |  3000 bytes free (73%)
[104337] [MAIN] Heap  : 43368 bytes initially | 19192 bytes used (44%) | 24176 bytes free (55%)
[104338] [MAIN] Heap  : 23048 contiguous bytes available (5% fragmentation)
[104338] [MAIN] Stack :  4096 bytes initially |  1440 bytes used (35%) |  2656 bytes free (64%)
[104338] 
[104339] [MAIN] Boot version: 7
[104341] [MAIN] Boot mode: 1
[104341] [MAIN] Last reset reason: Hardware Watchdog
[104342] [MAIN] Last reset info: Fatal exception:4 flag:1 (WDT) epc1:0x40104141 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[104342] 
[104342] [MAIN] Board: ITEAD_SONOFF_BASIC_BMX280
[104343] [MAIN] Support: API BROKER BUTTON DEBUG_TELNET DEBUG_WEB I2C LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL WEB 
[104343] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[104343] [MAIN] Sensors: BMX280 
[104345] [MAIN] WebUI image: SENSOR
[104345] 
[104346] [MAIN] Firmware MD5: 604c5a60e8fdfb34a55721b3bf95b228
[104346] [MAIN] Power: 3386 mV
[104347] [MAIN] Power saving delay value: 10 ms
[104347] 

---8<-------

[104347] +OK
[118942] [WEBSOCKET] Requested action: dbgcmd
[118951] [DEBUG] No crash info
[118951] +OK
```

mcspr, 2020-01-13:
HW WDT can be pretty much anything since it is like a cold boot situation, no stack trace is preserved from when the exception happened.
I also wonder if `erase.config` command can help? It will erase WiFi-related SDK settings and reboot after that.

davebuk, 2020-01-13:
I guess 'FACTORY RESET' does more than `erase.config`? I did that last night on the SWA1 switch and completely reset all settings. It didn't help.

mcspr, 2020-01-13:
`erase.config` only erases data related to the WiFi firmware
`factory.reset` erases espurna settings and it does not include `erase.config`

davebuk, 2020-01-13:
I've run it. Should the configured WiFi be removed and the device revert to AP mode?

davebuk, 2020-01-13:
With the SWA1 device, I've reverted back to my previous firmware. It's been stable now for going on 3 hours. The sonoff device has reset more than once.

mcspr, 2020-01-13:
No crashes for far, I'd start to think about blaming the specific AP that you started using, but...
Core versions switched SDK versions between 2.5.2 and 2.6.3, have you tried any of these options to compare with? This needs to be added to `build_flags = ...`:
`-DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK221` (i think this was included in 2.5.2 as default)
`-DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191024`
`-DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191105`
`-DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191122`
Default for 2.6.3 is `-DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703`
(src: https://github.com/esp8266/Arduino/blob/0d38ea7308fff8bb06c7bd2a4e1e016a96a2eee8/tools/platformio-build.py#L159-L198)

davebuk, 2020-01-13:
The AP has been in use for quite a while and espurna didn't have issues with it on previous builds. I'll try a different AP first and see what that does and look to build some SDK variants when time allows.

davebuk, 2020-01-14:
I left the SWA1 on `1.13.6dev core 2.5.2` and Sonoff_BMX on `1.14.1 core 2.6.3` and turned off the AP local to those devices. The AP is a D-LINK DIR-615. This meant the devices were connecting to my other AP (BT HOME HUB 4 which is just being used as an access point not a router). They had a very low signal, around -89dBm but were connected and working. Both devices stayed connected all night and no reboots.

I then replaced the DIR-615 with another HH4 in AP mode locally. Both devices stayed connected for at least 3 hours without issue. They were defiantly connected to this AP as their strength was now around -54dBm. 

I reinstated the DIR-615, ran the SWA1 on `1.14.1 core 2.6.3` and the Sonoff_BMX on `1.14.1 core 2.6.3 and SDK 2.2.1(cfd48f3)`. The SWA1 is back to rebooting (SOFT_WDT) after just under 2 hours, where the Sonoff_BMX is now stable.

For whatever reason, it looks like the newer SDK `2.2.2-dev(38a443e)` is having some issues with the DIR-615 AP but not the BT HOME HUB 4. The DIR-615 was setup in 'N-ONLY' mode where as the HH4 (from what I remember are in 'b/g/n' mode). I'll look to test with the DIR-615 in 'b/g/n' mode first before I look to replacing it with the HH4. 

Are there any other tests you would like me to try before reverting back to current latest versions of the SDK?

mcspr, 2020-01-14:
Probably, yes. Any identifiable markings on DIR-615? (...or internals photo with chip names?) 
https://wikidevi.wi-cat.ru/Special:Search?search=dir-615&go=Try+exact+match
https://openwrt.org/toh/d-link/dir-615 
shows a lot of revisions with different chips

At this point you can try out later revisions, 191105 and 191122? Just to note about SDK 3.0.0. - has incompatible flash layout with older versions, so downgrading after it will very likely break wifi connectivity completely. `erase.config` *may* fix things though, after the fact.
I wonder if @flucasro issue from #2064 is connected to SDK version too.

No other configuration test change come to mind.

davebuk, 2020-01-15:
Sorry, when I said latest version of SDK, I meant the one set by default for the 2.6.3 core (2.2.2-dev(38a443e)).

I have tried 191105 and 191122 and both are having the reboots with the DIR-615. 191122 seemed to reboot sooner for some reason.

WRT #2064, I installed the HH4 locally leaving the SWA1 on 1.13.6 dev and Sonoff_BMX on 1.14.1 SDK191122. The SWA1 connected to a lower signal AP and over an hour later is still connected with a -87dBm signal, even though an AP with the same SSID/password is now available. Pressing 'Scan Now' in the WiFi UI shows the 3 APs I have, and their signal levels, but the device stays on the one it was connected to before the stronger signal was available. Pressing the `Reconnect` button on the WebUI makes the device reconnect and it uses the strongest signal available (-56dBm). The Sonoff didn't connect at all and I needed to do 2 hard power cycles to get it to connect. When it did, it went straight to the strongest AP.

davebuk, 2020-01-15:
The DIR-615 is H/W: D4 P/N: . Although its a router, I was running it in `AP mode` so it was just a WiFi AP bridge to my main router.

mcspr, 2020-01-19:
Re: "WRT #2064"
As I also mentioned in the other issue, periodic scan is actually off by default in the #2088 patch. Sry about that, should've probably turned it on to avoid confusion. Unless it is on and not working 🤔 

I tried some different wireless hw and still not seeing any crashes. Only bug so far with sort-of recent Core versions is memory exhaustion from UDP, but I think that is solved with https://github.com/esp8266/Arduino/commit/3e6c25c21b51d9b1a868600b2fc8f4359f6f40a8 (2.6.3+)

davebuk, 2020-01-19:
Both devices running 2.6.3 core (2.2.2-dev(38a443e)) have been up now for 4 days with the BT HH4 acting as WiFi access point instead of the DIR-615. I'll close this issue as it seems the issue I had was related to external hardware.

Thanks for your help.

closed 2020-01-19 by davebuk