xoseperez/espurna #2112: mqtt rever reconnects

flucasro, 2020-01-16:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
*A clear and concise description of what the bug is.*

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by typing `info` via serial, terminal or in the debug tab in the web UI. The relevant information is that surrounded by the scissors-cut lines (`---8<-------`).*
*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*
* *Flash mode*
* *Device brand, model and version*

**Tools used**
* *Desktop operating system*
* *Browser & version*
* *IDE & version*
* *Compiler & version (if not embedded in IDE)*

**Additional context**
*Add any other context about the problem here.*

mcspr, 2020-01-17:
See https://github.com/marvinroger/async-mqtt-client/pull/156
If remote drops off the network for some reason, MQTT client will reconnect only after 15 minutes (internal TCP retry timeout)

I'd ping the PR there first, then we can think of adding it locally as forked version

flucasro, 2020-01-17:
Please see the log message
[739066] [MQTT] Retry CONNECTING (Now 513739066, Last 33600056, Delay 5000, Step 5000)

the "now", and "last" values... I think that there is something more to the 15min and the half open connections

mcspr, 2020-01-18:
But when does this happen? How to reproduce the issue?
Did broker disappear from the network temporarily / abruptly shutdown / IP address changed?

Log only says that neither connection or disconnection routine ever completed, not the reason why it is waiting. Looking at async-mqtt-client again, it never sets data-receive timeout for connection loop. If I load netcat listener on port 1883, it will never complete connection routine. see `netstat` command. Only after shutting down the 1883 listener it will cleanly break the connection, restoring the loop.

To avoid doing that: mcspr/async-mqtt-client@c1fcfd1
This will impose 10 second window to finish the connection after CONNECT packet is sent, since ping keepalive does not yet work.
Value is hard-coded tough, maybe it can be the same as keepalive?

I also noticed marvinroger/async-mqtt-client@7f1ba481a22d56ccf123e4b2f6e555d134c956d0 in the latest version, but we don't call those methods from mqtt callbacks.

closed 2020-03-26 by stale[bot]