xoseperez/espurna #2120: New type of airlux board?

shohart, 2020-01-22:
Hello, i am trying to get my strip controller to work, but it seems that it is a new board and colors keep messing around. so here is the board. It is the one with new rf remote module and it is labeled 
ZJ-WFMN-B v 1.1
[image]

as a result it can not reproduce green or yellow or orange color. only red and blue and puurple obviously.

Thanx a lot for a work, buds.

*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by typing `info` via serial, terminal or in the debug tab in the web UI. The relevant information is that surrounded by the scissors-cut lines (`---8<-------`).*
*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*
* *Flash mode*
* *Device brand, model and version*

**Tools used**
* *Desktop operating system*
* *Browser & version*
* *IDE & version*
* *Compiler & version (if not embedded in IDE)*

**Additional context**
*Add any other context about the problem here.*

mcspr, 2020-01-24:
Have you tried using *magichome* configuration instead?
https://github.com/xoseperez/espurna/releases/download/1.14.1/espurna-1.14.1-magichome-zj-wfmn-b-11.bin

shohart, 2020-01-24:
> Have you tried using _magichome_ configuration instead?
> https://github.com/xoseperez/espurna/releases/download/1.14.1/espurna-1.14.1-magichome-zj-wfmn-b-11.bin

oh... so dumb of me! anyways it works muck better! i do have all colors now, but red and green channels a messed up! should i just switch the wires or is there a config option?

mcspr, 2020-01-26:
GPIO configuration is hard-coded, so swapping wires is the way to go atm. Or if you want to wait for 1.14.2 there would be runtime config

shohart, 2020-01-27:
oh i see, so i just swapped wires then - the simpliest way - is the way! Cheers folks, thanks!

closed 2020-01-27 by shohart