xoseperez/espurna #2121: Faster Power-On for lights and relays

MaximeLussier, 2020-01-24:
**Is your feature request related to a problem? Please describe.**
On power-on, it takes up to a second to turn on the light/relay based on power-on configuration.
(e.g always-on, toggle, or last state)

**Describe the solution you'd like**
As soon as possible, before wifi and web/ota setup, check power-on configuration and apply it.

**Describe alternatives you've considered**
Could be possible to do a partial setup then complete once the rest has setup

**Additional context**
Reaction time to seem "instant" should remain under 300ms which is hard to achieve when waiting on wifi...

mcspr, 2020-01-24:
See https://github.com/xoseperez/espurna/pull/2116 (+ subsequent https://github.com/xoseperez/espurna/commit/2e891e3129133ea5025547d730682c1c4c162c26)
This was discussed out of the issue tracker - https://gitter.im/tinkerman-cat/espurna?at=5e206b6d0dbd9379ed470747

Main culprits:
- Arduino Core 2.3.0 only starts running initialization code at around 243msec after power on. With default build it takes ~50-100msec to call all global C++ object constructors. Then, setup() & loop() are called at around ~300-350msec timestamp.
- From our side, `info()` function used to have a lot of `yield()` slowing things down.
- In addition to that `info()` calculated MD5 of flash contents taking ~250msec

`info()` banner can be disabled completely via `DEBUG_LOG_MODE DebugLogMode::SkipBoot` setting (or `dbgLogMode => "2"` runtime). Switching to latest Core improves startup time by a lot, you should be able to see that basic relay provider already sets pin mode at 163ms after power on.

MaximeLussier, 2020-01-24:
Thanks, I will try with the latest Core and with DEBUG_LOG_MODE set to SkipBoot.

Any idea when the new core will be included in an espurna release?

mcspr, 2020-01-26:
Likely the one after the next one.