xoseperez/espurna #2122: MQTT Commands not work randomly

andreaconfa, 2020-01-25:
**Question**
Hi all, 
i have all my home running Shelly with Espurna and it was working great until this days.
The problem that I am experiencing is that randomly some devices do not respond to the command via mqtt until they are restarted. If I operate the relay directly from the web page, the state topic is correctly sent by espurna. MQTT is always connected even when the problem occurs and the wifi have strong signal.

**Device information**
Manufacture rALLTERCO
Device SHELLY1
Chip ID 2C8A20
Wifi MAC 3C:71:BF:2C:8A:20
SDK version1.5.3(aec24ac9)
Core version 2.3.0
Firmware name ESPURNA
Firmware version 1.13.3
Firmware revisione 2641b9
Firmware build date2018-10-08 17:54:44 
Firmware size 463200 
bytesFree space 581632 bytes

mcspr, 2020-01-26:
> Firmware version 1.13.3

Check with 1.14.1?

LSC-Labs, 2020-03-15:
Same problem...
i upgraded to the actual version... no change...
but all of my devices cannot sync with the ntp server !
so i assume it is an error in time sync so mqtt messages aren't accepted cause out of valid time window
Changing time server always throws the error "time server not reachable"

LSC-Labs, 2020-03-15:
Message for ntp is
ntp
[887967] [NTP] UTC Time  : 2036-02-07 06:28:41
[887968] [NTP] Local Time: 2036-02-07 07:28:41
[887968] +OK
ntp.sync
[944980] +OK
[952911] [NTP] Error: NTP server not reachable

closed 2020-07-25 by stale[bot]

DannyDeGaspari, 2020-10-19:
I'm having the same issue as the topic starter. I'm running v 1.14.1 on a Sonoff-S26. No MQTT response after a few days (1 - 3 days). The problem is happening more often now then a while ago, the system is running for a year now and no problems in the first months. 
Via the web interface, the relay can still be toggled. The device needs to be rebooted before MQTT is responsive again.

I also have a Sonoff-Mini, also has the same problem but less frequent. It can run for a few weeks before reboot is needed.

mcspr, 2020-10-21:
@DannyDeGaspari 

what do logs on the device look like? does it still try to send data when you change things from web?
can you try current development builds?
have you changed something in the network recently? i may be mistaken, but we *may* have issues when wifi network itself is unstable (which also may not be apparent at first glance). including silent loss of connectivity and / or causing mqtt client to hang itself

DannyDeGaspari, 2020-10-22:
When the device is not MQTT responsive anymore, nothing appears in the logs about MQTT, also not when the state is changed from web. If I don't reboot the device then most probably the web interface stops responding also after lets say 1 or 2 days.
Nothing changed in my network and I have another device (Sonoff mini) which keeps on working on the moment the Sonoff-S26 stops working. But also the mini can stop responding, but less frequent.
Can the MQTT client be restarted from the debug log ? I will also try the development build.

DannyDeGaspari, 2021-01-31:
I'm again in a non-responsive MQTT situation, with working web interface. This is the log:
[480012] [WEBSOCKET] #5 connected, ip: 192.168.0.153, url: /ws
[481881] [WEBSOCKET] Requested action: relay
[481882] [RELAY] #0 scheduled ON in 0 ms
[481884] [RELAY] #0 set to ON
[482886] [RELAY] Setting relay mask: 0b1
[500314] [WEBSOCKET] Requested action: relay
[500315] [RELAY] #0 scheduled OFF in 0 ms
[500325] [RELAY] #0 set to OFF
[501327] [RELAY] Setting relay mask: 0b0
[599544] [MAIN] Uptime: 4417566 seconds
[599545] [MAIN] Heap  : 34888 bytes initially | 15480 bytes used (44%) | 19408 bytes free (55%)
[599546] [MAIN] Power: 3238 mV
[599548] [MAIN] Time: 2021-01-31 11:07:36

Are there any test I can run or commands to enter to try to debug what is going on before I reboot the device ? If I reboot the device, the problem will be resolved again.

mcspr, 2021-02-02:
@DannyDeGaspari 
What's the MQTT keepalive setting on the device?
You might also want to try recent nightly builds, as 1.14.1 might have general WiFi connectivity problems with newer APs, causing these weird issues

DannyDeGaspari, 2021-02-02:
I'm using the default value of 300 for MQTT keepalive.

mcspr, 2021-02-02:
@DannyDeGaspari reducing it to 30 should also help. btw what is the heartbeat interval? this should be on the bottom of the ADMIN page

DannyDeGaspari, 2021-02-02:
Heartbeat interval is 300. 
I will try a lower keepalive value and see if it helps.