xoseperez/espurna #2123: Multiple devices from same config

orrpan, 2020-01-25:
I've tried the deltaco sh-p02, which uses the same configuration as Digoo NX SP202 (had to fix the calibration only). So it would be unnecessary to have a completely new setup for the device, but if I want correct `Manufacturer` and `Device` I need to change something. 

Like this

## platformio.ino
### before
```
[env:digoo-nx-sp202]
board = ${common.board_1m}
build_flags = ${common.build_flags_1m0m} -DDIGOO_NX_SP202

[env:digoo-nx-sp202-ota]
board = ${common.board_1m}
build_flags = ${common.build_flags_1m0m} -DDIGOO_NX_SP202
upload_port = ${common.ota_upload_port}
upload_flags = ${common.ota_upload_flags}

```
### after
```
[env:digoo-nx-sp202]
board = ${common.board_1m}
build_flags = ${common.build_flags_1m0m} -DDIGOO_NX_SP202

[env:digoo-nx-sp202-ota]
board = ${common.board_1m}
build_flags = ${common.build_flags_1m0m} -DDIGOO_NX_SP202
upload_port = ${common.ota_upload_port}
upload_flags = ${common.ota_upload_flags}

[env:deltaco-sh-p02]
board = ${common.board_1m}
build_flags = ${common.build_flags_1m0m} -DDELTACO_SH_P02

[env:deltaco-sh-p02-ota]
board = ${common.board_1m}
build_flags = ${common.build_flags_1m0m} -DDELTACO_SH_P02
upload_port = ${common.ota_upload_port}
upload_flags = ${common.ota_upload_flags}
```

## hardware.h
### before
```
#elif defined(DIGOO_NX_SP202)

    // Info
    #define MANUFACTURER                "DIGOO"
    #define DEVICE                      "NX_SP202"

    // Buttons
    #define BUTTON1_PIN                 0
```

### after
```
#elif (defined(DIGOO_NX_SP202) | defined(DELTACO_SH_P02))

    // Info
    #if defined(DELTACO_SH_P02)
        #define MANUFACTURER                "DELTACO"
        #define DEVICE                      "SH_P02"
    #else
        #define MANUFACTURER                "DIGOO"
        #define DEVICE                      "NX_SP202"
    #endif

    // Buttons
    #define BUTTON1_PIN                 0
```

## migrate.ino
no changes needed?

## arduino.h
no changes needed?

Or is this just unnecessary power to compile and store?

mcspr, 2020-01-26:
I'd rather not do `#elif (defined(DIGOO_NX_SP202) || defined(DELTACO_SH_P02))` here, comment would suffice + entry at https://github.com/xoseperez/espurna/wiki/Hardware

orrpan, 2020-01-26:
Yessir!

closed 2020-01-26 by orrpan