xoseperez/espurna #2125: Sonoff RF bridge can read remote control with 434.42 MHZ?

connected201, 2020-01-26:
hi, i have remote control like this:
[image]
[image]
i cannot read it! direct mode can resolve this? 
best regards.

mcspr, 2020-01-26:
Have you tried https://github.com/Portisch/RF-Bridge-EFM8BB1?

connected201, 2020-01-26:
> 
> 
> Have you tried https://github.com/Portisch/RF-Bridge-EFM8BB1?

Hi, your firmware will be read 434.42 MHZ?
if yes, i will flash it, i read datasheet for EFM8BB1, and i don't saw nothing about chip working frequency ( maybe because this chip is universal?)
my question is, if RF bridge can work for this range too?

mcspr, 2020-01-26:
It is not specifically about the frequency, but the protocol. Only recognisable one is 3-byte PT2262 (or sort-of compatible chips like EV1527).

EFM8BB1 refers to the secondary MCU on rfbridge, it handles rf decoding / encoding and communicates with esp8266 through UART.

trafoman2004, 2020-01-28:
sonoff rf bridge frequency is 433,92 so, no deal.

closed 2020-04-04 by stale[bot]