xoseperez/espurna #2126: Websocket Auth

Jip-Hop, 2020-01-26:
How can I authenticate via Websockets to toggle the relays?

I'm trying to control the relays on my device directly via a Websocket connection. When I run this example code in the console of a new Google Chrome tab, it toggles the relay without issues:

`webSocket = new WebSocket("ws://192.168.4.1/ws"); webSocket.addEventListener('open', (event) => { webSocket.send("{'action':'relay','data':{'id':0,'status':2}}") });`

But it only works after I have already logged into the Espurna web gui. When I didn't login and try to connect over Websockets, I get the following output in Chrome Devtools:

```
Out: {'action':'relay','data':{'id':0,'status':2}}
In: {"message": 10}
Close received after close
```

My goal is to toggle the relays from an ESP32. The ESP32 will receive messages via Bluetooth Low Energy and then toggle the relays on the Espurna device. The ESP32 is directly connected to the Espurna WiFi Access Point. But also on the ESP32 the Websocket connections closes immediately.

```
Connected to WiFi!
Connnection Opened
Got Message: {"message": 10}
Connnection Closed
```

So how can I authenticate from the ESP32 to be allowed to connect over Websockets? I tried with and without `Enable WS Auth` but couldn't get it working. It's not clear to me what `Enable WS Auth` is supposed to do and if it's related to this use case.

**Device information**

[139531] [MAIN] ESPURNA 1.14.1 (62ad7da3)
[139531] [MAIN] xose.perez@gmail.com
[139531] [MAIN] http://tinkerman.cat

[139532] [MAIN] CPU chip ID: 0xD164F1
[139532] [MAIN] CPU frequency: 80 MHz
[139532] [MAIN] SDK version: 1.5.3(aec24ac9)
[139532] [MAIN] Core version: 2.3.0
[139534] [MAIN] Core revision: 9826c6d
[139535] [MAIN] Build time: 1577763075
[139535] 
[139535] [MAIN] Flash chip ID: 0x144051
[139538] [MAIN] Flash speed: 40000000 Hz
[139538] [MAIN] Flash mode: DOUT
[139538] 
[139538] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[139542] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[139542] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[139542] [MAIN] Firmware size       :   484032 bytes /  119 sectors (   1 to  119)
[139543] [MAIN] Max OTA size        :   536576 bytes /  131 sectors ( 120 to  250)
[139543] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[139543] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[139543] 
[139543] [MAIN] EEPROM sectors: 251, 250
[139549] [MAIN] EEPROM current: 251
[139550] 
[139551] [MAIN] EEPROM:  4096 bytes initially |   335 bytes used ( 8%) |  3761 bytes free (91%)
[139551] [MAIN] Heap  : 34888 bytes initially | 17328 bytes used (49%) | 17560 bytes free (50%)
[139551] [MAIN] Stack :  4096 bytes initially |  1424 bytes used (34%) |  2672 bytes free (65%)
[139551] 
[139552] [MAIN] Boot version: 31
[139552] [MAIN] Boot mode: 1
[139554] [MAIN] Last reset reason: Power on
[139555] [MAIN] Last reset info: flag: 0
[139555] 
[139555] [MAIN] Board: ZHILDE_EU44_W
[139557] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[139557] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[139557] [MAIN] WebUI image: SMALL
[139557] 
[139560] [MAIN] Firmware MD5: 8d53779be58faeb803a8f0bf91bbc743
[139560] [MAIN] Power: 3217 mV
[139561] [MAIN] Power saving delay value: 10 ms
[139561] 

**Tools used**
MacOS 10.14.6 Mojave
Google Chrome Version 79.0.3945.130
ESP32, Arduino IDE 1.8.10

mcspr, 2020-01-26:
You need to access `/auth` with HTTP auth credentials, session is kept after that for some time. Check out the WebUI loader:
https://github.com/xoseperez/espurna/blob/cbd7064c5c99622204c04203de7e2e818871011c/code/html/custom.js#L2120-L2142

Jip-Hop, 2020-01-26:
Thanks for the reply!

I got it working now using Node.js. This example searches for all Espurna devices on the network, authenticates and on successful auth opens a Websocket connection to toggle all relays every 5 seconds.

```
const bonjour = require("bonjour-hap")();
const request = require("request");
const WebSocket = require("ws");

const username = "admin",
  password = "fibonacci",
  auth = "Basic " + Buffer.from(username + ":" + password).toString("base64");

function getJson(str) {
  try {
    return JSON.parse(str);
  } catch (e) {
    return false;
  }
}

function processService(service) {
  console.log(service);
  request(
    {
      url: `http://${service.addresses}/auth`,
      headers: {
        Authorization: auth
      }
    },
    function(error, response, body) {
      if (!error && response.statusCode == 200) {
        console.log(body);

        var nrOfRelays = 0;
        const ws = new WebSocket(`ws://${service.addresses}/ws`);

        ws.on("open", function open() {
          setInterval(function() {
            for (let i = 0; i < nrOfRelays; i++) {
              ws.send(`{'action':'relay','data':{'id':${i},'status':2}}`);
            }
          }, 5000);
        });

        ws.on("message", function incoming(data) {
          var data = getJson(
            data
              .replace(/\n/g, "\\n")
              .replace(/\r/g, "\\r")
              .replace(/\t/g, "\\t")
          );
          if (data) {
            Object.keys(data).forEach(function(key) {
              var value = data[key];
              console.log(value);
              // ---------------------------------------------------------------------
              // Relays
              // ---------------------------------------------------------------------

              if ("relayState" === key) {
                console.log(`New relays status: ${value.status}.`);
                return;
              }

              // Relay configuration
              if ("relayConfig" === key) {
                nrOfRelays = value.size;
                console.log(`Found ${value.size} relays.`);
                return;
              }
            });
          }
        });
      } else {
        console.log(error);
        console.log("Error");
      }
    }
  );
}

// find all Espurna devices
bonjour.find(
  {
    type: "arduino",
    txt: {
      app_name: "ESPURNA"
    }
  },
  processService
);
```

closed 2020-04-02 by stale[bot]