xoseperez/espurna #2127: Sunset and sunrise

luis-sanmiguel, 2020-01-27:
Hola,

No sé si puedo utilizar el castellano, probaré primero y si no recibo respuesta, quizás puede que lo intente en inglés.

Siempre he creído que sería interesante añadir una funcionalidad que permitiera tener en cuenta las horas del amanecer y atardecer para retrasar o adelantar determinadas acciones. Por ejemplo, en vez de encender una luz siempre a la misma hora, que en invierno puede ser muy tarde y en verano puede ser muy pronto, poder tener en cuenta como digo las horas del amanecer y del anochecer. De esta forma podría programar la luz para que se encendiera 1h después del atardecer y 1 h antes del amanecer.

Ahora mismo lo que hago es consultar estos datos vía la API de sunrise-sunset.org y en base a ellos planifico las tareas y acciones que quiero.

Bueno, pues ahí queda eso. Ya me dirás algo ... o no si es no es el idioma correcto

mcspr, 2020-01-29:
See https://github.com/xoseperez/espurna/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopen+sunrise 
There was an experimental sensor to send such data. It needs some cleanup though.
https://github.com/poulch74/espurna/blob/espurna-poulch74/code/espurna/sunrise.ino
https://github.com/poulch74/espurna/blob/espurna-poulch74/code/espurna/sensors/SunriseSensor.h
https://github.com/poulch74/espurna/blob/espurna-poulch74/code/espurna/libs/Sunrise.h

Scheduling part can be achieved through rpn?
https://github.com/xoseperez/espurna/wiki/RPN-Rules#rule-execution
Sensor code above hijacks analog sensor as magnitude, but that can be reworked. For example, sensor will return 0 for night and 1 for day.

Marking as duplicate of #1263

closed 2020-01-29 by mcspr

luis-sanmiguel, 2020-02-05:
Thank you sir.