xoseperez/espurna #2133: Allow to use latest ESPSoftwareSerial

mcspr, 2020-02-03:
**Is your feature request related to a problem? Please describe.**
Current PlatformIO references fairly old ESPSoftwareSerial
https://github.com/xoseperez/espurna/blob/cbd7064c5c99622204c04203de7e2e818871011c/code/platformio.ini#L109
While the latest version is included in the Core package. Main issue is that it has a different API, which slightly complicates things for anyone building with Arduino IDE

**Describe the solution you'd like**
String Serial management from sensor code, provide common way to receive either HardwareSerial or SoftwareSerial as `Stream`.
Using latest version should also improve performance with 115200 speed (if someone really needs it)

**Describe alternatives you've considered**
Use latest Core and only the latest SoftwareSerial API. One problem is that in case SWserial API changes again, all sensors need to match it

**Additional context**
re @irmishappy , building pow-r2 with Arduino IDE fails with:
```
sketch\config/../sensors/CSE7766Sensor.h: In member function 'virtual void CSE7766Sensor::begin()':
sketch\config/../sensors/CSE7766Sensor.h:123:90: error: no matching function for call to 'SoftwareSerial::SoftwareSerial(unsigned int&, const int&, bool&, int)'
             _serial = new SoftwareSerial(_pin_rx, SW_SERIAL_UNUSED_PIN, _inverted, 32);
                                                                                      ^
```

estebanz01, 2020-06-08:
I'm seeing this same issue with a sonoff S31.

```c++
n file included from /var/folders/rm/pn5y_6zj3dl0k195ggkvf4kw0000gn/T/arduino_build_945685/sketch/sensor.cpp:63:0:
/var/folders/rm/pn5y_6zj3dl0k195ggkvf4kw0000gn/T/arduino_build_945685/sketch/sensors/CSE7766Sensor.h: In member function 'virtual void CSE7766Sensor::begin()':
/var/folders/rm/pn5y_6zj3dl0k195ggkvf4kw0000gn/T/arduino_build_945685/sketch/sensors/CSE7766Sensor.h:138:55: error: 'SW_SERIAL_UNUSED_PIN' was not declared in this scope
                 _serial = new SoftwareSerial(_pin_rx, SW_SERIAL_UNUSED_PIN, _inverted, 32);
                                                       ^
```

mcspr, 2020-06-08:
https://github.com/plerup/espsoftwareserial/archive/3.4.1.zip is the library used in PlatformIO builds

One way this could work is by replacing `<CORE_INSTALLATION_DIRECTORY>/libraries/ESPSoftwareSerial/*` with contents of the zip, since SoftwareSerial is bundled with the esp8266 package

closed 2021-10-05 by mcspr