xoseperez/espurna #2139: Request: Web Interface: Switch on for xx

BuildTheRobots, 2020-02-05:
**Is your feature request related to a problem? Please describe.**
I use espurna to control my electric heaters and hot water immersion heater. Because of the built in scheduling I do not use any smart-home solution other than having the switches connected to my wifi. 
Quite often I would want to "top up" a heater or the water; to have it on for (eg) 15min, 30min or maybe an hour, but switching it on and remembering to turn it off manually can cause problems.

**Describe the solution you'd like**
Ideally something on the webpage which would let me toggle the relay on, but that used the pulse function to allow me to set a custom timeout. It would be great if the status page could report the relay as being not just "on" but "on for xxx minutes" or "on until <time> (timer|schedule)".

**Describe alternatives you've considered**
Switching on and trying to remember to switch it off manually - usually I forget.
Using the debug menu or curl to use the pulse function - as this is sporadic it's easy to forget the format of the command.

**Other**
Still thrillingly impressed with ESPurna and it's been doing me well over the last couple of years. The ability to have smart switches without any external dependencies needed is absolutely fantastic and makes it easy to experiment with smart home solutions without waking up to cold water.

mcspr, 2020-02-11:
> Because of the built in scheduling I do not use any smart-home solution other than having the switches connected to my wifi.
Quite often I would want to "top up" a heater or the water; to have it on for (eg) 15min, 30min or maybe an hour, but switching it on and remembering to turn it off manually can cause problems.

> Ideally something on the webpage which would let me toggle the relay on, but that used the pulse function to allow me to set a custom timeout.

Any ideas how that would look? (and in which part of the UI? b/c I am not really sure where that would even fit). What this sounds like to me at least is some kind of customisation widget in the ui, not the way it is done right now by hard-coding elements from our side. Debug panel already solves this via custom command, as you noted, which can be improved by adding "help" message like `pulse <relay-id> <seconds>` without any arguments?

External system will quite easily solve this, because you would have that custom dashboard with any action attached e.g. it would have switches that say 15m, 30m, 1h and send us pulse api command or off.

> It would be great if the status page could report the relay as being not just "on" but "on for xxx minutes" or "on until (timer|schedule)".

Related to the #1368. "on for xxx" via pulse can be done directly, *if* relay status change will know what part of the code turned it on or off. "on until schedule" or "timer" though... scheduler has debug log notifications about how much time there is until the next execution, so perhaps that part could also store some info for the ui. Or, the ui itself calculate the offset.

Pulse timer does not have any API regarding expiration time, only that it is either on or off. We do also have another proposal regarding timer limits #678, which also needs to be kept in mind when doing that.

goatchurchprime, 2020-02-14:
I think we want the features of this simple plug: https://www.belkin.com/us/p/P-F7C009/

[image]

You could put it into the schedule page -- a timeout a certain time after it has been switched on with the button.  

The user interface could be: 

1) Program in the timeout of 1hr through the webpage.
2) Press the button when off, and the power comes with the count-down timer starting at 1hr
3) Press the button when on (and counting) and it stays on and sets the count-down back to 1hr
4) Press and hold the button when on, and it turns off.
5) After the count-down has complete, turn off.  

This feature is simple because it's it's for purely off-line operation.  We don't need any complicated things such as reporting the count-down register to MQTT.  If there is an external system that is managing the sonoffs remotely, then it can already implement this turning off itself.

goatchurchprime, 2020-02-14:
An example of the timeout feature implemented in node-RED is included
[sonoff_timeout_flow.zip](https://github.com/xoseperez/espurna/files/4206459/sonoff_timeout_flow.zip)

[image]

It records the timestamp of the signal as it goes into the delay node so that later relay/on signals can suppress it.

mcspr, 2020-02-15:
> An example of the timeout feature implemented in node-RED is included
> [sonoff_timeout_flow.zip](https://github.com/xoseperez/espurna/files/4206459/sonoff_timeout_flow.zip)
> 
> [image]
> 
> It records the timestamp of the signal as it goes into the delay node so that later relay/on signals can suppress it.

Sorry if am stating the obvious, but we can already send `5` to the `ESPURNA_3B6F59/pulse/0` and it does almost the same as described in the previous comment? Device will invert it's current status and start the timer, which we can send 'refresh' by sending current status to `relay/0` (as sending 5 again will invert the status first and then set the timer).

Then, we can have a pre-set `relayTime0 => 5` so we can simply turn on or turn off with timer, depending on `relayPulse0` setting. These are 'Pulse' options in the Switches section of the WebUI.

mcspr, 2022-01-15:
For both #678 and this one
https://github.com/xoseperez/espurna/blob/932279f79fbbd69d9cca7d59e729c959204e3c38/CHANGELOG.md?plain=1#L149-L151
```markdown
- Remove internal pulse timer limit of 1 hour and 14 minutes, current limit should be around 47 days.
- Support extended time string with a unit suffix 'h' for hours, 'm' for minutes and 's' for seconds; such as, '5h' for 5 hours, '3m' for 3 minutes and '15s' for 15 seconds. For example, '1h35m'. Note that units are interpretted from the largest to the smallest, time string such as '1h2h' will be rejected and treated as 0 seconds instead of 1 hours plus 2 hours.
- Add `pulse <id> <time>` terminal command which accepts the same time string format or a floating 
point number of seconds (as in previous versions).
```

closed 2022-01-15 by mcspr