xoseperez/espurna #2150: web page doesn't save IP addresses

rmcbc, 2020-02-14:
With ESPURNA 1.14.2-dev the web page doesn't save the IP addresses.

I change the IP address save it, then it restarts and is back in the original.

mcspr, 2020-02-15:
Please follow the issue template and include more info than "web page doesn't save", maybe some screenshots?

rmcbc, 2020-02-15:
[image]
[image]
[image]
[image]
[image]
[image]

rmcbc, 2020-02-15:
Above you can see the sequence.
Thanks.

mcspr, 2020-02-15:
What does `get ip0` show in the Debug?

rmcbc, 2020-02-15:
[image]

rmcbc, 2020-02-15:
This was after upgrading from 1.14.1 to 1.14.2-dev.

mcspr, 2020-02-15:
I see, thanks.
What I am thinking now is when on Debug:
- `set ip0 192.168.1.9`
- `get ip0` will show `192.168.1.9`
- `reload`
- `get ip0` will show nothing
I missed settings eraser checking for hardcoded settings

rmcbc, 2020-02-15:
This is my custom.h file:

`// -----------------------------------------------------------------------------
// -----------------------------------------------------------------------------
//                                 defaults.h
// -----------------------------------------------------------------------------
// -----------------------------------------------------------------------------

// -----------------------------------------------------------------------------
// General
// -----------------------------------------------------------------------------

// Default hostname will be ESPURNA-XXXXXX, where XXXXXX is last 3 octets of chipID
#define HOSTNAME                "SonoffBasic09"

// -----------------------------------------------------------------------------
// -----------------------------------------------------------------------------
//                                  general.h
// -----------------------------------------------------------------------------
// -----------------------------------------------------------------------------

//------------------------------------------------------------------------------
// HEARTBEAT
//------------------------------------------------------------------------------

#define HEARTBEAT_INTERVAL          900000    // 15 min. // Interval between heartbeat messages (in ms)

// Values that will be reported in heartbeat
#define HEARTBEAT_REPORT_STATUS     1
#define HEARTBEAT_REPORT_SSID       0
#define HEARTBEAT_REPORT_IP         0
#define HEARTBEAT_REPORT_MAC        0
#define HEARTBEAT_REPORT_RSSI       0
#define HEARTBEAT_REPORT_UPTIME     0
#define HEARTBEAT_REPORT_DATETIME   0
#define HEARTBEAT_REPORT_FREEHEAP   0
#define HEARTBEAT_REPORT_VCC        0
#define HEARTBEAT_REPORT_RELAY      1
#define HEARTBEAT_REPORT_LIGHT      0
#define HEARTBEAT_REPORT_HOSTNAME   0
#define HEARTBEAT_REPORT_DESCRIPTION 0
#define HEARTBEAT_REPORT_APP        0
#define HEARTBEAT_REPORT_VERSION    0
#define HEARTBEAT_REPORT_BOARD      0
#define HEARTBEAT_REPORT_LOADAVG    0
#define HEARTBEAT_REPORT_INTERVAL   0

// -----------------------------------------------------------------------------
// WEBSOCKETS
// -----------------------------------------------------------------------------

// This will only be enabled if WEB_SUPPORT is 1 (this is the default value)
#define WS_AUTHENTICATION           0           // WS authentication ON by default (see #507)

// -----------------------------------------------------------------------------
// ALEXA
// -----------------------------------------------------------------------------

// This setting defines whether Alexa support should be built into the firmware
#define ALEXA_SUPPORT               1               // Enable Alexa support by default (10.84Kb)

// This is default value for the alexaEnabled setting that defines whether
// this device should be discoberable and respond to Alexa commands.
// Both ALEXA_SUPPORT and alexaEnabled should be 1 for Alexa support to work.
#define ALEXA_ENABLED               0

//------------------------------------------------------------------------------
// GENERAL
//------------------------------------------------------------------------------

[... 385 lines cut ...]
        #define BUTTON1_PIN         5
        //#define BUTTON1_MODE        BUTTON_SWITCH
        #define BUTTON1_MODE        BUTTON_PUSHBUTTON // | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
        #define BUTTON1_RELAY       1

        #define BUTTON1_PRESS       BUTTON_MODE_TOGGLE
        #define BUTTON1_CLICK       BUTTON_MODE_NONE
        #define BUTTON1_DBLCLICK    BUTTON_MODE_NONE
        #define BUTTON1_LNGCLICK    BUTTON_MODE_NONE
        #define BUTTON1_LNGLNGCLICK BUTTON_MODE_NONE

        // Relays
        #define RELAY1_PIN          4
        #define RELAY1_TYPE         RELAY_TYPE_NORMAL

        // Non-core modules
        #define ALEXA_SUPPORT               1
        #define DOMOTICZ_SUPPORT            1
        #define HOMEASSISTANT_SUPPORT       1
        #define I2C_SUPPORT                 0
        #define SCHEDULER_SUPPORT           1
        #define SENSOR_SUPPORT              0
        #define THINGSPEAK_SUPPORT          0

#endif`

rmcbc, 2020-02-15:
If I do a Factory reset it show work, the ip0, ip1, ip2 and ip3 show be corrected but is not.

mcspr, 2020-02-15:
I mean, can you repeat command sequence from above? Storing `ip0` value, `reload` and checking if it is still there via `get`

rmcbc, 2020-02-15:
[image]
[image]

rmcbc, 2020-02-15:
@mcspr in the first screen I did the set then the get
in the second reload and get again

mcspr, 2020-02-15:
So I see the issue then.

*Also a nice reminder that we probably should show the typed command in the webui debug log :)*

mcspr, 2020-02-15:
cc @Tofandel , as it is somewhat related to the previous discussion related to wifi settings ui

`_wifiClean(num)` does it's thing based on the old assumption that empty setting string should be removed. The issue is when `WIFI1_SSID` is set to something, that means that `ssid0` is never created. Changing `ip0` (static IP) causes settings reload and a call to `_wifiClean()`, which sees that `ssid0` is empty and unintentionally removes `ip0` (with every other network setting index 0).

PR #2151 (2020-02-18): wifi: respect hardcoded settings when checking for deletion
fix #2150, keeping the existing behaviour deleting empty ssid
small fix for wps / smartconfig to detect hardcoded settings
update comments for WIFIN_..., since we have up to 5 now
closed 2020-02-18 by mcspr