xoseperez/espurna #2152: InfluxDB: measurements created, but no data

tbernaerd, 2020-02-16:
I have a Sonoff Pow R2 with Espurna flashed onto it. I made my own build with the instructions in VS Code just to have influxdb support. Build seems to work, but adding data to Influx does not work so far.

The debug log in the web UI:

```
[540559] [INFLUXDB] Sending to 192.168.0.15:8086
[540563] [INFLUXDB] Connected to 192.168.0.15:8086
[540705] [INFLUXDB] Failure response after 142ms
[540951] [INFLUXDB] Disconnected
```

In influx, I can see the new measurements created but no data:

```
Connected to http://localhost:8086 version 1.7.9
InfluxDB shell version: 1.7.9
> use espurna
Using database espurna
> show measurements
name: measurements
name
----
apparent
current
energy
factor
freeheap
loadavg
power
reactive
rssi
ssid
uptime
vcc
voltage

```

Any ideas on how I can troubleshoot this further? The debug log does not seem too verbose.

mcspr, 2020-02-17:
Have you checked influxdb logs?

mcspr, 2020-02-17:
'Failure' notice is erroneous: 
https://github.com/xoseperez/espurna/blob/7862b52060b465b5a1b9363fd235cd5540330763/code/espurna/influxdb.ino#L58-L59
`sizeof(char[])` != `strlen(char[])`, checking for HTTP string it failed to read it properly. Fixing that in #2153 (plus some unrelated refactoring that was bound to happen at some point)

So far I am not seeing any issues with influx part:
```
[17/Feb/2020:15:26:33 +0000] "POST /write?db=test&p=%5BREDACTED%5D&u=test HTTP/1.1" 204 0 "-" "-" e129c79b-5199-11ea-8158-0242ac110002 415489
```
```
> select time,value from vcc where device = 'testdevice' order by time desc limit 10
name: vcc
time                value
----                -----
1581953397930141583 3418
1581953097545009682 3420
1581952797586094950 3418
1581952497827285398 3420
1581952197544267180 3418
1581951897560531068 3418
1581951597526623977 3418
1581951297942935012 3418
1581950997484544597 3420
1581950697802125858 3418
```

tbernaerd, 2020-02-17:
You are right, I do have data in Influx. I got confused while looking at another Influxdb instance, which is also why the Influx logs were not very useful. 

Changes to get rid of the erroneous notice will be better to not confuse future users though, thanks for that.

PR #2153 (2020-02-18): influxdb: fix http response parsing, refactor module scope
fix #2152 

fix strncmp going out-of-intended-bounds
(forgetting '\0', because `char var[] = "abcde"; sizeof(var) != strlen(var)`)

change scope of variables a bit
closed 2020-02-18 by mcspr