xoseperez/espurna #2160: HC-SR501 PIR to control Sonoff Basic internal relay.

davebuk, 2020-02-19:
I plan to connect an HC-SR501 PIR to a GPIO in a Sonoff Basic. I want the PIR to operate the relay to switch a light on/off.

I see from post https://github.com/xoseperez/espurna/issues/503#issuecomment-362274817 others have used a button definition.

Can I use `#define BUTTON2_RELAY	1` to turn the relay ON/OFF when the PIR is triggered? 

The other idea I had was to define:

```
    // Info
    #define MANUFACTURER        "ITEAD"
    #define DEVICE              "SONOFF_BASIC"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          12
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // PIR Digital IN
    #define DIGITAL_SUPPORT      1
    #define DIGITAL1_PIN         16  // OR equivalent 
    #define DIGITAL1_PIN_MODE    INPUT
    #define DIGITAL1_DEFAULT_STATE 	0
```
Will adding `#define DIGITAL1_RELAY	1` bind the digital sensor to the relay?

mcspr, 2020-02-20:
Check out `defaults.h` or `sensors.h` for available flags. Button should work if PIR operates like button, switching to low temporarily? This may change which event it picks (CLICK, LNGCLICK etc.)

There is no `DIGITAL1_RELAY`, so it won't pick up something like that. This snippet will though (somewhere in espurna.ino setup() for example, at the end):
```cpp
SensorReadBroker::Register([](const String& magnitude, unsigned char, double value, const char*) {
    if (magnitude.equals("digital")) {
        relayStatus(0, (value > 0));
    }
});
```

davebuk, 2020-02-20:
I built using `button3` on a sonoff mini as per #503. I haven't got the PIR yet but it goes HIGH when triggered and back to LOW. I guess if I configure ```#define BUTTON3_MODE BUTTON_PUSHBUTTON | BUTTON_DEFAULT_LOW``` this should work?

mcspr, 2020-02-20:
Button will default to LOW without `BUTTON_DEFAULT_HIGH`, there is no `BUTTON_DEFAULT_LOW` type (also see `types.h`)

davebuk, 2020-02-20:
Thanks for the clarification. I'll rebuild and try it out when the PIR arrives.

davebuk, 2020-02-24:
The following seems to work.

```
#elif defined(ITEAD_SONOFF_BASIC)  // _PIR  PIR Relay trigger

    #define MANUFACTURER        "ITEAD"
    #define DEVICE              "SONOFF_BASIC_PIR"

    //My config
    #define DEBUG_SERIAL_SUPPORT    0
    #define ALEXA_SUPPORT    0
    #define DOMOTICZ_SUPPORT        0
    #define HOMEASSISTANT_SUPPORT   0
    #define THINGSPEAK_SUPPORT      0
	
	// Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1
    #define BUTTON2_PIN         3       //USE RX Pin GPIO3 for switch. **DISABLE DEBUG_SERIAL_SUPPORT**
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON
    #define BUTTON2_RELAY       1
    #define BUTTON2_PRESS       BUTTON_MODE_ON
    #define BUTTON2_CLICK       BUTTON_MODE_OFF
    #define BUTTON2_DBLCLICK    BUTTON_MODE_NONE
    #define BUTTON2_LNGCLICK    BUTTON_MODE_OFF
    #define BUTTON2_LNGLNGCLICK BUTTON_MODE_OFF
	
    // Relays
    #define RELAY1_PIN          12
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    1
```

As expected, I get `EVENT 1` when the PIR senses movement (LOW to HIGH) and the relay switches ON. After no detected movement and setting a short time delay, less than 2 seconds, I then get an `EVENT 4`  (High back to LOW) and the relay switches OFF. Setting a longer time delay, say 10 seconds, I then get `EVENT 5`.

Is there a limit to how long espurna will keep the `PRESSED` state as valid? Will the firmware be happy with being in that state for say 10 minutes before being switched back to LOW and reporting the `EVENT 5`?

mcspr, 2020-02-24:
> Is there a limit to how long espurna will keep the PRESSED state as valid? Will the firmware be happy with being in that state for say 10 minutes before being switched back to LOW and reporting the EVENT 5?

Nope. When state bounces back to the default, event handler measures for how long it stayed in that state so we get one of CLICK (2), LNGCLICK (4) or LNGLNGCLICK (5). Only conceivable issue length value overflow, but you would need to hold the button for ~49 days

davebuk, 2020-02-24:
Excellent, 49 days should be fine! 😁 Thanks for the advise.

closed 2020-02-24 by davebuk