xoseperez/espurna #2165: BME280 pressure value over reading.

davebuk, 2020-02-26:
I have a BME280 environment sensor connected to a Sonoff Basic. It works fine but the pressure reading seems to be about 10 mbar too high. There are some comments about a similar situation here: https://github.com/rwaldron/johnny-five/issues/1274#issue-200767502 but I don't know which library espurna uses for the BME280 or if that link is even relevant.

The sensor section has an offset section for temp and humidity but not for pressure. Can this be added?

mcspr, 2020-02-26:
See config/sensors.h BMX280_PRESSURE comment.
1 -> 1x oversample
2 -> 2x
3 -> 4x
4 -> 8x (value mentioned in the other issue)
5 -> 16x

That is something missing from runtime settings.

davebuk, 2020-02-26:
No difference when building using `#define BMX280_PRESSURE 4` or `#define BMX280_PRESSURE 5` in `sensors.h`. I did have the sensor read rate set to 10 seconds and report every 12 for a two minute update, but I have changed it to 1 minute read, report every 2. I'll see if it changes overnight.

Maybe the BME280 has become faulty. I'm sure it used to be more accurate.

davebuk, 2020-02-27:
Sensor is still 10 mbar over reading. As I use the reading via mqtt to openHAB, I have created a proxy item to subtract 10. I'll see if the error gets worse over time.

Thanks.

closed 2020-02-27 by davebuk

mcspr, 2020-02-28:
OK, we will see then :)

btw, based on the issues linked at the end, you might see the bmx chip heating up more than usual when using high oversampling values.
I also wonder if standby times can change things too. See sensor.h comments, still.

davebuk, 2020-02-28:
I guess it can set all of the definitions, e.g. `#define BMX280_STANDBY  1` in custom.h file rather than editing sensors.h?

davebuk, 2020-02-29:
For reference I tried: `#define BMX280_PRESSURE 4`,	`#define BMX280_PRESSURE 5` both using `#define BMX280_STANDBY 0 and 5`. None of these make a difference to the error.

I even tried the settings below but that didn't help.
```
#define BMX280_PRESSURE 1
#define BMX280_STANDBY 5
#define BMX280_MODE 3
```

mcspr, 2020-03-02:
Re 
> I guess it can set all of the definitions, e.g. #define BMX280_STANDBY 1 in custom.h file rather than editing sensors.h?
> Does it matter the order in which the settings are defined?

We use c++ preprocessor rules here. Anything with `#ifndef FLAG` / `#if not defined(FLAG)` allows us to check before applying defaults. e.g. in sensors it checks for `#ifndef BMX280_STANDBY` before setting it to 1.
`-DFLAG` in build flags takes precedence (it is loaded even before custom.h), but it has the same effect as custom header `#define FLAG`. So you would see:
```
test.cpp:62: warning: "FLAG" redefined
   62 | #define FLAG 1
      |
<command-line>: note: this is the location of the previous definition
```
When flag needs to be changed directly in the header file.