xoseperez/espurna #2166: [DOMOTICZ] Error parsing data

karmacoma92, 2020-02-27:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
Error parsing data keeps on appearing with DOMOTICZ:

> [268971] [MQTT] Received domoticz/out => {
> "Battery" : 255,
> "RSSI" : 12,
> "description" : "",
> "dtype" : "General",
> "hwid" : "5",
> "id" : "00000001",
> "idx" : 63,
> [268982] [DOMOTICZ] Error parsing data
> [268991] [MQTT] Received domoticz/out =>
> "name" : "Cloud Cover",
> "nvalue" : 0,
> "stype" : "Percentage",
> "svalue1" : "47.00",
> "unit" : 1
> }
> 
> [268998] [DOMOTICZ] Error parsing data
> [282142] [SENSOR] DHT22 @ GPIO14 - temperature: 21.6C
> [282143] [SENSOR] DHT22 @ GPIO14 - humidity: 42%
> [292409] [MQTT] Received domoticz/out => {
> "Battery" : 255

It also happens in another device with an analog sensor:
> [030787] [DOMOTICZ] Error parsing data
> [030793] [MQTT] Received domoticz/out => "name" : "Cloud Cover",
> 	"nvalue" : 0,
> 	"stype" : "Percentage",
> 	"svalue1" : "33.00",
> 	"unit" : 1
> }
> 
> [030803] [DOMOTICZ] Error parsing data
> 

**Steps to reproduce**
It just happens 

**Expected behavior**
[DOMOTICZ] Error parsing data

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
[000178] [MAIN] ESPURNA 1.14.2.PA5-dev (200219)
[000178] [MAIN] xose.perez@gmail.com
[000178] [MAIN] http://tinkerman.cat

[000180] [MAIN] CPU chip ID: 0x8E1600
[000183] [MAIN] CPU frequency: 80 MHz
[000186] [MAIN] SDK version: 2.2.2-dev(38a443e)
[000191] [MAIN] Core version: 2.6.3
[000194] [MAIN] Core revision: 3d128e5c
[000197] [MAIN] Build time: 1582416957
[000200] 
[000202] [MAIN] Flash chip ID: 0x146085
[000205] [MAIN] Flash speed: 40000000 Hz
[000208] [MAIN] Flash mode: DOUT
[000211] 
[000212] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[000219] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[000226] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[000234] [MAIN] Firmware size       :   569376 bytes /  140 sectors (   1 to  140)
[000241] [MAIN] Max OTA size        :   450560 bytes /  110 sectors ( 141 to  250)
[000248] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[000255] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[000262] 
[000264] [MAIN] EEPROM sectors: 251, 250
[000267] [MAIN] EEPROM current: 250
[000270] 
[000271] [MAIN] EEPROM:  4096 bytes initially |  1302 bytes used (31%) |  2794 bytes free (68%)
[000279] [MAIN] Heap  : 40984 bytes initially |  5280 bytes used (12%) | 35704 bytes free (87%)
[000288] [MAIN] Stack :  4096 bytes initially |  1232 bytes used (30%) |  2864 bytes free (69%)
[000296] 
[000297] [MAIN] Boot version: 31
[000300] [MAIN] Boot mode: 1
[000302] [MAIN] Last reset reason: External System
[000307] [MAIN] Last reset info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[000320] 
[000322] [MAIN] Board: GENERIC_ESP01S_DHT11_10
[000325] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THINGSPEAK WEB 
[000341] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[000345] [MAIN] Sensors: DHTXX 
[000347] [MAIN] WebUI image: SENSOR
[000350] 
[000352] [MAIN] Power: 2454 mV
[000354] [MAIN] WiFi Sleep Mode: MODEM
[000357] 
*If you cannot get this info from the device, please answer this questions:*
Core version: 2.6.3
ESPURNA 1.14.2.dev
curl and esptool
GENERIC_ESP01S_DHT11_10

**Tools used**
Ubuntu 19.10
Firefox 73.0.1 (64-bit)
PlatformIO Home 3.1.0·Core 4.2.1

**Additional context**
*Add any other context about the problem here.*

mcspr, 2020-02-28:
How can I reproduce domoticz installation? (config files, os etc.)
Which os domoticz runs on?

karmacoma92, 2020-02-28:
Hi,

Domoticz Version: 4.11740
OS is raspbian buster
regarding installation files what do you need exactly?

Thanks
Oscar

mcspr, 2020-03-01:
nvm, I have installed beta domoticz. but, I don't see the issue described here when setting up dummy switch.

What is the MQTT broker domoticz connects to? Does running any other client (on phone, `mosquitto_sub` etc.) show the same 2-parter json or is it a single message? Can you manually send the same json payload?

karmacoma92, 2020-03-02:
Ok I'll try but I don't think I'll be very successful, ...

karmacoma92, 2020-03-07:
Hi, 

finally I got some time to test this. It happens with all devices, at least all that I have checked. For example in one with DHT11

this is what I got from the DEBUG:

>[463319] [MQTT] Received domoticz/out => {
>	"Battery" : 255,
>	"RSSI" : 12,
>	"description" : "",
>	"dtype" : "General",
>	"hwid" : "5",
>	"id" : "00000001",
>	"idx" : 63,
>[463333] [DOMOTICZ] Error parsing data
>[463347] [MQTT] Received domoticz/out => 
>	"name" : "Cloud Cover",
>	"nvalue" : 0,
>	"stype" : "Percentage",
>	"svalue1" : "0.00",
>	"unit" : 1
>}

while this what I got from the broker:

[File_from_MQTT_broker.txt](https://github.com/xoseperez/espurna/files/4302309/File_from_MQTT_broker.txt)

and this is a tcpdump in the broker:
[file2.zip](https://github.com/xoseperez/espurna/files/4302319/file2.zip)

Apparently there is nothing in the neither in the json file nor in the tcpdump file, ...

Could you tell me next step? 
thanks
Oscar

mcspr, 2020-03-07:
Could you modify `code/espurna/mqtt.ino` like this and run a second test?
```diff
diff --git a/code/espurna/mqtt.ino b/code/espurna/mqtt.ino
index 06c79f67..109f72e3 100644
--- a/code/espurna/mqtt.ino
+++ b/code/espurna/mqtt.ino
@@ -975,6 +975,11 @@ void mqttSetup() {
             _mqttOnDisconnect();
         });
         _mqtt.onMessage([](char* topic, char* payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total) {
+            DEBUG_MSG_P(PSTR("[MQTT] %s pkt:%u:%c:%c msg:%u:%u:%u\n"),
+                topic,
+                properties.qos, (properties.dup ? 'y' : 'n'), (properties.retain ? 'y' : 'n'),
+                len, index, total
+            );
             _mqttOnMessage(topic, payload, len);
         });
         _mqtt.onSubscribe([](uint16_t packetId, uint8_t qos) {
```
I think the issue is on our side, as we don't check for total length of the packet from the current batch, see packets №29 and №31 from the capture.

And I would expect to see two consecutive logs about domoticz/out showing different index but the same total, at least that would allow us to properly parse things by buffering the previous part.

karmacoma92, 2020-03-08:
then do you need me to take a new tcpdump, here you have the output from the console:

>[702148] [MQTT] Received domoticz/out => {
>	"Battery" : 255,
>	"RSSI" : 12,
>	"description" : "",
>	"dtype" : "General",
>	"hwid" : "5",
>	"id" : "00000001",
>	"idx" : 63,
>	
>[702163] [DOMOTICZ] Error parsing data
>[702181] [MQTT] domoticz/out pkt:0:n:n msg:99:126:225
>[702182] [MQTT] Received domoticz/out => "name" : "Cloud Cover",
>	"nvalue" : 0,
>	"stype" : "Percentage",
>	"svalue1" : "2.00",
>	"unit" : 1
>}
>
>[702189] [DOMOTICZ] Error parsing data
>[709873] [MQTT] domoticz/out pkt:0:n:n msg:304:0:304
>[709874] [MQTT] Received domoticz/out => {
>	"Battery" : 255,
>	"RSSI" : 12,
>	"description" : "",
>	"dtype" : "Temp + Humidity",
>	"hwid" : "13",
>	"id" : "82026",
>	"idx" : 26,
>	"name" : "ESP07-Caldera-T+H",
>	"nvalue" : 0,
>	"stype" : "THGN122/123/132, THGR122/228/238/268",
>	"svalue1" : "15.30",
>	"svalue2" : "63.30",
>	"svalue3" : "1",
>	"unit" : 1
>}

and here you have the output from the broker:

>domoticz/out {
>	"Battery" : 255,
>	"RSSI" : 12,
>	"description" : "",
>	"dtype" : "General",
>	"hwid" : "5",
>	"id" : "00000001",
>	"idx" : 63,
>	"name" : "Cloud Cover",
>	"nvalue" : 0,
>	"stype" : "Percentage",
>	"svalue1" : "2.00",
>	"unit" : 1
>} 

here you have the file 

[file6.zip](https://github.com/xoseperez/espurna/files/4303655/file6.zip)

kind regards and thanks a lot for your help!
Oscar

karmacoma92, 2020-03-08:
Hi 

I just found this:   

> [303707] [MQTT] domoticz/out pkt:0:n:n msg:227:0:227
> [303710] [MQTT] Received domoticz/out => {
> 	"Battery" : 255,
> 	"RSSI" : 12,
> 	"description" : "",
> 	"dtype" : "General",
> 	"hwid" : "5",
> 	"id" : "0",
> 	"idx" : 6,
> 	"name" : "Visibility-Out-Madrid",
> 	"nvalue" : 0,
> 	"stype" : "Visibility",
> 	"svalue1" : "16.1",
> 	"unit" : 1
> }
> 
> [303735] [MQTT] domoticz/out pkt:0:n:n msg:233:0:233
> [303738] [MQTT] Received domoticz/out => {
> 	"Battery" : 255,
> 	"RSSI" : 12,
> 	"description" : "",
> 	"dtype" : "General",
> 	"hwid" : "5",
> 	"id" : "00000100",
> 	"idx" : 51,
> 	"name" : "Ozone Sensor",
> 	"nvalue" : 0,
> 	"stype" : "Custom Sensor",
> 	"svalue1" : "337.5000",
> 	"unit" : 1
> }
> 
> [303763] [MQTT] domoticz/out pkt:0:n:n msg:225:0:225
> [303774] [MQTT] Received domoticz/out => {
> 	"Battery" : 255,
> 	"RSSI" : 12,
> 	"description" : "",
> 	"dtype" : "General",
> 	"hwid" : "5",
> 	"id" : "00000001",
> 	"idx" : 63,
> 	"name" : "Cloud Cover",
> 	"nvalue" : 0,
> 	"stype" : "Percentage",
> 	"svalue1" : "5.00",
> 	"unit" : 1
> }
> 
> [323059] [SENSOR] DHT11 @ GPIO2 - temperature: 21.5°C
> [323059] [SENSOR] DHT11 @ GPIO2 - humidity: 48%

So it is not always the problem appear and just in case you would like to compare
thanks
Oscar

mcspr, 2020-03-10:
Thanks! tcpdump (as previous one) & show the problem:
```
[702181] [MQTT] domoticz/out pkt:0:n:n msg:99:126:225
                                              ^^^
```

I just need to think a bit what exactly makes sense here as API wrapper. We should not cache everything based on `length`, because maximum size of MQTT payload could be up to 268,435,456 bytes. But domoticz module could perhaps track `index` and store some data temporarily.

karmacoma92, 2020-03-14:
Hi Max, 

the fact that "Some checks were not successful - 1 failing and 1 successful checks" is preventing me from testing this, do you still are expecting me to do it?

thanks 
Oscar

mcspr, 2020-03-14:
Looking at Codacy again:
> code/espurna/mqtt.ino
Skipping configuration 'MQTT_MAX_PACKET_SIZE;MQTT_SUPPORT' since the value of 'MQTT_MAX_PACKET_SIZE' is unknown. Use -D if you want to check it. You can use -U to skip it explicitly.
    if (!len || (len >= MQTT_MAX_PACKET_SIZE) || (total >= MQTT_MAX_PACKET_SIZE)) return;

> Skipping configuration 'MQTT_MAX_PACKET_SIZE;MQTT_SUPPORT' since the value of 'MQTT_MAX_PACKET_SIZE' is unknown. Use -D if you want to check it. You can use -U to skip it explicitly.
    static char message[MQTT_MAX_PACKET_SIZE] = {0};

Is a "non issue", since MQTT_MAX_PACKET_SIZE flag is provided by the platformio.ini config:
https://github.com/xoseperez/espurna/blob/032f921fc4c4758c80c5c3068b190b608acb8624/code/platformio.ini#L79
I would make changes to expose it as custom.h config though, but you should be fine testing it as-is.

karmacoma92, 2020-03-14:
ok, I'll do that

karmacoma92, 2020-03-15:
Hi 

There is a bug, at least for me, in mqtt.ino it appears 

> MQTT_MAX_PACKET_SIZE 

while in platformio.ini:

> MQTT_mAX_PACKET_SIZE

that solves the problem of 'MQTT_MAX_PACKET_SIZE' but there is still one issue 

> PlatformIO/Projects/espurna/code/espurna/mqtt.ino: In function 'void _mqttInfo()':
> PlatformIO/Projects/espurna/code/espurna/mqtt.ino:437:9: error: '_mqtt_connecting' was not declared in this scope
 >        _mqtt_connecting ? "CONNECTING" : "WAITING",
 >        ^
> espurna/debug.h:36:42: note: in definition of macro 'DEBUG_MSG_P'
>     #define DEBUG_MSG_P(...) debugSend_P(__VA_ARGS__)

which I do not know how to fix.....
thanks 
Oscar

mcspr, 2020-03-15:
Have you cherry-picked the buffering commit? The issue in mqttInfo is fixed in the subsequent commit, I kind of expected you to checkout the full patch list.

I have reworked configuration a bit and forcepushed new version. Please check it out again either by
- https://patch-diff.githubusercontent.com/raw/xoseperez/espurna/pull/2181.patch
- git-remote-add'ing my repo at https://github.com/mcspr/espurna and using 'mqtt/stream-buf' branch
- `git fetch origin pull/2181/head:pr2181` and `git checkout pr2181` (which is something github does, https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/checking-out-pull-requests-locally#modifying-an-inactive-pull-request-locally)

karmacoma92, 2020-03-15:
Excuse me Max, 
but could you tell me, how can I go back? how can I undo it?
thanks

karmacoma92, 2020-03-15:
Hi there, I think I'll wait for you to push the changes to the dev branch
thanks

mcspr, 2020-03-15:
What do you mean by 'undo'? Can you use my branch directly or did you add some more commits on top of you own? You can simply roll the branch back to an older commit [1]:

`$ git branch -f your-branch new-HEAD-commit-hash` to reset branch to specific commit, without my changes. Then, apply the patch above.
`$ git rebase -i HEAD~4` to show an interactive branch editor that will allow to pick again / drop / edit 4 latest commits. You can drop my commits and keep only yours. If you have fetched the current repo, you can also add series of patches by hash by adding `pick 
bb7b2a3` newline `pick 
8be6d30` etc.

[1] https://git-scm.com/book/en/v2/Git-Branching-Branches-in-a-Nutshell

karmacoma92, 2020-03-15:
Hi Max, 

sorry but I'm a newbie here and I do not know how to manage with versions. The thing is I have added some stuff of mine on top of standard. 

going back, could you please tell me what I'm doing wrong?

> $ git remote -v
> mqtt-stream-buf	https://github.com/mcspr/espurna/tree/mqtt/stream-buf (fetch)
> mqtt-stream-buf	https://github.com/mcspr/espurna/tree/mqtt/stream-buf (push)
> origin	https://github.com/karmacoma92/espurna.git (fetch)
> origin	https://github.com/karmacoma92/espurna.git (push)
> $ git fetch mqtt-stream-buf pull/2181/head:pr2181 
> fatal: repository 'https://github.com/mcspr/espurna/tree/mqtt/stream-buf/' not found

thanks again!

karmacoma92, 2020-03-15:
Hi Max,

sorry but I'm a newbie here and I do not know how to manage with versions. The thing is I have added some stuff of mine on top of standard.

going back, could you please tell me what I'm doing wrong?

    $ git remote -v
    mqtt-stream-buf https://github.com/mcspr/espurna/tree/mqtt/stream-buf (fetch)
    mqtt-stream-buf https://github.com/mcspr/espurna/tree/mqtt/stream-buf (push)
    origin https://github.com/karmacoma92/espurna.git (fetch)
    origin https://github.com/karmacoma92/espurna.git (push)
    $ git fetch mqtt-stream-buf pull/2181/head:pr2181
    fatal: repository 'https://github.com/mcspr/espurna/tree/mqtt/stream-buf/' not found

thanks again!

mcspr, 2020-03-15:
Ok, let's just use the repo approach then. Issue is that the remote url should be `https://github.com/mcspr/espurna`, not `.../tree/...`. In the current state:
```
$ git remote rename mqtt-stream-buf mcspr
$ git remote set-url mcspr https://github.com/mcspr/espurna
$ git fetch mcspr mqtt/stream-buf
$ git checkout -b pr2181 remotes/mcspr/mqtt/stream-buf
```
Intent here is that later down the line you can always use `mcspr` remote to fetch my branches.

karmacoma92, 2020-03-16:
Hi,I think smething is not working, I'll explain myself, this is what I'm receiving at the DUT:

>    [808510] [MQTT] Sending miCasa/8E1659-DHT11-Spare2/loadavg => 1 (PID 1)
>    [830341] [SENSOR] DHT11 @ GPIO2 - temperature: 19.5°C
>    [830342] [SENSOR] DHT11 @ GPIO2 - humidity: 47%
>    [860346] [SENSOR] DHT11 @ GPIO2 - temperature: 19.5°C
>    [860346] [SENSOR] DHT11 @ GPIO2 - humidity: 47%
>    [890349] [SENSOR] DHT11 @ GPIO2 - temperature: 19.5°C
>    [890350] [SENSOR] DHT11 @ GPIO2 - humidity: 47%
>    [920360] [SENSOR] DHT11 @ GPIO2 - temperature: 20.5°C
>    [920360] [SENSOR] DHT11 @ GPIO2 - humidity: 47%
>    [950362] [SENSOR] DHT11 @ GPIO2 - temperature: 19.5°C
>    [950363] [SENSOR] DHT11 @ GPIO2 - humidity: 47%
>    [980369] [SENSOR] DHT11 @ GPIO2 - temperature: 20.5°C
>    [980370] [SENSOR] DHT11 @ GPIO2 - humidity: 48%
>    [010371] [SENSOR] DHT11 @ GPIO2 - temperature: 20.5°C
>    [010372] [SENSOR] DHT11 @ GPIO2 - humidity: 48%
>    [040382] [SENSOR] DHT11 @ GPIO2 - temperature: 20.5°C
>    [040383] [SENSOR] DHT11 @ GPIO2 - humidity: 48%

while this is what I'm receiving in another device

> [145739] [SENSOR] DHT22 @ GPIO14 - temperature: 22.3°C
> [145740] [SENSOR] DHT22 @ GPIO14 - humidity: 55%
> [160745] [SENSOR] DHT22 @ GPIO14 - temperature: 22.3°C
> [160746] [SENSOR] DHT22 @ GPIO14 - humidity: 55%
> [173978] [MQTT] domoticz/out pkt:0:n:n msg:304:0:304
> [173979] [MQTT] Received domoticz/out => {
> 	"Battery" : 255,
> 	"RSSI" : 12,
> 	"description" : "",
> 	"dtype" : "Temp + Humidity",
> 	"hwid" : "13",
> 	"id" : "82026",
> 	"idx" : 26,
> 	"name" : "ESP07-Caldera-T+H",
> 	"nvalue" : 0,
> 	"stype" : "THGN122/123/132, THGR122/228/238/268",
> 	"svalue1" : "10.30",
> 	"svalue2" : "85.50",
> 	"svalue3" : "3",
> 	"unit" : 1
> }
> 
> [175750] [SENSOR] DHT22 @ GPIO14 - temperature: 22.4°C
> [175751] [SENSOR] DHT22 @ GPIO14 - humidity: 55%
> [190757] [SENSOR] DHT22 @ GPIO14 - temperature: 22.4°C

shouldn't I be receiving something from the other nodes at the same time as well?

mcspr, 2020-03-16:
Log is just missing. If you were to publish something manually on `domoticz/out`:
```shell
$ mosquitto_pub -t 'domoticz/out' -m '1'
```
On the device:
```
[046451] [DOMOTICZ] Error parsing data
```

karmacoma92, 2020-03-16:
I'm getting this in the device

`[966374] [DOMOTICZ] Error parsing data`

but nothing in the broker.....

mcspr, 2020-03-16:
I'll restore the log asap, but it is harmless though? Sensors data that you receive are not used by the device, its just logging them

karmacoma92, 2020-03-16:
okay

mcspr, 2020-03-16:
See the latest commit, pull etc.

I may have added off-by-one issue because of index, but it seems to be at the correct offset (meaning msg[len + index] = '\0' part)

karmacoma92, 2020-03-16:
it seems to be working now:

````
}

[286149] [MQTT] topic=domoticz/out mqtt:qos=0:dup=n:ret=n msg:len=233:idx=0:total=233
[286156] [MQTT] Received domoticz/out => {
	"Battery" : 255,
	"RSSI" : 12,
	"description" : "",
	"dtype" : "General",
	"hwid" : "5",
	"id" : "00000100",
	"idx" : 51,
	"name" : "Ozone Sensor",
	"nvalue" : 0,
	"stype" : "Custom Sensor",
	"svalue1" : "328.6000",
	"unit" : 1
}

[286181] [MQTT] topic=domoticz/out mqtt:qos=0:dup=n:ret=n msg:len=226:idx=0:total=226
[286193] [MQTT] Received domoticz/out => {
	"Battery" : 255,
	"RSSI" : 12,
	"description" : "",
	"dtype" : "General",
	"hwid" : "5",
	"id" : "00000001",
	"idx" : 63,
	"name" : "Cloud Cover",
	"nvalue" : 0,
	"stype" : "Percentage",
	"svalue1" : "21.00",
	"unit" : 1
}
````

karmacoma92, 2020-03-16:
I think this one was the one to blame:

````
[768048] [MQTT] topic=domoticz/out mqtt:qos=0:dup=n:ret=n msg:len=122:idx=0:total=226
[768062] [MQTT] topic=domoticz/out mqtt:qos=0:dup=n:ret=n msg:len=104:idx=122:total=226
[768062] [MQTT] Received domoticz/out => {
	"Battery" : 255,
	"RSSI" : 12,
	"description" : "",
	"dtype" : "General",
	"hwid" : "5",
	"id" : "00000001",
	"idx" : 63,
	"name" : "Cloud Cover",
	"nvalue" : 0,
	"stype" : "Percentage",
	"svalue1" : "12.00",
	"unit" : 1
}
````
Am I right? do you think it fits?

mcspr, 2020-03-16:
Looks like it is working.

karmacoma92, 2020-03-16:
Yes, I agree

karmacoma92, 2020-03-17:
I think you can close this issue

PR #2181 (2020-03-20): Async MQTT: buffer some data
fix #2166 
@karmacoma92, I think this should work now.

We already buffer the data to inject '\0' at the end, but we never re-use the buffer. Instead, allocate `char buf[MQTT_MAX_PACKET_SIZE]` and place the data there until we got the full message. Static allocation, avoid dynamic `char buf[len+1]`. Default buffer size is 1024 bytes.
Also avoid using strlcpy, since we already know the length (and because of weird len+1)
closed 2020-03-20 by mcspr