xoseperez/espurna #2167: When or how flash a Blitzwofif BW-SHP7?

braserito66, 2020-02-27:
Hi, I own a BW-SHP7 dual socket, and I wonder when or how it is possible to flash it.
Is there any other device with same binary? If necessary I could open and investigate where to connect comms and DI00, lines

mcspr, 2020-02-28:
See https://templates.blakadder.com/blitzwolf_SHP7.html for the pinout

We don't have any pre-defined hw yet, but see VANZAVANZU_SMART_WIFI_PLUG_MINI in  [`hardware.h`](https://github.com/xoseperez/espurna/blob/dev/code/espurna/config/hardware.h) as an example.

braserito66, 2020-02-29:
Ok, I'll try to flash it, first in order to make a copy of original firmware, I'll to make a copy connecting power, Tx, Rx, and I00 wires as shown:
[image]
[image]
[image]

braserito66, 2020-02-29:
esptool.py works, is possible read and write at memory chip, using such connections.
Next is using arduino ide modify hardware.h to a working config.
Is there any guide to follow to do that? using (https://templates.blakadder.com/blitzwolf_SHP7.html for the pinout)
Thanks

dreamfalcon, 2020-03-25:
I managed to flash tasmota but would prefer to use espurna.
I'm trying to convert the tasmota template to espurna using the BLITZWOLF_BWSHPX_V23 as a example, but is not easy for a newbie.

brafaelsilva, 2020-06-03:
Hello,
Is it possible to have individual power using this socket? Ex: current power of in slot device A is 100w and current power of device in slot b is 250w ? Or can we only have the total power from both ?
thanks

pepsonEL, 2020-10-14:
Hi
I have a problem. I try flash SHP7 and i can not. I connect GND from SHP7 to GND, I00 also to GND, TX to RX on FTDI, RX to TX on FTDI, VCC to VCC and not found chip to flash. Why ? WHen connect FTDI to USB i connect I00 to GND and after about 1-2sec diconect it.

pepsonEL, 2020-10-16:
Anybody can help me.... please

WhoCares1337, 2021-04-30:
I managed to flash it OTA using the latest tuya-convert.

GNDeSouza, 2021-05-03:
@pepsonEL, did you manage to program it?  I am having the same problem.

WhoCares1337, 2021-05-03:
I think we still need to wait a bit for a binary that supports all the features 😅

GNDeSouza, 2021-05-03:
Just some information that may help others...
1) I am not using a SHP7, but a Gosund WP2, which I found out to be quite similar to each other, but still, read this with care;
2) I was able to program my WP2s using tuya-convert 2.4.3 and OTA -- out of the box.  But be careful, the manufacturing date has huge impact on your success (newer firmwares don't seem to be tuya compatible at all -- reportedly 2.4.3, 2.4.4, or 2.4.5);
3) after uploading my own code and experimenting with the GPIOs (the documented numbers were not working), I lost OTA and had to crack the plug open.  I connected the wires (Tx, Rx, Vcc, GND, GPIO0) and after a few trials (watch for Vcc=3.3V, if instead it is close or lower than 3.1V, it won't program) and was able to 'play' more with the GPIOs; 
4) some configurations of the GPIOs as 'input' or 'output' will prevent the device from communicating with the outside (e.g. UART), and even turn off or disturb the WiFi to the point it won't work anymore.
I am still not able to determine the function of all GPIOs (as relay1, or relay2, etc...)  If any one know those for sure, I would appreciate any hints.

mcspr, 2021-05-05:
For WP2, this should be:
https://templates.blakadder.com/gosund_WP2.html

Names for both templates can be referenced here:
https://tasmota.github.io/docs/GPIO-Conversion/
https://github.com/xoseperez/espurna/wiki/Buttons-and-switches#tasmota-templates

For OT, something like (obviously, untested)
```cpp

#elif defined(BLITZWOLF_BWSHP7)

    // Info
    #define MANUFACTURER                "BLITZWOLF"
    #define DEVICE                      "BWSHP7"

    #define DEBUG_SERIAL_SUPPORT        0

    // Relays
    #define RELAY1_PIN                  14
    #define RELAY2_PIN                  16

    // Buttons
    #define BUTTON1_PIN                 0
    #define BUTTON1_RELAY               1

    #define BUTTON2_PIN                 12
    #define BUTTON2_RELAY               2

    // LEDs
    #define LED1_PIN                    1
    #define LED1_PIN_INVERSE            1

    #define LED2_PIN                    13
    #define LED2_MODE                   LED_MODE_RELAY
    #define LED2_RELAY                  1
    #define LED2_PIN_INVERSE            1

    #define LED3_PIN                    2
    #define LED3_MODE                   LED_MODE_RELAY
    #define LED3_RELAY                  2
    #define LED3_PIN_INVERSE            1

    #ifndef HLW8012_SUPPORT
    #define HLW8012_SUPPORT             1
    #endif
    #define HLW8012_SEL_PIN             3
    #define HLW8012_CF1_PIN             5
    #define HLW8012_CF_PIN              4

    #define HLW8012_SEL_CURRENT         HIGH
    #define HLW8012_CURRENT_RATIO       25740
    #define HLW8012_VOLTAGE_RATIO       313400
    #define HLW8012_POWER_RATIO         3414290
    #define HLW8012_INTERRUPT_ON        FALLING
```
Notice that hardware.h section needs `DEBUG_SERIAL_SUPPORT 0` to allow us to use the GPIO1 and GPIO3 freely, otherwise things will break when serial driver will read / write things.

Only notable difference from other similar HLW devices is `HLW8012_SEL_CURRENT`, which needs to be HIGH so we don't treat current as voltage and voltage as current when receiving the data. 
(ref. SELi in the tasmota template, if I understood it correctly) 

Ratios are also questionable, I'd advice to manually calibrate.

GNDeSouza, 2021-05-05:
Yeah... Those don't work with my version of WP2 (V1.1 2018) and the simple program I wrote to test the GPIOs. I am collecting more information and will post here (and at Tamota) once I have it all listed.