xoseperez/espurna #2168: S20 no SoftAP (or sign of life) after flashing. Serial Monitor outputs garbage, all bauds

nstrelow, 2020-02-28:
**Problem**
I have a following problem:
Sonoff S20->flashed espurna -> SoftAP is not firing up, no leds are blinking

* erases and flashes fine without error
* tried both itead-sonoff-s20 and espurna-core
* only garbage on Serial Monitor (tried all baud rates)
* Uploading an arduino sketch works 100% fine: Can print to serial and turn on LED (GPIO13)
* also plugged it into wall socket

**Device information**
* 1.14.1 using prebuilt binaries: Tried itead-s20, espurna-core
* DOUT
* Itead Sonoff S20

**Tools used**
* Windows 10 (current Insider)
* esptool.py (newest, python 3.8)

mcspr, 2020-02-28:
Couple of things to note:
-  "Uploading an arduino sketch" means latest Core release 2.6.3? It installs different sdk, might cause problems without full flash erase / choosing option to erase SDK settings. .bin files use 2.3.0, but you can also try `nodemcu-lolin-latest.bin` that builds with recent version (nothing really different between s20 and nodemcu binaries, only default cfg. note that you will need to flash next .bin wired too, OTA will break because of different flash layout settings)
- "tried all baud rates"
115200 is the default rate for us. 74880 should show bootloader messages.
- "DOUT" was kind of my first though, but flash erase would've fixed that.

nstrelow, 2020-02-28:
* Yes, Core release 2.6.3. Good to know that it might cause problems.
* Will try nodemcu-lolin-latest.bin

Immediately after flashing + buidling via PlatformIO I got this in 115200:
```
Fatal exception (0): 
epc1=0x402357cc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```
When I replug it, Serial prints garbage again.

Any way of resetting the SDK settings for sure?

nstrelow, 2020-02-28:
@mcspr You are a hero!
`nodemcu-lolin-latest.bin` works!
I can press the button to turn the switch on and off. Also SoftAP is there.

Flashing itead-s20 afterwards, I again get the error.

Tried to use the Serial commands:
- factory.reset
- erase.config

nstrelow, 2020-02-28:
More info:
Downgraded to ESP Core 2.3.0 and flashed an Arduino sketch.
SAME problem, same garbage on Serial.
Will investigate further and see if I can downgrade.

mcspr, 2020-03-01:
> Any way to remove this incompatibility? Tried all kinds of ESP.eraseConfig() but nothing seems to make it work with the s20 prebuilt binary of espurna.

`esptool.py write_flash --erase-all` or `esptool.py erase_flash` + `write_flash` does not work? Doing eraseConfig work before it is even needed.

btw Core version is configurable through the `platform = ...` setting in the platformio.ini, either in global `[env]` or in `[env:...]`.

nstrelow, 2020-03-01:
I tried `esptool.py erase_flash + write_flash` multiple times, same error.

> btw Core version is configurable through the platform = ... setting in the platformio.ini, either in global [env] or in [env:...].

good to know, so I could build my own with a newer version.

But right now, `nodemcu-lolin-latest` is able to do everything I need.

closed 2020-05-07 by stale[bot]