xoseperez/espurna #2169: Sonoff Basic R2's turning off/freezing w. high pitched buzzing

fragande, 2020-03-03:
I'm having problems with my two Sonoff Basic R2's running Espurna 1.14.1. They both randomly turn off and freeze (non accessible) randomly. Sometimes they'll work for weeks, sometimes dies again within minutes of re-plugging. But it has become much more frequent recently. When they do this they emit a high pitched buzzing noise until disconnected from mains. I suspect this is purely a hardware problem but I'm wondering if anyone else has had this problem and if maybe there's a way to repair them.

I recently updated to 1.14.1 but it made no difference. I've also tried Tasmota firmware but it made no difference. This is what the latest crash log looks like:
```
[545893] [DEBUG] Latest crash was at 1725765660 ms after boot
[545894] [DEBUG] Reason of restart: 2
[545894] [DEBUG] Exception cause: 28
[545894] [DEBUG] epc1=0x4024a00c epc2=0x00000000 epc3=0x00000000
[545900] [DEBUG] excvaddr=0x00000044 depc=0x00000000
[545905] sp=0x3fffff40 end=0x3fffffb0 saved=0x0070

[545909] [DEBUG] >>>stack>>>
[DEBUG] 3fffff40: 4010481f 90e14ded 60000600 4023c4b1 
[DEBUG] 3fffff50: 4024b671 3fff73a0 3ffef1a0 cfa7201b 
[DEBUG] 3fffff60: 60000600 3ffef188 3ffef5fc 00000000 
[DEBUG] 3fffff70: 402356fd 3ffef1a0 3ffef188 402356fd 
[DEBUG] 3fffff80: 40235742 3fffdab0 00000000 3fffdcb0 
[DEBUG] 3fffff90: 3ffef1b8 3fffdab0 00000000 4022ce87 
[DEBUG] 3fffffa0: 40000f49 40000f49 3fffdab0 40000f49 
[DEBUG] 3fffffb0: 5e304024 00004024 d9360000 7bfd3ffe 
[DEBUG] <<<stack<<<
```

mcspr, 2020-03-14:
HW issue seems the case :/ Something with the power supply. Points of crash might not mean much if that is the case.

fragande, 2020-03-15:
> 
> 
> HW issue seems the case :/ Something with the power supply. Points of crash might not mean much if that is the case.

Yeah that's what I'm thinking too. Maybe some degrading capacitors or similar? Seems these units aren't the most well made out there. Have ordered a pair of S26's instead. Thank you for confirming my suspicion, I don't think there's a reason to keep this open.

closed 2020-03-15 by fragande