xoseperez/espurna #2184: Help, first approach

luconezena, 2020-03-17:
Sorry my english google, I'm italian
Hi, I need your precious help
Downloaded all the libraries, if I compile Espurna I get this

Arduino:1.8.5 (Windows 10), Scheda:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

button:492: error: 'T' has not been declared

 unsigned long _buttonGetSetting(const char* key, unsigned char index, T default_value) {

                                                                       
mqtt:281: error: variable or field '_mqttApplySetting' declared void

 void _mqttApplySetting(T& current, T& updated) {

                        ^

mqtt:281: error: 'T' was not declared in this scope

mqtt:281: error: 'current' was not declared in this scope

 void _mqttApplySetting(T& current, T& updated) {

                           ^

mqtt:281: error: 'T' was not declared in this scope

 void _mqttApplySetting(T& current, T& updated) {

                                    ^

mqtt:281: error: 'updated' was not declared in this scope

 void _mqttApplySetting(T& current, T& updated) {

                                       ^

mqtt:289: error: variable or field '_mqttApplySetting' declared void

 void _mqttApplySetting(T& current, const T& updated) {

                        ^

mqtt:289: error: 'T' was not declared in this scope

mqtt:289: error: 'current' was not declared in this scope

 void _mqttApplySetting(T& current, const T& updated) {

                           ^

mqtt:289: error: expected primary-expression before 'const'

 void _mqttApplySetting(T& current, const T& updated) {

                                    ^

mqtt:297: error: variable or field '_mqttApplyTopic' declared void

 void _mqttApplyTopic(T& current, const char* magnitude) {

                      ^

mqtt:297: error: 'T' was not declared in this scope

mqtt:297: error: 'current' was not declared in this scope

 void _mqttApplyTopic(T& current, const char* magnitude) {

                         ^

mqtt:297: error: expected primary-expression before 'const'

 void _mqttApplyTopic(T& current, const char* magnitude) {

                                  ^

settings:145: error: 'R' does not name a type

 R getSetting(const settings_key_t& key, R defaultValue) {

 ^

settings:176: error: 'T' does not name a type

 bool setSetting(const settings_key_t& key, const T& value) {

                                                  ^

settings:176: error: ISO C++ forbids declaration of 'value' with no type [-fpermissive]

 bool setSetting(const settings_key_t& key, const T& value) {

                                                     ^

settings:207: error: 'T' has not been declared

 String getSetting(const String& key, unsigned char index, T defaultValue) {

                                                           ^

settings:212: error: 'T' has not been declared

 bool setSetting(const String& key, unsigned char index, T value) {

                                                         ^

exit status 1
'T' has not been declared

Questo report potrebbe essere più ricco di informazioni abilitando l'opzione
"Mostra un output dettagliato durante la compilazione"
in "File -> Impostazioni"

ColinShorts, 2020-03-17:
My best advice would be to try [Platformio](https://github.com/xoseperez/espurna/wiki/PlatformIO) , it works really well and it's easy to use!

closed 2020-05-23 by stale[bot]

mcspr, 2020-05-23:
> Arduino:1.8.5

For posterity sake, see 5bb852834ef8b407dad969116783c05deb7db246, #1579
Current version (1.8.12) of the Arduino IDE should work OK. We also got rid of problematic .ino files altogether in the `dev` branch.