xoseperez/espurna #2187: Support RF KAKU protocol used by DIO chacon devices

schmurtzm, 2020-03-17:
**Is your feature request related to a problem? Please describe.**
I recently buy DIO RF plugs, these are not supported by espruna due to a new RF 433 protocol : KAKU. 

**Describe the solution you'd like**
An update of the RF protocols supported.

**Describe alternatives you've considered**

OpenMQTTGateway supports this RF protocol, [he speak about it here](https://1technophile.blogspot.com/2017/09/kaku-new-protocol-supported-by.html). It can be a solution but I use espruna since a long time and I'm satisfied by the stability and functionalities.

**Additional context**
[image]

mcspr, 2020-03-19:
Does it work with the ZgatewayRF or ZgatewayPilight option in the openmqttgateway?
RCSwitch implementation there looks extended with quite a lot of new protocols

schmurtzm, 2020-03-19:
Yes I’ve tried with ZgatewayPilight and it works (not with ZgatewayRF). But I’m disappointed by the speed of the Pilight implementation in openmqttgateway : it takes about 2 seconds between the reception and the mqtt message.

The few RFlink sources also exposes a [lot of protocols](https://github.com/jwdb/rflink/tree/master/Plugins), there is an [esp conversion](https://github.com/Stef-aap/RFLink-ESP) available on github with few protocols (which includes KAKU) : it can be a way too.

mcspr, 2020-03-22:
Will probably try with pilight option first. Rflink seems like too much manual work involved :)

schmurtzm, 2020-03-23:
Yes, you're probably right, there's a [ESPilight library](https://github.com/puuu/ESPiLight) and a an implementation of this library in [MQTT433gateway](https://github.com/puuu/MQTT433gateway) from the same author so it can helps .... ;)

Just for information, in my investiguation I also found [a source of RFlink-ESP by couin3](https://github.com/couin3/RFLink) that keep compatibility with the original RFlink plugins. The code is more segmented and easy to handle. It's probably less handful than [ESPilight library](https://github.com/puuu/ESPiLight) but still interesting thanks to this compatibility with RFlink plugins.

Speaking about these modified RFlink sources for ESP, I also like a detail in [the first source of RFlink-ESP  by Stef-aap](https://github.com/Stef-aap/RFLink-ESP) that I particularly like  : he said in the readme : "Protocols are more generic, so you need less protocols". And it's really true : with a true RFlink (based on arduino mega 2560) my 433mhz devices are detected as EV1527 , TriState, Eurodomest or AB400D. With RFlink-ESP by Stef-aap all my stuff are detected as EV1527 : it is easier in the source code and easier to make it work with a lot of devices !