xoseperez/espurna #2189: Hi, I have a sonoff basic r3, and can't find the right image for that device. I see several itead sonoff basic images. can anyone help me telling which one is the right

fadudba, 2020-03-19:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by typing `info` via serial, terminal or in the debug tab in the web UI. The relevant information is that surrounded by the scissors-cut lines (`---8<-------`).*
*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*
* *Flash mode*
* *Device brand, model and version*

**Tools used**
* *Desktop operating system*
* *Browser & version*
* *IDE & version*
* *Compiler & version (if not embedded in IDE)*

**Additional context**
*Add any other context about the problem here.*

closed 2020-03-20 by mcspr

mcspr, 2020-03-20:
...-itead-sonoff-basic.bin

-r2-dht, -r2-dallas are meant to specify GPIO pin that is different between revisions.

fadudba, 2020-03-22:
> ...-itead-sonoff-basic.bin
> 
> 
> 
> -r2-dht, -r2-dallas are meant to specify GPIO pin that is different between revisions.

Thanks!!