xoseperez/espurna #2192: UV sensor SI1145

HilverinkJ, 2020-03-20:
Is it possible to support this sensor?

[https://www.adafruit.com/product/1777](url)

mcspr, 2020-03-22:
For reference:
https://cdn-shop.adafruit.com/datasheets/Si1145-46-47.pdf
https://github.com/adafruit/Adafruit_SI1145_Library/blob/master/Adafruit_SI1145.cpp

> The operation of the Si1145/46/47 can be described as two measurement groups bound by some common factors.

> The PS Measurement group consists of the three PS measurements while the ALS Measurement group consists of the Visible Light Ambient Measurement (ALS_VIS), the Infrared Light Ambient Measurement (ALS_IR) and the Auxiliary measurement (AUX). 

> Each measurement group has three measurements each. The Channel List (CHLIST) parameter enables the specific measurements for that measurement grouping.

(PS - Proximity Sensor, ALS - Ambient Light Sensor?)

UI index specifically:
> To enable UV reading, set the EN_UV bit in CHLIST, and configure UCOEF [3:0] to the default values of 0x29, 0x89, 0x02, and 0x00. If the sensor will be under an overlay that is not 100% transmissive to sunlight.

> Typically, after 285 µs, AUX_DATA will contain a 16-bit value representing 100 times the sunlight UV Index. Host software must divide the results from AUX_DATA by 100.

mcspr, 2020-03-22:
Any I2C sensor can be used as an example, or Adafruit library directly (which already does some initialization with ::begin(), so we don't have to -> https://github.com/adafruit/Adafruit_SI1145_Library/blob/5391bd9977f14f85e7c99c16ebcb0abf5b139393/Adafruit_SI1145.cpp#L29-L97)

HilverinkJ, 2020-03-24:
I'm not a programmer but I gif it a try.
in platformio.ini added 
lib_deps =
`https://github.com/adafruit/Adafruit_SI1145_Library#1.1.0`

`[env:hilverink-esp8266-screens]
board = ${common.board_4m}
build_flags = ${common.build_flags_4m1m} -DUSE_CUSTOM_H -DHILVERINK_SCREENS`

As example I used sensor/VEML6075Sensor.h
added file SI1145Sensor.h
`// -----------------------------------------------------------------------------`
`// SI1145 Sensor over I2C`
`// -----------------------------------------------------------------------------`

`#if SENSOR_SUPPORT && SI1145_SUPPORT`

`#pragma once`

`#include <Arduino.h>`
`#include "Adafruit_SI1145.h"`

`#include "I2CSensor.h"`

`class SI1145Sensor : public I2CSensor {`

`     public:`

`       SI1145Sensor() : I2CSensor() {`
`            //_count = 1;`
`            _sensor_id = SENSOR_SI1145_ID;`
`            Adafruit_SI1145 _si1145 = Adafruit_SI1145();`
 `       }`

`        void begin() { `
`           if (!_si1145->begin()) {`
`                return;`
`            };`
`            _ready = true;`
`        }`

  `      // ---------------------------------------------------------------------`
 `       // Sensor API`
 `       // ---------------------------------------------------------------------`

`        // Descriptive name of the sensor`
`        String description() {`
 `           char buffer[25];`
 `           snprintf(buffer, sizeof(buffer), "SI1145 @ I2C (0x%02X)", _address);`
`            return String(buffer);`
`        }`

`        // Descriptive name of the slot # index`
`        String slot(unsigned char index) {`
`            return description();`
  `      };`

 `       // Type for slot # index`
 `       unsigned char type(unsigned char index) {`
 `           if (index == 0) return MAGNITUDE_UVI;`
  `          return MAGNITUDE_NONE;`
  `      }`

   `     // Pre-read hook (usually to populate registers with up-to-date data)`
   `     void pre() {`

  `          _error = SENSOR_ERROR_OK;`

  `          double value;`

 `           value = _si1145->readUV();`
 `           _uvi = value / 100;`
`        }`

`        // Current value for slot # index`
`        double value(unsigned char index) {`
`            if (index == 0) return _uvi; `
 `           return 0;`
 `       }`

`    protected: `
`        Adafruit_SI1145 * _si1145 = NULL;`
`        double _uvi = 0;`
};`
   
`#endif // SENSOR_SUPPORT && SI1145_SUPPORT`

In config/custom.h added
`#define I2C_SDA_PIN 4` 
`#define I2C_SCL_PIN 5 `
`#define BMX280_SUPPORT 1`
`#define BH1750_SUPPORT 1`
`#define SI1145_SUPPORT 1`

In sensors.ino added
`    #if SI1145_SUPPORT
    {
        SI1145Sensor * sensor = new SI1145Sensor();
        sensor->setAddress(SI1145_ADDRESS);
        _sensors.push_back(sensor);
    }
    #endif`

In types.h added
`#define SENSOR_SI1145_ID            39`

In board.ino added
`    #if SI1145_SUPPORT`
`        "SI1145 "`
`    #endif``

Now there is this error. 

[image]

HilverinkJ, 2020-03-24:
Sorry

mcspr, 2020-03-24:
(I have reformatted your comment a bit, instead of \` line \` not it it has \`\`\`cpp at the top and \`\`\` at the end of code block. That way formatting works)

```cpp
SI1145Sensor() : I2CSensor() {
            //_count = 1; <<<<
            _sensor_id = SENSOR_SI1145_ID;
            Adafruit_SI1145 _si1145 = Adafruit_SI1145(); <<<<
        }
```

`_si1145` attached to the class that you later try to use is uninitialized. Line in question creates _si1145 variable local to the SI1145Sensor() constructor function, not the one you later try to use as `_si1145->readUV();`

So just:
```cpp
_si1145 = new Adafruit_SI1145();
```
Note that `_count = 1` should also be there, it won't detect any magnitudes because default count is  0.

HilverinkJ, 2020-03-24:
That works. Thanks.

One thing is just missing. There is no sensor address SI1145 @ I2C (0x00) visible.

[image]

mcspr, 2020-03-24:
It is never set.
https://github.com/xoseperez/espurna/blob/387d4cea58736e2fd0fec971fe3854cd1b29772a/code/espurna/sensors/AM2320Sensor.h#L52-L53
Technically, it should be a good practice to add such check to make sure device is actually there. This could happen some time before adafruit::begin(), returning from ours begin() if address is 0

HilverinkJ, 2020-03-24:
I added this. It looks great now...

It's my intention to close my sunscreens when there is no rain (Digatal #0) sensor.
It looks for me like a 2 stap rokket. With rules I turn on/off Switch #0.  
`$digital0 0 gt 0 relay`
[image]
Switch #0 turns domoticz device on/off
[image]
Is there a simple whay to do soo?

mcspr, 2020-03-27:
I guess we can send digital sensor directly, you don't want additional 'relay' to trigger the action?

We can:
- send sensor numeric value directly. So it would send by itself:
```json
{"idx":224,"nvalue":$digital0,"svalue":""}
```
Although, I am not sure what nvalue *is* exactly; integer, float, unsigned etc. In case of switch it seems to react on either 0 or 1 as inputs.
- support something lik e`$digital0 224 dcz_nvalue`, which would do the same, but perhaps with more conditions attached (like, depending on some other sensor)

Note that I don't use Domoticz, so I am only guessing what it actually needs as input data. Maybe we are missing some new API that can already do that.

HilverinkJ, 2020-03-27:
> I guess we can send digital sensor directly, you don't want additional 'relay' to trigger the action?
Yes that's it..

[https://www.domoticz.com/wiki/MQTT]
Type light/switch to turn on/off.
{"command": "switchlight", "idx": 224, "switchcmd": "On" }
{"command": "switchlight", "idx": 224, "switchcmd": "Off" }

mcspr, 2020-03-28:
I'll add the sensor code to the dev branch then, since it is working.

---

My main question how exactly we send domoticz message.
Our generic sensor output is float, so we can only use svalue (as wiki/MQTT suggests to look at the code, nvalue is strictly integer value). I would assume switch would not react to svalue like "1.0"?

RPN has more lenience, but we still need to have some code to send the data. For example:
rpnrules.ino code:
```cpp
(down in the rpnSetup())
...
    rpn_operator_set(_rpn_ctxt, "dcz_switchlight", [](rpn_context& ctxt) {
        float cmd, idx;
        rpn_stack_pop(ctxt, idx);
        rpn_stack_pop(ctxt, cmd);
        char buffer[128] = {0};
        snprintf(
            buffer, sizeof(buffer),
            "{\"command\":\"switchlight\",\"idx\":%u,\"switchcmd\":\"%s\"}",
            idx, (cmd >= 0) : "On" ? "Off"
        );
        mqttSendRaw(getSetting("dczTopicIn", DOMOTICZ_IN_TOPIC).c_str(), buffer);
    });
...
```
Where `(cmd >= 0)` can check for $digital0 value

HilverinkJ, 2020-03-28:
OK this will work for me..
How can I implement these changes?

mcspr, 2020-03-29:
Do you mean domoticz message side? Have you tried the snippet above? Just add it at the end of  rpnSetup() function, after `#if NTP_SUPPORT ... #endif` block

HilverinkJ, 2020-03-29:
This results in an compile error.

[image]

mcspr, 2020-03-29:
My bad, fixed the code above.

HilverinkJ, 2020-03-29:
Thanks a lot it's working.

HilverinkJ, 2020-03-30:
The code is working but it is slowly killing my Domoticz server.

mcspr, 2020-03-30:
It does not survive the MQTT message every 6 seconds (or whatever sensor read interval is set to)? Or logging it is the problem?

Another suggested change, maybe we need to delay sensor reading -> rpn until value actually changes?
```diff
diff --git a/code/espurna/rpnrules.ino b/code/espurna/rpnrules.ino
index f9ca9486..c9ff37a8 100644
--- a/code/espurna/rpnrules.ino
+++ b/code/espurna/rpnrules.ino
@@ -12,6 +12,9 @@ Copyright (C) 2019 by Xose Pérez <xose dot perez at gmail dot com>
 #include "relay.h"
 #include "rpnrules.h"

+#include <map>
+#include <limits>
+
 // -----------------------------------------------------------------------------
 // Custom commands
 // -----------------------------------------------------------------------------
@@ -101,13 +104,20 @@ void _rpnConfigure() {

 void _rpnBrokerCallback(const String& topic, unsigned char id, double value, const char*) {

-    char name[32] = {0};
+    static std::map<String, double> last;
+    const bool found = last.find(topic) != last.end();
+
+    if (!found || (fabs(last[topic] - value) > std::numeric_limits<double>::epsilon())) {
+        char name[32] = {0};

-    snprintf(name, sizeof(name), "%s%u", topic.c_str(), id);
-    rpn_variable_set(_rpn_ctxt, name, value);
+        snprintf(name, sizeof(name), "%s%u", topic.c_str(), id);
+        rpn_variable_set(_rpn_ctxt, name, value);
+
+        _rpn_last = millis();
+        _rpn_run = true;
+    }

-    _rpn_last = millis();
-    _rpn_run = true;
+    last[topic] = value;

 }
```

mcspr, 2020-03-30:
Although... If we follow the idea that rules describe the desired state, and not sequence of actions, what I can do instead is add `changed` operator:
```$digital0 changed 224 dcz_sendswitch```
Storing $digital0 previous value and exit rule if it is the same as the last time

I would need to update the rules underlying lib to a more recent version though. Which I wanted to do anyway.

HilverinkJ, 2020-03-30:
How do I handle that changed lib from you?

mcspr, 2020-03-30:
https://github.com/mcspr/rpnlib is the library itself
Changes mostly are internal as seen in this example diff - https://github.com/mcspr/rpnlib/commit/3a04c815e691de6cb9fc39fc4351f7059c7022a5#diff-722036d88083221fb60ee22839fe48f1
Soon, maybe? I can't exactly tell as there are some changes needed for MQTT part too. But we are getting really off-topic here :)

HilverinkJ, 2020-03-30:
I changed the code like this:

`void _rpnBrokerCallback(const String& topic, unsigned char id, double value, const char*) {`
`    static std::map<String, double> last;`
`    const bool found = last.find(topic) != last.end();`
`    if (!found || (fabs(last[topic] - value) > std::numeric_limits<double>::epsilon())) {`
`        char name[32] = {0};`
`        snprintf(name, sizeof(name), "%s%u", topic.c_str(), id);`
`        rpn_variable_set(_rpn_ctxt, name, value);`
`        _rpn_last = millis();`
`        _rpn_run = true;`
`    }`
`    last[topic] = value;`
`}`
But there are still a lot of readings.
[image]

mcspr, 2020-03-30:
2nd try above.
Does this mean sensor read time is set to 1 second?

mcspr, 2020-03-30:
Even a 3rd one... In case of MQTT disconnection, sensor data will be sent asap.

HilverinkJ, 2020-03-31:
Yes sensor read time is set to 1 sec. I have changed this to 6 sec.

mcspr, 2020-03-31:
Rule execution is wired to sensor read, thus messages fire every second. Patch above only protects from that specific instance, not every other condition. But you can disable 'Sticky variables' to purge $digital0 value from memory until sensor sets it again.

HilverinkJ, 2020-03-31:
In Domoticz a sensor is updated with i.e. someting like this;
` {"command": "udevice",  "idx" : 7, "nvalue" : 0, "svalue" : "75.00"}`
a switch is operated by something like this:
`{"command": "switchlight", "idx": 50, "switchcmd": "On" }`
Why using the rules system.
[image]
In case of a digital sensor it's only possible to operate a switch device in domoticz. In that case is it posible to activate a device also just to fill in the device idx number over here.

mcspr, 2020-03-31:
It is exactly how it works with switches, we just send nvalue 0 or 1 to a specific index.
https://github.com/xoseperez/espurna/blob/62f5a34340f3fbe4d62b74d924e8f6506501dffc/code/espurna/domoticz.ino#L252

Like I said above, we only send svalue for sensor and 0 for nvalue:
https://github.com/xoseperez/espurna/blob/62f5a34340f3fbe4d62b74d924e8f6506501dffc/code/espurna/sensor.ino#L1862

So when you were using previous configuration with `relay` rule, it worked around that issue.

mcspr, 2020-04-05:
What was the solution in the end? Do we still need to change something / add `changed` / have a different way to send sensor data?

HilverinkJ, 2020-04-05:
You can solve it like this;
file types.h
`#define HUMIDITY_NORMAL                 0       // > %30`
`#define HUMIDITY_COMFORTABLE        1       // > %45`
`#define HUMIDITY_DRY                          2       // < %30`
`#define HUMIDITY_WET                          3       // > %70`

file sensors.ino
 `   #if DOMOTICZ_SUPPORT`
 `   {`
 `       char key[15];`
 `       snprintf_P(key, sizeof(key), PSTR("dczMagnitude%d"), index);`
 `       if (magnitude.type == MAGNITUDE_HUMIDITY) {`
`            int status;`
`            if (value > 70) {`
`                status = HUMIDITY_WET;`
`            } else if (value > 45) {`
 `               status = HUMIDITY_COMFORTABLE;`
 `           } else if (value > 30) {`
 `               status = HUMIDITY_NORMAL;`
`           } else {`
 `               status = HUMIDITY_DRY;`
 `           }`
 `           char status_buf[5];`
 `           itoa(status, status_buf, 10);`
 `           domoticzSend(key, buffer, status_buf);`
 `       } else {`
 `           domoticzSend(key, 0, buffer);`
  `      }`
 `   }`
`    #endif // DOMOTICZ_SUPPORT`
`}`

HilverinkJ, 2020-04-05:
Sorry this is on the wron place. #2208

HilverinkJ, 2020-04-06:
> What was the solution in the end? Do we still need to change something / add changed / have a different way to send sensor data?

First off all I will appreciate if you are willing to adopt the SI1145 sensor to the system.
And yes there is something to change. I discovered that we have a huge misunderstanding.
[image]

On this place do I have to define a Domoticz device that have to be updated with the found data.
In case of switch # 1 is that idx 218, a domoticz **switch device** also. 
[image]

In case of a domoticz sensor you have to send nvalue and svalue, but incase of a dcz-switch you have to send On or Off.
And in case of a digital sensor [0:1] it's only possible to define a domoticz switch device so, in that case you have to send On or Off also.
[image]

mcspr, 2020-04-07:
> dcz-switch you have to send On or Off.

Can be as easy as this?
```cpp
    // Special case to allow us to use it with switches directly
    } else if (MAGNITUDE_DIGITAL == type) {
        int nvalue = (buffer[0] >= 48) ? (buffer[0] - 48) : 0;
        domoticzSend(key, nvalue, buffer);
```
Am able to control switch via `'{"idx": 2,"nvalue": 0,"svalue": "1"}'` and `'{"idx": 2,"nvalue": 1,"svalue": "1"}'`, it does not care about the svalue property.

HilverinkJ, 2020-04-07:
I'm not so sure about that because my Domoticz server on Synology is killed after a couple of hours.
To be sure to have unwanted changes, I have cloned espurna new on my system and will test this tonight.

mcspr, 2020-04-07:
After https://github.com/xoseperez/espurna/pull/2215 it should send digital sensor data exactly like switch data, assuming relays are working fine?
Note that it will send data every 'report', each N readings. This can be changed to 'send every change' by `snsMaxDelta3` to `1.0` (`set snsMaxDelta3 1.0` in the debug panel command line).
*If I am wrong about `3` for index, check out what `magnitudes` command outputs, at the beginning of the `digital/...` topic.*

PR #2216 (2020-04-07): sns: add SI1145 sensor
Resolve #2192 (at least sensor issue by itself)
closed 2020-04-07 by mcspr