xoseperez/espurna #2193: Install Tasmota 8.2.0 with Tasmotizer on an Magic Home ESP-12S

Robert126, 2020-03-21:
I'm trying to flash my Magic Home ESP-12S RGB led driver with Tasmota 8.2.

1) I downloaded Tasmota.bin (version 8.2.0 Elliot) file from https://github.com/arendst/Tasmota/releases
2) followed installation instructions https://github.com/tasmota/tasmotizer#installation-and-how-to-run and installed Tasmotizer. 
pip3 install Tasmotizer
run Tasmotizer.py

3) Run Tasmotizer 
select port: tty.usbserial-1420
BIN file open tasmota.bin
select Backup original firmware
select Erase before flashing

4) Follow instructions https://github.com/arendst/Tasmota/wiki/MagicHome-LED-strip-controller

5) after: 
Connect the 12V power supply to MagicHome. As GPIO00 is connected to GND, board will go into flash mode. Disconnect GPIO00 after few seconds.
I receive a "Failed to connect to ESP8266: Timed out waiting for packet header"

The problem report gives the following

Process:               Python [54807]
Path:                  /usr/local/Cellar/python@3.8/3.8.1/Frameworks/Python.framework/Versions/3.8/Resources/Python.app/Contents/MacOS/Python
Identifier:            org.python.python
Version:               3.8.1 (3.8.1)
Code Type:             X86-64 (Native)
Parent Process:        zsh [96522]
Responsible:           Terminal [30460]
User ID:               502

Date/Time:             2020-03-21 15:54:02.581 +0100
OS Version:            Mac OS X 10.15.3 (19D76)
Report Version:        12
Anonymous UUID:        CB8D7D1B-4D25-7285-2C0A-06A3E6E7D0F2

Sleep/Wake UUID:       EB406C93-2835-4465-8289-D6C1294FA353

Time Awake Since Boot: 210000 seconds
Time Since Wake:       4500 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff717d97fa __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff71896bc1 pthread_kill + 432
2   libsystem_c.dylib             	0x00007fff71760a1c abort + 120
3   QtCore                        	0x000000010498bde9 qt_message_fatal(QtMsgType, QMessageLogContext const&, QString const&) + 9
4   QtCore                        	0x000000010498d534 QMessageLogger::fatal(char const*, ...) const + 202
5   QtCore                        	0x000000010499460d QThread::~QThread() + 189
6   QtCore.abi3.so                	0x00000001046bc7ac sipQThread::~sipQThread() + 44
7   sip.cpython-38-darwin.so      	0x00000001050644d4 forgetObject + 180
8   sip.cpython-38-darwin.so      	0x00000001050640ee sipWrapper_dealloc + 14
9   org.python.python             	0x0000000103bd2247 subtype_dealloc + 764
10  org.python.python             	0x0000000103bb9ff9 dict_dealloc + 172
11  sip.cpython-38-darwin.so      	0x00000001050622e2 sipSimpleWrapper_clear + 178
12  sip.cpython-38-darwin.so      	0x00000001050642d9 sipWrapper_clear + 25
13  sip.cpython-38-darwin.so      	0x00000001050640f6 sipWrapper_dealloc + 22
14  org.python.python             	0x0000000103bd2247 subtype_dealloc + 764
15  org.python.python             	0x0000000103ba747d frame_dealloc + 155
16  org.python.python             	0x0000000103b94a4c function_code_fastcall + 208
17  org.python.python             	0x0000000103b96afe method_vectorcall + 256
[... 517 lines cut ...]
__UNICODE                          564K        1 
mapped file                      369.4M       34 
shared memory                      632K       14 
===========                     =======  ======= 
TOTAL                              1.1G     1008 
TOTAL, minus reserved VM space     1.1G     1008 

Model: MacBookPro11,4, BootROM 196.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 2,2 GHz, 16 GB, SMC 2.29f24
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453120
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453120
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x152), Broadcom BCM43xx 1.0 (7.77.106.3 AirPortDriverBrcmNIC-1440.1)
Bluetooth: Version 7.0.3f5, 3 services, 18 devices, 1 incoming serial ports
Network Service: AirPort, AirPort, en0
Network Service: USB 10/100/1000 LAN, Ethernet, en4
Serial ATA Device: APPLE SSD SM0512G, 500,28 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
USB Device: USB2.0-Serial
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1

Hopefully some one can help out?

Robert

Robert126, 2020-03-21:
I changed to tasmota-lite.bin (version 8.2.0 Elliot)

Robert@Macintosh ~ % tasmotizer.py
esptool.py v2.8
Serial port /dev/tty.usbserial-1420
Connecting........_
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: a0:20:a6:0d:27:d0
Uploading stub...
esptool.py v2.8
Serial port /dev/tty.usbserial-1420
Connecting...
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: a0:20:a6:0d:27:d0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.9s
Compressed 477328 bytes to 332518...

Message is:
Failed to write compressed data to flash after seq 0 (result was C100)

Any suggestions?

mcspr, 2020-03-27:
No idea, you should probably ask on Tasmota discord instead?

closed 2020-03-27 by mcspr