xoseperez/espurna #2194: Need help to flash Smart Life wifi plug from Arduino IDE

stiw47, 2020-03-22:
Hi,

Vendor is Lonsonho and it is this switch: https://a.aliexpress.com/_d6zi1aD 
I had converted device with tuya-convert successfully, device worked few days with espurna-1.14.1-smartlife-mini-smart-socket.bin I had some issues with losing wifi, so I wanted to re-flash the firmware. Unfortunately, I flashed espurna-1.14.1-espurna-core-1MB.bin and didn't knew that this one has no web interface. So I am stuck with only possibility to build firmware in Arduino IDE and flash it via network port, if I understood correct in two days of googling and struggling :)

Unfortunately, I am not the dev and it seems this is above my skills :(
I done everything from here: https://github.com/xoseperez/espurna/wiki/ArduinoIDE , but obviously missed something, since no matter what config I make (editing of code/espurna/config/arduino.h), I always finish with lot of errors in Arduino console.

Last one case: 

- Additional board manager https://arduino.esp8266.com/stable/package_esp8266com_index.json added to preferences 
- "Generic ESP8266 Module" was selected as board in Tools/Board
- boards.local.txt was added to C:\Users\iys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3
- All libraries installed, either from Sketch/Include Library/Manage Libraries and from zip files
- MQTT_MAX_PACKET_SIZE was raised to 1024 in C:\Users\iys\Documents\Arduino\libraries\PubSubClient\src\PubSubClient.h
- folloving was enabled in code/espurna/config/arduino.h: #define SMARTLIFE_MINI_SMART_SOCKET ; #define OTA_ARDUINOOTA_SUPPORT 1 ; #define WEB_SUPPORT 0 ; #define DHT_SUPPORT 1
- didn't touch code/espurna/config/general.h
- flash size set to 1M (1 EEPROM Sector, no SPIFFS)
- everything was set same as on your screenshot below this link: https://github.com/xoseperez/espurna/wiki/ArduinoIDE#building--flashing-the-firmware

After all of this, I got following error: 

> Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), no dtr (aka ck), 26 MHz, 40MHz, DOUT (compatible), 1M (1 EEPROM Sector, no SPIFFS), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200, Disabled (Recommended)"
> 
> button:492: error: 'T' has not been declared
> 
>  unsigned long _buttonGetSetting(const char* key, unsigned char index, T default_value) {
> 
>                                                                        ^
> 
> mqtt:282: error: variable or field '_mqttApplySetting' declared void
> 
>  void _mqttApplySetting(T& current, T& updated) {
> 
>                         ^
> 
> mqtt:282: error: 'T' was not declared in this scope
> 
> mqtt:282: error: 'current' was not declared in this scope
> 
>  void _mqttApplySetting(T& current, T& updated) {
> 
>                            ^
> 
> mqtt:282: error: 'T' was not declared in this scope
> 
>  void _mqttApplySetting(T& current, T& updated) {
> 
>                                     ^
> 
> mqtt:282: error: 'updated' was not declared in this scope
> 
>  void _mqttApplySetting(T& current, T& updated) {
> 
>                                        ^
> 
> mqtt:290: error: variable or field '_mqttApplySetting' declared void
> 
>  void _mqttApplySetting(T& current, const T& updated) {
> 
>                         ^
> 
> mqtt:290: error: 'T' was not declared in this scope
> 
> mqtt:290: error: 'current' was not declared in this scope
> 
>  void _mqttApplySetting(T& current, const T& updated) {
> 
>                            ^
> 
[... 238 lines cut ...]
> C:\Users\iys\OneDrive - P3 Group\Desktop\espurna-dev\code\espurna\homeassistant.ino:401:26:   required from here
> 
> C:\Users\iys\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObjectSubscript.hpp:43:3: error: no type named 'type' in 'struct ArduinoJson::TypeTraits::EnableIf<false, ArduinoJson::JsonObjectSubscript<const char*> >'
> 
> C:\Users\iys\OneDrive - P3 Group\Desktop\espurna-dev\code\espurna\mqtt.ino: In function 'void mqttFlush()':
> 
> mqtt:874: error: 'ArduinoJson::DynamicJsonBuffer' has no member named 'clear'
> 
>      jsonBuffer.clear();
> 
>                 ^
> 
> exit status 1
> 'T' has not been declared
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.

There were another errors also with different configuration in code/espurna/config/arduino.h since I tried different ways 10 times at least.
I would like to flash any firmware which would enable wifi interface, so I can come back to espurna-1.14.1-smartlife-mini-smart-socket.bin from web interface/admin panel/upgrade , if this is possible.

I hope that above is not too much confusing, I am ready to provide any additional needed info.

Please consider that I am not dev and be gently :) , many thanks in advance.

mcspr, 2020-03-22:
(woosh :)
> Arduino: 1.8.5

I think I already saw this error some time ago... Please try updating to a more recent version, arduino.cc lists 1.8.12 as the latest one.

Not sure which ArduinoJson version is used?
https://github.com/bblanchon/ArduinoJson/archive/v5.13.5.zip should be used to match what we use to build .bin

stiw47, 2020-03-22:
> (woosh :)
>
>
> I think I already saw this error some time ago... Please try updating to a more recent version, arduino.cc lists 1.8.12 as the latest one.
>
> Not sure which ArduinoJson version is used? [...]

Thanks for fast answer. I installed Arduino IDE 1.8.12 and installed ArduinoJson library 5.13.5. With Arduino library, I made mistake: Installed both - from Sketch/Include Library/Manage Libraries and also manually imported .zip from your link. So new error is below:

> c:/users/iys/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot open linker script file eagle.flash.1m0m1s.ld: No such file or directory
> 
> collect2.exe: error: ld returned 1 exit status
> 
> Multiple libraries were found for "ArduinoJson.h"
>  Used: C:\Users\iys\Documents\Arduino\libraries\ArduinoJson
>  Not used: C:\Users\iys\Documents\Arduino\libraries\ArduinoJson-5.13.5
> exit status 1
> Error compiling for board Generic ESP8266 Module.

I think that only lines related with xtensa-lx106-elf-gcc are important, correct? I mean, part related with ArduinoJson library is stating that I have multiple same libraries and that compile will use one installed from library manager, not manually imported and I installed 5.13.5 from library manager.

[image]

 Can I get more help :)

stiw47, 2020-03-22:
Yup, I removed double library and end up only with following error:

> c:/users/iys/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot open linker script file eagle.flash.1m0m1s.ld: No such file or directory
> 
> collect2.exe: error: ld returned 1 exit status
> 
> exit status 1
> Error compiling for board Generic ESP8266 Module.
>

stiw47, 2020-03-22:
I found a linker script "eagle.flash.1m0m1s.ld". There is in git source code folder: espurna-dev\dist\ld\latest I tried to put eagle.flash.1m0m1s.ld script into C:\Users\iys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3 but got same error as above again. Am I on right direction and where should this script be located? :)

mcspr, 2020-03-22:
See https://github.com/xoseperez/espurna/tree/dev/dist/ld
.ld files should go into $ARDUINO_ESP8266/tools/sdk/ld directory

stiw47, 2020-03-22:
> 
> 
> See https://github.com/xoseperez/espurna/tree/dev/dist/ld
> .ld files should go into $ARDUINO_ESP8266/tools/sdk/ld directory

Thanks again, but no success. I put all linker scripts in `C:\Users\iys\AppData\Local\Arduino15\packages\esp8266\tools\sdk\ld` and got same error again. This is content of my `C:\Users\iys\AppData\Local\Arduino15\packages\esp8266\tools\sdk\ld` :

[image]

Just to mention: There were no `sdk/ld` directory inside of `C:\Users\iys\AppData\Local\Arduino15\packages\esp8266\tools\`, I created it manually.

Mention2: There are two more lines on top of the error code. They appear straight on beginning of compile process, but I think they are related with no internet connection, cause I'm connected to AP of smart plug because of flashing. So below would be complete error log:

> Error downloading https://arduino.esp8266.com/stable/package_esp8266com_index.json
> Error downloading https://downloads.arduino.cc/packages/package_index.json
> c:/users/iys/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot open linker script file eagle.flash.1m0m1s.ld: No such file or directory
> 
> collect2.exe: error: ld returned 1 exit status
> 
> exit status 1
> Error compiling for board Generic ESP8266 Module.

mcspr, 2020-03-22:
I mean `tools` dir inside of `C:\Users\iys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3`, you can see stock scripts there. Our files are only supposed to go inside of the current Core directory, they are not shared just like boards.local.txt

stiw47, 2020-03-23:
> 
> 
> I mean `tools` dir inside of `C:\Users\iys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3`, you can see stock scripts there. Our files are only supposed to go inside of the current Core directory, just like boards.local.txt

Thanks a lot once again, compile problem was solved.

never ending story, new problem appear :) . At the end of compile process, when should start to upload sketch, it says:

`Type board password to upload a new sketch`

If I either leave password field empty and click on "Upload" button, or type last admin password set by me, it says:

```
Authenticating...FAIL
09:17:40 [ERROR]: Authentication Failed
```

If I type default `fibonacci` password, which is btw current password, I know that because there was hard reset and I'm using fibonacci now to connect on switch AP, it says:

```
Authenticating...FAIL
09:17:49 [ERROR]: No Answer to our Authentication
```

This is complete log of compiling and upload try, including errors:

> Executable segment sizes:
> 
> 
> IROM   : 564832          - code in flash         (default or ICACHE_FLASH_ATTR) 
> 
> 
> IRAM   : 29584   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
> 
> 
> DATA   : 1568  )         - initialized variables (global, static) in RAM/HEAP 
> 
> 
> RODATA : 9132  ) / 81920 - constants             (global, static) in RAM/HEAP 
> 
> 
> BSS    : 30224 )         - zeroed variables      (global, static) in RAM/HEAP 
> 
> 
> Sketch uses 605116 bytes (59%) of program storage space. Maximum is 1023984 bytes.
> Global variables use 40924 bytes (49%) of dynamic memory, leaving 40996 bytes for local variables. Maximum is 81920 bytes.
> Authenticating...FAIL
> 09:17:36 [ERROR]: No Answer to our Authentication
> Authenticating...FAIL
> 09:17:40 [ERROR]: Authentication Failed
> Authenticating...FAIL
> 09:17:49 [ERROR]: No Answer to our Authentication
> Authenticating...FAIL
> 09:18:00 [ERROR]: Authentication Failed

I'm not sure should I close this issue, since now have different issue?

Thx.

mcspr, 2020-03-23:
You might want to check the firewall? espota.py script is supposed to bind to a local port which will be used to communicate with the other side.
Also see https://github.com/xoseperez/espurna/wiki/OTA#using-espotapy. You should have both python install and espota, in a global `tools` and Core's respectively.

closed 2020-03-23 by mcspr

stiw47, 2020-03-23:
> 
> 
> Also see https://github.com/xoseperez/espurna/wiki/OTA#using-espotapy. You should have both python install and espota, in a global `tools` and Core's respectively.
> 

Actually, from beginning, I only needed this. Can't believe, after whole day of acrobatics in Arduino IDE and trying to compile code, problem solved in one single command :)

> python espota.py --progress --ip 192.168.4.1 --auth fibonacci --file espurna-1.14.1-smartlife-mini-smart-socket.bin

Thanks anyway :)