xoseperez/espurna #2196: Conversion from joules to kwh gives always zero

dreamfalcon, 2020-03-25:
Hi,

I updated all my devices to 1.14.1 and started having a problem with the Total Energy value.
In some devices if I select kWh the value is always zero. On most devices is displaying the correct decimal value.
Tried reseting the energy value, but no luck.

[image]

[image]

dreamfalcon, 2020-03-25:
Rebooting the devices solved the problem.

closed 2020-05-31 by stale[bot]

mcspr, 2020-05-31:
Should be fixed in dev btw. The specific issue was with decimal places and unit conversion not being properly reloaded from runtime settings.