xoseperez/espurna #2199: LED option is shown on webUI even though no LEDs defined.

davebuk, 2020-03-29:
I have been using a sonoff basic and PIR sensor to control some outside lights. The original setup used `GPIO3 (Rx)` pin as it was the easiest to access. For some reason, the sonoff has stopped responding on `GPIO3` and wouldn't report the button presses. Even pulsing the pin with a piece of wire between GND and 3V doesn't show a response in the webUI debug screen.

I have changed the build to use the `LED GPIO (13)` and removed the LED from the board. I commented out the `#define LED` statements in my custom.h file. The firmware built still had the `LED` option enabled so I don't think the GPIO would have allowed me to use it as an input? As it was 1.14.2dev there was an option to change the GPIO for the LED, but whatever setting I tried to save, it would revert back to there being an `LED` on `GPIO 13`.

I have built again setting `#define LED_SUPPORT 0` which has removed the LED options and `GPIO13` is working as a button input for my PIR.

In summary, by **not** defining any LEDs for an environment, should `LED_SUPPORT  0` be set by default? This would be the same as when `SENSOR_SUPPORT` is not shown as an option if a sensor hasn't been defined.

mcspr, 2020-03-30:
> As it was 1.14.2dev there was an option to change the GPIO for the LED, but whatever setting I tried to save, it would revert back to there being an LED on GPIO 13.

I have missed the ledGPIO part, .js file here need updating + gulp rebuild
https://github.com/xoseperez/espurna/blob/387d4cea58736e2fd0fec971fe3854cd1b29772a/code/html/custom.js#L277
+ `, "ledGPIO"

Setting was never updated, but `set ledGPIO0 ...` would've worked.

mcspr, 2020-03-30:
> In summary, by not defining any LEDs for an environment, should LED_SUPPORT 0 be set by default? This would be the same as when SENSOR_SUPPORT is not shown as an option if a sensor hasn't been defined.

Might be useful too, when all LED[1-8]_PIN == GPIO_NONE. But unlike sensor support it does not actually take a lot of space, which is the main reason of disabling support for the other.

closed 2020-03-30 by mcspr

davebuk, 2020-03-31:
So **without** setting `LED_SUPPORT 0` I should now be able to change `LED1` to another or even **none** for its GPIO assignment and use the old LED GPIO as a `BUTTON` input rather than an output as defined in my custom.h file? Or is it still best to set `LED_SUPPORT 0` if there is no requirement for LEDs in custom.h environment build?

mcspr, 2020-03-31:
The issue was that every entry on the LEDs page was creating `ledGPIO => something` without index  instead of `ledGPIO# => something#` with index, which led module expects. Thus LED1_PIN was never overridden (or the existing ledGPIO0 kv that was added every boot in the previous versions).

LED_SUPPORT 0 is fine though? You just need to make sure there is no default LED1_PIN based on the board type. You can always define MANUFACTURER and DEVICE flags to avoid using any specific base

PR #2202 (2020-03-30): webui: ledGPIO is a list
fix #2199