xoseperez/espurna #2208: Domoticz device "Barometer" crashing.

HilverinkJ, 2020-04-02:
Defind complete new device:
[image]
With device idx 233:
[image]
Reading data from BME@80:
[image]
JSON result from device 233:
[image]
Updating data to idx 233:
[image]
No values displayed anymore:
[image]
In JSON result the value "Barometer: 1008.75" is not there anymore.

JSON wiki information about "Barometer"
[https://www.domoticz.com/wiki/Domoticz_API/JSON_URL's#Barometer](url)
The above sets the parameters for a Barometer device from hardware type 'General'
IDX = id of your device (This number can be found in the devices tab in the column "IDX")
BAR = Barometric pressure
BAR_FOR = Barometer forecast

Barometer forecast can be one of:
0 = Stable
1 = Sunny
2 = Cloudy
3 = Unstable
4 = Thunderstorm

In file "types.h" there is no ENVIROMMENTAL information for barometer.

mcspr, 2020-04-05:
Huh. Sending:
```
$ mosquitto_pub -t 'domoticz/in' -m '{"idx": 3,"nvalue": 0,"svalue": "1020.1"}'
$ mosquitto_pub -t 'domoticz/in' -m '{"idx": 3,"nvalue": 0,"svalue": "1020.1;3"}'
```
First message shows blank barometer, second one correctly displays the value.

But how exactly we should determine "forecast" value?

HilverinkJ, 2020-04-05:
You can find all that information here:
https://www.domoticz.com/wiki/Domoticz_API/JSON_URL's#Barometer

mcspr, 2020-04-05:
Nope. What is forecast value exactly from the Domoticz POV and why it wants us to set it?
 
Do we need to calculate some values on the device itself or what?
https://en.wikipedia.org/wiki/Cold_front

Or does it mean to simply translate value like these do:
https://google.com/search?q=analog+barometer&tbm=isch
(Which is kind of strange, since it knows the value already. Which is why I thought about the first option)

mcspr, 2020-04-05:
(re https://github.com/xoseperez/espurna/issues/2208#ref-issue-585359121)

I have found piece of code that domoticz uses:
https://github.com/domoticz/domoticz/blob/6027b1d9e3b6588a901de42d82f3a6baf1374cd1/hardware/I2C.cpp#L1092-1193
Which is not a simple value range match like humidity

HilverinkJ, 2020-04-05:
Wow.. No thats not simple

HilverinkJ, 2020-04-05:
I think this is OK.

`			int nforecast = bmpbaroforecast_cloudy;` 
`			if (pressure <= 980)`
`				nforecast = bmpbaroforecast_thunderstorm;`
`			else if (pressure <= 995)`
`				nforecast = bmpbaroforecast_rain;`
`			else if (pressure >= 1029)`
`				nforecast = bmpbaroforecast_sunny;`

mcspr, 2020-04-05:
Maybe. It does not match `(x > 995) && (x < 1029)`? We can also just send invalid number like -1 or 5 to set barometer sensor to the Unknown state.
 
.pdf also describes a simple approach:
> Below simple C code from the DEMOAPEXSENSOR demo kit calculates which weather symbol to display on the LCD screen.
• **CurrentAltitude** - (m) Altitude in meters that is entered into the system by the user for that current static location.
• **Pweather** - (kPa) Pressure at the current altitude. It is calculated using the Height (m) to Pressure (kPa) exponential
equation, inputting CurrentAltitude in meters. This is the ideal pressure for the current location on a stable relatively
sunny day.
• **decPcomp** - (kPa) Value of compensated pressure from MPL115A. 
```python
 ///////////////////////
//SIMPLE WEATHER SECTION
///////////////////////
Pweather = (101.3 * exp(((float)(CurrentAltitude))/(-7900)));
Simpleweatherdiff = decPcomp - Pweather;
if (Simpleweatherdiff > 0.25)
Simpleweatherstatus = 1; // Sunny
if ((Simpleweatherdiff <= 0.25) or (Simpleweatherdiff >= (-0.25)))
Simpleweatherstatus = 2; // Cloudy
if (Simpleweatherdiff < (-0.25))
Simpleweatherstatus = 4; // Thunderstorm / Rain
```
Which can at least match 3 states, but not Stable / Unstable

HilverinkJ, 2020-04-06:
Please simplify it a bit:

`#define BAROMETER_UNKNOWN 		5	<  980	`					
`#define BAROMETER_THUNDERSTORM 	4	<= 990	`					
`#define BAROMETER_UNSTABLE 		3	<= 1008		`				
`#define BAROMETER_CLOUDY_RAIN 	2	<= 1020		`				
`#define BAROMETER_SUNNY 		1	>= 1036		`				
`#define BAROMETER_STABLE 		0	>  1036		`

[image]

HilverinkJ, 2020-04-06:
#define BAROMETER_SUNNY 1 <= 1036

mcspr, 2020-04-07:
After https://github.com/xoseperez/espurna/pull/2215 it adds `;-1` to the pressure reading string before sending it to domoticz.
I will add forecast stuff eventually, I'd like to keep the original intent and actually try to have some basis on our forecast :) Maybe with temperature, too. I don't see domoticz code basing any results on temperature changes, while it probably should.