xoseperez/espurna #2209: Pet Drinking Fountain WIFI

Camposmansi, 2020-04-03:
Hi,
Buy https://www.amazon.es/gp/product/B07GZHD7RZ/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&psc=1

Has an ESP8266, attached images...

[image]
[image]

Thank you

ColinShorts, 2020-04-03:
The pictures are not very clear/sharp. I count 3 led's, a buzzer/speaker and a switch, and it doesn't look like there is anything complicated in there. The only issue might be controlling the water level asking that is handled by the esp.

Have you tried flashing espurna or identifying the pins for the LEDs, etc?

Camposmansi, 2020-04-03:
it's a magnetic switch

ColinShorts, 2020-04-03:
You should be able to identify most of it using the debug console (assuming the buzzer just makes a noise when voltage is applied).

Start by taking a full backup of the ROM (instructions should be on the wiki) and then flash any generic image.

When you get it running, login and had to the debug console. Type "gpio" and you'll see the state of the pins. Press and hold the button, run gpio again and you will hopefully see one of the gpio states has changed. Take note. Repeat for the magnetic switch....

Then you can check each of the remaining gpio like this:

gpio X 1

Or

gpio X 0

Hopefully you'll see one of the LEDs light up.

Camposmansi, 2020-04-03:
can i make a backup ??? I am going to try

I don't see any generic image, sorry

I don't know how to read but I'm also going to try ...

Then he said something

ColinShorts, 2020-04-03:
https://github.com/xoseperez/espurna/wiki/Backup

https://github.com/xoseperez/espurna/releases

I guess the 1mb core image would work for testing.

Camposmansi, 2020-04-03:
Found all ....
red led = MTDI GPIO12
Green led = GPIO5
White led = SD_DATA_2 GPIO9
Switch = GPIO0
Water Sensor = MTMS GPIO14
Relay source = GPIO15
Now how do I spurn it and it works well?

Thank you very much but thank you very much

Camposmansi, 2020-04-04:
I have observed that the speaker is missing, but I did not find the gpio, but it is something that is not important either, but when I disassemble it again I look for it more carefully