xoseperez/espurna #2219: Power plug specific settings not respected

reaper7, 2020-04-10:
Today I updated my HYKKER_SMART_HOME_POWER_PLUG to latest dev 
but some defined for this device settings like SENSOR_ENERGY_UNITS and probably RATIOS are not respected.
So...device show more than ~330V instead ~236 and Energy units as J (instead defined kWh)

When I back to dev commit from Apr 2, 2020 then all works fine again.

is the reason for this? :
https://github.com/xoseperez/espurna/commit/cae50fa5445afac661e89eed86b5e2122da8a9d7
and will be fixed soon? 
```
Further sensor refactoring... soon.
```

mcspr, 2020-04-10:
> RATIOS

That is something I missed bc it was in the configuration part, but you should be able to use runtime overrides in WebUI? Or see following key values <- flags association:
`pwrRatioC` <- `HLW8012_CURRENT_RATIO`
`pwrRatioV` <- `HLW8012_VOLTAGE_RATIO`
`pwrRatioP` <- `HLW8012_POWER_RATIO`

> SENSOR_ENERGY_UNITS

Is reset to default in settings when previous value is used, but should work when set as a flag? See current `eneUnits` setting value. I will probably revert that part though :/ Currently `dev` version tracks global enumeration of 'units', and previously it used magnitude-specific index, so numbers are different. But, it should probably keep using local indexes to allow us to change things in the future

reaper7, 2020-04-10:
Both can be solved by setting the correct values in the web interface
but I often do some experiments on devices and equally often I have to do Factory Reset via webinterface...
..so, it would be good to have fully working default settings for specific devices from hardware.h :)

I will temporarily use dev from Apr 2, 2020

OT. I don't want to open another issue but
I noticed a problem with the time display.
I have defined server and time zone in custom.h
```
  #define NTP_SUPPORT                 1
  #define NTP_SERVER                  "homegateway.lan"
  #define NTP_TIMEZONE                TZ_Europe_Warsaw
```
the time refresh process is running correctly, 
timezone is also correctly decoded because I see in webinterface:
```
CET-1CEST,M3.5.0,M10.5.0/3
```
but on webinterface I still see time as UTC (18h instead 20h)

mcspr, 2020-04-12:
That was meant to point out to a missing code! Not dismissing the issue :) I'll bundle some other changes in the mean time regarding other configuration options. This will be included:
```patch
From abf5f638fff531b1b500a9061cc3f8918dde40af Mon Sep 17 00:00:00 2001
From: Maxim Prokhorov <prokhorov.max@outlook.com>
Date: Sun, 11 Apr 2020 22:52:36 +0000
Subject: [PATCH] hlw8012: load hardware-specific ratios

---
 code/espurna/sensor.ino | 3 +++
 1 file changed, 3 insertions(+)

diff --git a/code/espurna/sensor.ino b/code/espurna/sensor.ino
index 516a40be..81632ec8 100644
--- a/code/espurna/sensor.ino
+++ b/code/espurna/sensor.ino
@@ -1317,6 +1317,9 @@ void _sensorLoad() {
         sensor->setSEL(getSetting("snsHlw8012SelGPIO", HLW8012_SEL_PIN));
         sensor->setCF(getSetting("snsHlw8012CfGPIO", HLW8012_CF_PIN));
         sensor->setCF1(getSetting("snsHlw8012Cf1GPIO", HLW8012_CF1_PIN));
+        sensor->setCurrentRatio(HLW8012_CURRENT_RATIO);
+        sensor->setVoltageRatio(HLW8012_VOLTAGE_RATIO);
+        sensor->setPowerRatio(HLW8012_POWER_RATIO);
         sensor->setSELCurrent(HLW8012_SEL_CURRENT);
         _sensors.push_back(sensor);
     }
--
2.25.2

```

closed 2020-04-28 by mcspr

reaper7, 2020-05-01:
@mcspr - I checked this with current dev branch and still not working properly.
I compiled only with HYKKER_SMART_HOME_POWER_PLUG uncommented and WEB DEBUG enabled,
so values from HLW8012 are completely incorrect.
Commands via web debug interface return null values for pwrRatioC, pwrRatioP, pwrRatioV...
for e.g. _get pwrRatioC_ return:
```
[482675] [WEBSOCKET] Requested action: dbgcmd
[482677] > pwrRatioC =>
[482677] +OK
```
shouldn't there be values declared for HYKKER_SMART_HOME_POWER_PLUG from hardware.h?

When I rewrite default values for HYKKER via web debug interface:
_set pwrRatioC 25740_ (and the other two values):
```
[650833] [WEBSOCKET] Requested action: dbgcmd
[650841] > pwrRatioC => "25740"
[650841] +OK
```
then the readings are correct.

mcspr, 2020-05-01:
Huh. I do see that. HLW lib resets multipliers aka ratios in begin(), so I kind of missed that setting ratios right after new() does not really change anything. I'll see about configure method then.

Also, some strange c++ feature noticed in the Emon sensor - having structs `A` and `B : public A` we can't have `A::some()`+`A::some(arg)` and try to call `B::some(arg)` if B has an override for `some` without the arg. This somewhat complicates what I wanted to do previously...

mcspr, 2020-05-01:
(re: resetting multipliers)
Defaults are based on resistor values, also set in begin():
https://github.com/xoseperez/espurna/blob/e60c561f7023a3d63eab72e207276ef63ea1754b/code/espurna/sensors/HLW8012Sensor.h#L143
We can also change those in hw.h to match the ratio?

PR #2230 (2020-04-28): Various sensor fixes
fix #2223
fix #2219

- fix domoticz API floats in nvalue
- fix pressure constrain 0...100
- adjust BaseSensor api to allow different units per magnitude
- hlw8012 set energy in pre-read, getter returns last value