xoseperez/espurna #2220: Both can be solved by setting the correct values in the web interface

mcspr, 2020-04-11:
I noticed a problem with the time display.
I have defined server and time zone in custom.h
```
  #define NTP_SUPPORT                 1
  #define NTP_SERVER                  "homegateway.lan"
  #define NTP_TIMEZONE                TZ_Europe_Warsaw
```
the time refresh process is running correctly, 
timezone is also correctly decoded because I see in webinterface:
```
CET-1CEST,M3.5.0,M10.5.0/3
```
but on webinterface I still see time as UTC (18h instead 20h)

_Originally posted by @reaper7 in https://github.com/xoseperez/espurna/issues/2219#issuecomment-612157081_

mcspr, 2020-04-11:
Gist of the issue:
UI receives raw seconds unix timestamp and we use `(new Date(timestamp * 1000)).toISOString()` to print it (Stripping T and Z symbols, as it is ISO string output). This is done because NTPCLientLib used to output unix timestamp in local timezone.
Current NTP library uses UTC, but string still shows the same data :)

- ui code stores timestamp as `now` variable and ticks it +1 every second *locally*
https://github.com/xoseperez/espurna/blob/2dcf1b2bd80ea2474275254bec722aa8905bf6c5/code/html/custom.js#L2187
Purpose of that is to show sort-of current device time, see span[name="now"], input[name="now"]. span[name="ago"] is a separate entity that only tracks seconds since the latest "uptime" received.
- js `toLocaleString()` can render timestamp in browser timezone, but it is kind of pointless.
- device itself already can print datetime via ntpDateTime()

reaper7, 2020-04-11:
so, basically setting the correct time zone only matters when scheduler is used ?

mcspr, 2020-04-11:
Right now, yes, this is only a display issue. See `ntp` terminal output for example.

yo8aiv, 2020-05-29:
hi. i'm kind of having the same issue .. but weird .. when i set UTC+2 .. in debug shows wrong by cutting 2h instead of adding them .. if i put -2 it will do the correct thing while in status page nothing seems to change. 

Probably unrelated !!
using pzem 004t with voltage level shifter : observing that values (power, voltage, etc) aren't changing instantly  .. compared to another pzem016 i have. 
is this normal ?? 

having sensors set to read and report every 1sec

THX.

mcspr, 2020-05-29:
See https://github.com/xoseperez/espurna/blob/e26ba9f52ecd60afbbed2ec13d80f4072dad9607/code/espurna/config/general.h#L1551-1558
`UTC+2` means -2h per the specification, you need to use `UTC-2`. Or, like it happens to work with `-2`

mcspr, 2020-09-30:
Closing via aec13cef63d12ccf1f03a72f55079c6bdb1c120b

closed 2020-09-30 by mcspr